PowerBroker2/ELMduino #75: ELM timeout while calling any function eg .rpm() if ELM RX TX is connected with wire to Arduino

Som4567, 2021-04-06:
Every time I tried to connect ELM with Arduino ( UNO or Mega) it successfully gets connected to ELM but the moment it calls any function it gives error 'ELM timeout'

I face this issue only with the wire connection of RX and TX. If I use FTDI wire to directly read from ELM using ScanTool then it works perfectly.
It works perfectly with BLE and wifi. 

I am using 38400 baud rate for ELM port.

PowerBroker2, 2021-04-06:
Please post your code and all debug printouts

Som4567, 2021-04-09:
Hi @PowerBroker2 

sorry for delay in response. I was out of town so couldn't check it earlier.
Please find attached log.
[ELM_output_debug.txt](https://github.com/PowerBroker2/ELMduino/files/6283820/ELM_output_debug.txt)

PowerBroker2, 2021-04-09:
Looks like you're communicating with the ELM327 successfully, however, the ELM327 can't get the info it needs from the car. Hence the "SEARCHING...UNABLE TO CONNECT". Most likely you need to specify a protocol that works for your vehicle:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.h#L7-L22

You can do so by specifying a protocol in the `begin()` function:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.cpp#L6-L25

Som4567, 2021-04-13:
@PowerBroker2 thanks a lot for your support. It was a wiring issue with MCP2551. Now I can communicate with ELM.

closed 2021-04-13 by Som4567

MGC1STOKI, 2021-04-25:
Im having the same issue with my wired ELM327 even after selecting the correct protocol:

`#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial mySerial(11, 12); // RX, TX
#define ELM_PORT mySerial

ELM327 myELM327;

float voltage;
char protocol = ISO_15765_11_BIT_500_KBAUD;

void setup()
{
  Serial.begin(115200);
  ELM_PORT.begin(38400);
  myELM327.begin(ELM_PORT, protocol);
  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }
  Serial.println("Connected to ELM327");
  delay(2000);
  if (myELM327.sendCommand(DISP_CURRENT_PROTOCOL) == ELM_SUCCESS)
  {
    Serial.print("PROTOCOL: ");
    Serial.println(myELM327.payload);
  }

}

void loop()
{
  float tempVoltage = myELM327.ctrlModVoltage();

  if (myELM327.status == ELM_SUCCESS)
  {
    voltage = tempVoltage;
    Serial.print("Volatage: "); 
    Serial.print(voltage);
    Serial.println(" v");
  }
  else
  {
    Serial.print(F("\tERROR: "));
    Serial.println(myELM327.status);
    delay(100);
    
  }
}`

Using the `DISP_CURRENT_PROTOCOL` command to check what protocol it should be using shows that it is still set to auto even after setting it in the `begin()` function.
The log shows that it is still sending AT TP A0 when presumably i would expect it to send AT TP A6.

I know this is definitely the right protocol for my car as it is what the Torque app works with when set.

> 19:40:34.197 -> Clearing input serial buffer
19:40:34.197 -> Sending the following command/query: AT D
19:40:34.235 -> 	Received char: \r
19:40:34.235 -> 	Received char: O
19:40:34.235 -> 	Received char: K
19:40:34.235 -> 	Received char: \r
19:40:34.235 -> 	Received char: \r
19:40:34.235 -> 	Received char: >
19:40:34.235 -> Delimiter found
19:40:34.235 -> All chars received: OK
19:40:34.340 -> Clearing input serial buffer
19:40:34.340 -> Sending the following command/query: AT Z
19:40:34.340 -> 	Received char: ⸮
19:40:34.340 -> 	Received char: ⸮
19:40:34.340 -> 	Received char: i
19:40:34.340 -> 	Received char: ⸮
19:40:34.340 -> 	Received char: \r
19:40:35.140 -> 	Received char: \r
19:40:35.140 -> 	Received char: \r
19:40:35.140 -> 	Received char: E
19:40:35.140 -> 	Received char: L
19:40:35.140 -> 	Received char: M
19:40:35.140 -> 	Received char: 3
19:40:35.140 -> 	Received char: 2
19:40:35.140 -> 	Received char: 7
19:40:35.140 -> 	Received char:  
19:40:35.140 -> 	Received char: v
19:40:35.140 -> 	Received char: 1
19:40:35.140 -> 	Received char: .
19:40:35.140 -> 	Received char: 5
19:40:35.140 -> 	Received char: \r
19:40:35.140 -> 	Received char: \r
19:40:35.140 -> 	Received char: >
19:40:35.140 -> Delimiter found
19:40:35.177 -> All chars received: iELM327v15
19:40:35.250 -> Clearing input serial buffer
19:40:35.250 -> Sending the following command/query: AT E0
19:40:35.250 -> 	Received char: ⸮
19:40:35.250 -> 	Received char: ⸮
19:40:35.250 -> 	Received char: +
19:40:35.250 -> 	Received char: ⸮
19:40:35.250 -> 	Received char: 5
19:40:35.250 -> 	Received char: ⸮
19:40:35.250 -> 	Received char: O
19:40:35.250 -> 	Received char: K
19:40:35.250 -> 	Received char: \r
19:40:35.250 -> 	Received char: \r
19:40:35.285 -> 	Received char: >
19:40:35.285 -> Delimiter found
19:40:35.285 -> All chars received: 5OK
19:40:35.359 -> Clearing input serial buffer
19:40:35.359 -> Sending the following command/query: AT S0
19:40:35.392 -> 	Received char: O
19:40:35.392 -> 	Received char: K
19:40:35.392 -> 	Received char: \r
19:40:35.392 -> 	Received char: \r
19:40:35.392 -> 	Received char: >
19:40:35.392 -> Delimiter found
19:40:35.392 -> All chars received: OK
19:40:35.496 -> Clearing input serial buffer
19:40:35.496 -> Sending the following command/query: AT AL
19:40:35.496 -> 	Received char: O
19:40:35.496 -> 	Received char: K
19:40:35.496 -> 	Received char: \r
19:40:35.496 -> 	Received char: \r
19:40:35.496 -> 	Received char: >
19:40:35.496 -> Delimiter found
19:40:35.496 -> All chars received: OK
19:40:35.602 -> Clearing input serial buffer
19:40:35.602 -> Sending the following command/query: AT TP A0
19:40:35.602 -> 	Received char: O
19:40:35.602 -> 	Received char: K
19:40:35.602 -> 	Received char: \r
19:40:35.602 -> 	Received char: \r
19:40:35.602 -> 	Received char: >
19:40:35.602 -> Delimiter found
19:40:35.602 -> All chars received: OK
19:40:35.602 -> Attempting to connect to ELM327...
19:40:36.954 -> Connected to ELM327
19:40:38.954 -> PROTOCOL: AUTO
19:40:39.970 -> 	ERROR: 7
19:40:41.066 -> 	ERROR: 7
19:40:41.171 -> Volatage: 0.00 v
19:40:42.169 -> 	ERROR: 7
19:40:43.291 -> 	ERROR: 7
19:40:43.364 -> Volatage: 0.00 v
19:40:44.379 -> 	ERROR: 7
19:40:45.495 -> 	ERROR: 7
19:40:46.586 -> 	ERROR: 7
19:40:47.672 -> 	ERROR: 7

PowerBroker2, 2021-04-25:
```C++
myELM327.begin(ELM_PORT, protocol);
```

^ That is not a valid use of the library's `begin()` function. Please read the documentation:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.cpp#L6-L26

PowerBroker2, 2021-04-25:
This line is also incorrect:

```C++
char protocol = ISO_15765_11_BIT_500_KBAUD;
```

MGC1STOKI, 2021-04-25:
update - changing `myELM327.begin(ELM_PORT, protocol);` to `myELM327.begin(ELM_PORT, true, 1000, protocol);` did produce an AT TP A6 but with the same result

MGC1STOKI, 2021-04-25:
changing `char protocol = ISO_15765_11_BIT_500_KBAUD;` to `char protocol = "ISO_15765_11_BIT_500_KBAUD"` produces this respose:

> 19:58:31.628 -> Clearing input serial buffer
19:58:31.628 -> Sending the following command/query: AT TP A⸮
19:58:31.628 -> 	Received char: ?
19:58:31.628 -> 	Received char: \r
19:58:31.628 -> 	Received char: \r
19:58:31.628 -> 	Received char: >
19:58:31.628 -> Delimiter found
19:58:31.628 -> All chars received: 
19:58:31.628 -> Setting protocol via AT TP A%c did not work - trying via AT SP %c
19:58:31.628 -> Clearing input serial buffer
19:58:31.628 -> Sending the following command/query: AT SP ⸮
19:58:31.662 -> 	Received char: ?
19:58:31.662 -> 	Received char: \r
19:58:31.662 -> 	Received char: \r
19:58:31.662 -> 	Received char: >
19:58:31.662 -> Delimiter found
19:58:31.662 -> All chars received: 
19:58:31.662 -> Setting protocol via AT SP %c did not work
19:58:31.662 -> Attempting to connect to ELM327...
19:58:32.990 -> Connected to ELM327

PowerBroker2, 2021-04-25:
This is worse:

```C++
char protocol = "ISO_15765_11_BIT_500_KBAUD"
```

Try this instead:

```C++
myELM327.begin(ELM_PORT, true, 1000, ISO_15765_11_BIT_500_KBAUD);
```

MGC1STOKI, 2021-04-25:
`myELM327.begin(ELM_PORT, true, 1000, ISO_15765_11_BIT_500_KBAUD);` produces the same response