PowerBroker2/ELMduino #76: ESP32 Wifi: getting weird responses

hrtkpf, 2021-04-07:
Hello,

I was testing ELMduino with a OBD2 Wifi adapter and an ESP32. I read #52 and I know there may be problems with this configuration. I'm trying to read the coolant temperature of a BMW from 2014. Sometimes, I get correct numbers, but very often I just get weird responses.
After some debugging, I found out that these "weird responses" might actually contain the correct value but some weird symbols as well, which could be misinterpreted by my code.

I just tested with two different sketches, one by me and the other one being your "ESP32_test.ino", adapted to work with Wifi.
I will only post my sketch and the output for now because I don't want the initial issue to be too long, but I can also post the debug sketch and its output later on if desired.
I also added some comments starting with "NOTE:". Unrelated parts of the sketch, like printing to a display, are omitted.

My sketch:
```
#include <WiFi.h>
#include <ELMduino.h>

// ELM dongle IP address
IPAddress ELMserver(192, 168, 0, 10);
WiFiClient client;
ELM327 elm;

int lastUpdatedCoolant, updateAgeCoolant = 0;
int coolantTemp = 0;

void setup(){
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.begin("V-LINK");

  Serial.println("Connecting to WiFi...");

  while (WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    delay(500);
  }

  // could improve wireless connectivity with ESP (maybe)
  client.setNoDelay(true); // NOTE: due to the error in the output, I don't know if this has any influence at all

  if(client.connect(ELMserver, 35000)){
    Serial.println("ELM connected");
  }else{
    Serial.println("ELM connection failed");
    delay(5000);
    ESP.restart();
  }

  elm.begin(client, true);
}

void loop(){
  // Coolant
  float readCoolantTemp = elm.engineCoolantTemp();

  Serial.println(readCoolantTemp);

  if(elm.status == ELM_SUCCESS && readCoolantTemp < 200 && readCoolantTemp >= 0){
    coolantTemp = readCoolantTemp;
    lastUpdatedCoolant = millis();
  }else{
    // NOTE: this fails because the value is too high, but elm.status is actually ELM_SUCCESS
    Serial.println("Could not read Coolant");
    elm.printError();
  }

  if(elm.status == ELM_TIMEOUT){
    ESP.restart();
  }

  updateAgeCoolant = (millis() - lastUpdatedCoolant) / 1000;

[... 294 lines cut ...]
Received char: >
Delimiter found
All chars received: 410528�
Expected response header: 4105
Single response detected
64-bit response: responseByte_0: 200
responseByte_1: 2
responseByte_2: 0
responseByte_3: 0
responseByte_4: 0
responseByte_5: 0
responseByte_6: 0
responseByte_7: 0
672.00
Could not read Coolant
Received: 410528�
ELM_SUCCESS
0 C, 44s
```

Sometimes, I get the correct number (without the weird symbols) which is then correctly interpreted by the code, but while logging this output, that didn't work.
Thanks to [this page](https://www.obdsol.com/knowledgebase/obd-software-development/reading-real-time-data/), I was able to manually decode the response values (after removing weird symbols) and they correctly decode to 0 °C in this case.

Do you think those weird symbols originate from hardware issues like described in #52? Like I said, the raw numbers seem to be correct.
Also, thank you for maintaining this very useful library! :)

PowerBroker2, 2021-04-07:
I don't know why you're getting the `�` char, but I'll add a filter in the parsing function to throw out any non-alphanumeric ascii chars. This should fix your issue

> Also, thank you for maintaining this very useful library! :)

Happy to help and really appreciate people find it useful!

PowerBroker2, 2021-04-07:
Try testing the main branch and if it works, I'll tag out a new version. If not, please post the latest debug printout

hrtkpf, 2021-04-07:
Thank you for your fast response! I will test the code tomorrow again and report back ASAP. 

But doesn’t it need to be `!isalnum(recChar)` because we want to continue on non-alphanumerical chars? Or am I overseeing something?

hrtkpf, 2021-04-08:
So I just tested the updated code and it works flawlessly now! Apparently no hardware issues, just some timeouts here and there, but that's probably normal.

Thank you!

PowerBroker2, 2021-04-08:
Fixed in [release 2.5.5](https://github.com/PowerBroker2/ELMduino/releases/tag/2.5.5)

closed 2021-04-08 by PowerBroker2