PowerBroker2/ELMduino #78: How do I select a different ODB protocol

Maverick1182, 2021-04-24:
Hello, this library looks very useful but whilst you've provided the list of different ODB protocols, I'm now sure what the syntax is to use a different one from standard to ensure the communication protocol is correct for my car (Mazda)

PowerBroker2, 2021-04-24:
You can specify an OBD protocol in the `begin()` function like this:

```C++
myELM327.begin(Serial1, true, 1000, SAE_J1850_PWM_41_KBAUD);
```

`SAE_J1850_PWM_41_KBAUD` is just an example, but you can replace it with any of the following:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.h#L7-L22

Maverick1182, 2021-04-25:
Thanks, will try this out, is the default set to AUTOMATIC? In which case it will somehow find the right protocol anyway? 

Also, I assume that that "Serial1, true, 1000" part in your example are relating to setting the baud rate? and not required for the setting of the protocol?

PowerBroker2, 2021-04-25:
> Thanks, will try this out, is the default set to AUTOMATIC? 

Yup

> In which case it will somehow find the right protocol anyway?

Automatic, I assume

> Also, I assume that that "Serial1, true, 1000" part in your example are relating to setting the baud rate?

No, please read the documentation:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.cpp#L6-L26

Also, `ELM_PORT` is a UART port (like `Serial` or `Serial1`) and `myELM327` is an instance of the main library's class.

closed 2021-04-25 by PowerBroker2