PowerBroker2/ELMduino #80: response problem

mr-brune, 2021-05-08:
I tried the [ESP8266 WiFi.ino](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP8266_WiFi/ESP8266_WiFi.ino) but i did get a `elm timeout` or a response not decected so i tried the [ESP8266 test](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP8266_test/ESP8266_test.ino) and if i send `010C` i didnt get a response but with `0010C` i got the right response. In a simulator it works flawlessy but in my car i have this problem using a wifi odb. I tried changing protocol but i got the same result.

PowerBroker2, 2021-05-08:
Did you run the code in debug mode where you get serial prints on what ELMduino is doing? Run your code in debug mode and post the output. This will give me a good idea as to what is going wrong.

Also, please post your whole code in tags.

Lastly, see [this issue](https://github.com/PowerBroker2/ELMduino/issues/52)

PowerBroker2, 2021-05-22:
No response, assuming you got it fixed on your end

closed 2021-05-22 by PowerBroker2