PowerBroker2/ELMduino #81: Throttle position calculation

victorChB, 2021-05-09:
I was testing the code and the throttle values did not make sense to me. I looked at the source code and the calculation you do is 255/100 * response. Shouldn't it be 100/255?  The values from my car dont go all the way to 100 % but from what I read that depends from manufacturer. Modifying the calculation to 100/255 = 0.39 I get more reasonable values.

Regards

PowerBroker2, 2021-05-09:
Here's the current function in question:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.cpp#L907-L928

As I have it now, the returned value is calculated as:

value / 2.55 = value / (255 / 100) = (value * 100) / 255

Which matches [the PID spec](https://en.wikipedia.org/wiki/OBD-II_PIDs). Am I missing something here?

victorChB, 2021-05-10:
It is correct sorry for my mistake. The origin of this issue I posted is because I am getting some really strange values. The output I get from the throttle position in debug mode is the following:

```
Throttle:Service: 1
PID: 17
Normal length query detected
Query string: 0111
Clearing input serial buffer
Sending the following command/query: 0111
	Received char: 4
	Received char: 1
	Received char: 1
	Received char: 1
	Received char: 1
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 411111000000
Expected response header: 4111
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 17
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
111848104.00
```

The last printed line is done by calling `throttle() `and printing the result it returns directly with `Serial.println()`. After checking the code I am not sure that the received hex string from the ELM327 is being parsed correctly. The response is for this case `411111000000`. If I understand correctly the OBD II payload is after the first two bytes for this case i.e. `11000000`. This should be 0x11 or 17 decimal. And instead, the software is taking as response 0x00000000110000 which corresponds to the calculation of `111848104.00` at the last line (response/2.55).

Is my interpretation correct or is the throttle position response correctly parsed and calculated? 

Here you will find further info from my Toolchain options:
ESP32 Board Version 1.0.6
Arduino Version 1.8.9

PowerBroker2, 2021-05-10:
Possible workaround **at the sketch level**: You could replace all `myELM327.throttle()` calls with `newThrottle()` calls using this function in your sketch:

```C++
float newThrottle()
{
  if (myELM327.queryPID(SERVICE_01, THROTTLE_POSITION)) 
    return ((myELM327.findResponse() << 32) / 2.55); 
}
```

victorChB, 2021-05-10:
So after a closer look at AndroOBD, I saw that the app parses the raw value from the payload by just using the length it is expecting. For the throttle, it's 1 byte so it discards any other bytes that are in the payload. For the log of the first post on this issue its `11000000`. So instead of parsing `11000000` as 0x11000000, it just uses the first byte 0x11. I suspect that is the same thing with the RPM.

The info I found from AndroOBD :
It has a CSV with all the metadata from each Service and PID with the expected bytes, offset, bitmask,etc. The throttle position entry is:

Byte length = 1
Bit length = 8
Bit Mask = 0xFF

[CSV Entry](https://github.com/fr3ts0n/AndrOBD/blob/e56623f48464a899d08c2e4ba04ba3fd9ac06975/library/src/main/java/com/fr3ts0n/ecu/prot/obd/res/pids.csv#L32)

The implementation of using the CSV info to parse the payload is  [here](https://github.com/fr3ts0n/AndrOBD/blob/e56623f48464a899d08c2e4ba04ba3fd9ac06975/library/src/main/java/com/fr3ts0n/ecu/EcuDataItem.java#L164-L171)

So as you said my car (maybe the ELM327?) is appending trailing zeroes but the AndroOBD app ignores them and parses them correctly as well as the RPM. 

Now that I remember I also tried printing the time in seconds the engine has started with your API but it gave me big numbers as well. 

I will try to implement the CSV implementation to parse the raw data and check if all the PIDS that my car supports can be printed out correctly on the weekend. 

To be continued...

PowerBroker2, 2021-05-10:
> I saw that the app parses the raw value from the payload by just using the length it is expecting

That's probably a better way to do it, but the reason why I use every byte the ELM327/car responds with is to make custom PIDs processing easier and I want to retain that functionality. I'm still under the assumption it's a bug in your ELM327's software and I've never had anyone have this problem till now.

If anyone else has this problem and is reading, please let me know in a reply to this issue with your ELM327 manufacturer and software version info.

PowerBroker2, 2021-05-11:
Been thinking and this could be a decent compromise. Replace `throttle()` in the cpp to this and see if it works:

```C++
 float ELM327::throttle() 
 { 
 	if (queryPID(SERVICE_01, THROTTLE_POSITION)) 
 		return ((findResponse() >> (((recBytes / 2) - 2) * 8)) / 2.55); 
  
 	return ELM_GENERAL_ERROR; 
 } 
```

If it works, I'm considering using a similar edit to the one above to the rest of the functions in case anyone else is using an ELM327 like yours.

victorChB, 2021-05-16:
I did not clearly understand the calculation you proposed for the response.

recBytes/2 = Number of bytes received due to 2 bytes per hex representation in a String
-2   = Header size?
*8 = 8 bits in a byte ?

So the formula would be 

(Number of bytes received  - Header size)*8 = Total bits of the data payload

So for the above example it would be

Received String= 411111000000
recBytes= 12
Plugging the numbers in the equation would be
((12/2)-2)*8 =  32

The Response as mentioned from the library is  0x11000000. Bit shifting 32 bits would get you zero.

If there is something that I misinterpreted let me know but I am not sure that would work.

PowerBroker2, 2021-05-16:
I'm very much the absent minded professor type and make dumb mistakes like this all the time. Here's what I actually meant:

```C++
 float ELM327::throttle() 
 { 
 	if (queryPID(SERVICE_01, THROTTLE_POSITION)) 
 		return ((findResponse() >> ((payBytes - 2) * 4)) / 2.55); 
  
 	return ELM_GENERAL_ERROR; 
 } 
```

Where:

- `payBytes - 2` is the difference between the number of hex chars received in the payload and the number of chars expected to be in the payload of the response
- The difference in bytes from the note above is divided by 2 since there are 2 hex chars per byte, HOWEVER, we must also multiply by 8 since there are 8 bits in a byte, hence the `* 4`
- After bit shifting the proper amount, the scaling factor of `100 / 255` is applied

Test this. If it works, I'll clean it up, apply it to the rest of the functions and tag out a new version for you.

PowerBroker2, 2021-05-22:
Once I tag out the next version, this issue will be fixed

closed 2021-05-22 by PowerBroker2