PowerBroker2/ELMduino #83: Getting either zero values or extremely high wrong values from every request

legocrack1, 2021-05-21:
Hi, Im looking for some help about a little project in which im working on.

Im trying to get some data from my OBD2 ELM327 adapter, then the data will be uploaded to one server through HTTP request with a SIM800L gprs module, I can confirm that everything is working fine except the data that im receiving from the OBD, every time I do a request I get the message "Response not detected", this cause extremely high wrong values or 0 values some times, the ELM327 is working fine, i can get the right data with the Torque App, also i was able to try the code with an ELM327 simulator on android, the code works fine with the simulator, for some reason i only have this problem when I use the real adapter.

[image]

```
void loop() {

  if (myELM327.status == ELM_SUCCESS) {

    //rpm = (uint32_t)tempRPM;
    rpm = myELM327.rpm();
    consumoCombustible = myELM327.fuelRate();
    millaje = myELM327.distTravelWithMIL();
    torque = myELM327.torque();
    temperaturaTomaDeAire = myELM327.intakeAirTemp();
    temperaturaRefrigerante = myELM327.engineCoolantTemp();
    temperaturaAmbiente = myELM327.ambientAirTemp();
    temperaturaAceite = myELM327.oilTemp();
    //potencia=myELM327.rpm(); //La potencia no se puede obtener

    DEBUG_PORT.println("RPM: " + (String)rpm);
    DEBUG_PORT.println("Consumo de combustible: " + (String)consumoCombustible);
    DEBUG_PORT.println("Millaje: " + (String)millaje);
    DEBUG_PORT.println("Torque: " + (String)torque);
    DEBUG_PORT.println("TemperaturaTomaDeAire: " + (String)temperaturaTomaDeAire);
    DEBUG_PORT.println("TemperaturaRefrigerante: " + (String)temperaturaRefrigerante);
    DEBUG_PORT.println("TemperaturaAmbiente: " + (String)temperaturaAmbiente);
    DEBUG_PORT.println("TemperaturaAceite: " + (String)temperaturaAceite);
    //DEBUG_PORT.println("Potencia: "+(String)potencia);

    //Orden de las variables: RPM,Consumo de combustible,Millaje,Torque,Temperatura de la toma de aire,Temperatura del refrigerante,Temperatura ambiente,Temperatura del aceite,potencia

    //Envio con todos los datos para cuando el servidor este preparado
    //gsm_http_post("param="+(String)rpm+" "+(String)consumoCombustible+" "+(String)millaje+" "+(String)torque+" "+(String)temperaturaTomaDeAire+" "+(String)temperaturaRefrigerante+" "+(String)temperaturaAmbiente+" "+(String)temperaturaAceite+" "+(String)potencia);

    //Envio de solo dos datos para pruebas en el servidor de testing
    gsm_http_post("param=" + (String)rpm + " " + (String)consumoCombustible);

    delay(tiempoDeActualizacion);

  } else {
    DEBUG_PORT.println("El OBD2 tuvo un problema recibiendo la informacion del coche, reintentando....");
    delay(1000);
  }

}
```

PowerBroker2, 2021-05-21:
Please copy and paste printouts directly into the issue **instead of taking photos of your screen.**

1. Run [an example sketch](https://github.com/PowerBroker2/ELMduino/tree/master/examples) and see if you can get RPM to work
2. Please run your code in debug mode and post all the printouts you get
3. Your sketch is written incorrectly, you need to follow the required format for each individual value (see below):

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino#L45-L53

legocrack1, 2021-05-21:
Hi, thank you for the quick answer, I used your ESP32 Bluetooth example and still not working, I got this serial output in a loop:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ELM327v15
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char:  
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
[... 358 lines cut ...]
	Received char: N
	Received char: O
	Received char:  
	Received char: D
	Received char: A
	Received char: T
	Received char: A
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: NODATA
ELM responded with errror "NO DATA"
Expected response header: 415C
Response not detected
RPM: 0
Consumo de combustible: 0
Millaje: 0
Torque: 4294967171
TemperaturaTomaDeAire: 4294967256
TemperaturaRefrigerante: 4294967256
TemperaturaAmbiente: 4294967256
TemperaturaAceite: 4294967256

```

PowerBroker2, 2021-05-21:
Thanks! Much better.

It seems you're having [the same issue as this person](https://github.com/PowerBroker2/ELMduino/issues/81), where you're getting data, just that your ELM327 is adding extra chars in the response and is throwing off my parsing algorithm.

I do have a possible fix I suggested [here](https://github.com/PowerBroker2/ELMduino/issues/81#issuecomment-841869999), but OP hasn't responded on whether or not it worked. In that case, could you do the following:

1. Open ELMduino.cpp in an IDE or text editor of your choice
2. Replace the contents of the `rpm()` function in that cpp file with this:

```C++
 	if (queryPID(SERVICE_01, ENGINE_RPM)) 
 		return ((findResponse() >> ((payBytes - 2) * 4)) / 4.0); 
  
 	return ELM_GENERAL_ERROR; 
```

3. Test the example sketch and see if it works
4. Post the debug prints and post here

legocrack1, 2021-05-22:
Hi, I tried do modify the library like you said but looks like "payBytes" is a local variable of other scope or something like that,  I have a compile error:

```
C:\Users\Misael\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:767:33: error: 'payBytes' was not declared in this scope

    return ((findResponse() >> ((payBytes - 2) * 4)) / 4.0); 
```

PowerBroker2, 2021-05-22:
Dang, you're right - it's a local variable. Here is where it's defined:

https://github.com/PowerBroker2/ELMduino/blob/ad580d996257e88f18d9f14d4e47e3197bfc6807/src/ELMduino.cpp#L2507

Could you make it a public class global variable and try again?

legocrack1, 2021-05-22:
Ok, I tried but im still getting weird values

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ELM327v15
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char:  
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: F
	Received char: C
	Received char: E
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0FCE0000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 206
	responseByte_3: 15
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 66289664
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: F
	Received char: B
	Received char: E
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0FBE0000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 190
	responseByte_3: 15
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 66027520
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: F
	Received char: A
	Received char: D
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0FAD0000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 173
	responseByte_3: 15
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 65748992
```

PowerBroker2, 2021-05-22:
Are you sure you replaced the right function with the correct contents? Can you put some extra debug statements inside to see what's going on?

My real question is why the f- is your ELM327 is adding erroneous trailing zeros to each response...

legocrack1, 2021-05-22:
Haha, i was about to edit my answer, you are right, I did not edit the rpm function, now Im getting 0 values , heres the output: 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C00000000
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ELM327v15
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char:  
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C00000000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 0
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C00000000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 0
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C00000000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 0

```

legocrack1, 2021-05-22:
Haha, I was about to edit my answer, you are right, I did not edit the rpm function with the new code. Thank you so much, now everything looks fine, the RPM value is right, looks like this is the correction we been looking for

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 3
	Received char: F
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D3F0000
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ELM327v15
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char:  
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D410000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 65
	responseByte_3: 13
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 3
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 4
	Received char: 8
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D480000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 72
	responseByte_3: 13
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 3
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 5
	Received char: F
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D5F0000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 95
	responseByte_3: 13
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 3

```

Now only have to correct the function of the other values, if you can deploy a new version of the library it will be very usefull

PowerBroker2, 2021-05-22:
Wait, you sure an RPM value of 3 is right or am I reading your debug statements wrong?

> if you can deploy a new version of the library it will be very usefull

I will once I verify beyond all doubt that this method works

legocrack1, 2021-05-22:
Oh excuse me, you're right, I rush, initially I didn't think about it well and the value of three seemed to me a correct value since the car is not moving but you're right, 3 is too low a value

PowerBroker2, 2021-05-22:
I had a small mistake, try this:

```C++
 	if (queryPID(SERVICE_01, ENGINE_RPM)) 
 		return ((findResponse() >> ((payBytes - 4) * 4)) / 4.0); 
  
 	return ELM_GENERAL_ERROR; 
```

legocrack1, 2021-05-22:
It looks like everything is working now, Im reading consistent values and most important, the RPM corresponds with the RPM of the Torque App

```
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 7
	Received char: B
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D7B
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ELM327v15
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char:  
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 7
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D700000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 112
	responseByte_3: 13
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 860
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: D
	Received char: 6
	Received char: 3
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410C0D630000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 99
	responseByte_3: 13
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 856
```
I really thank you, it looks like finally this is the solution

PowerBroker2, 2021-05-22:
Ok, I'll update the lib and tag out a new version today

closed 2021-05-22 by PowerBroker2

PowerBroker2, 2021-05-22:
Actually, before I tag out the next version, can you check that the master branch works properly? Thanks!

legocrack1, 2021-05-22:
Excuse me, I don't have much knowledge with github or branches, can you explain in more detail what I should do?

legocrack1, 2021-05-22:
Hi, I tried the the master branch with your examples and the RPM works fine, also i tried with my code and almost everything works fine, still some problems with some values like oil temperature, torque, traveled distance and fuel consumption, the functions that present this problems are: 

oilTemp()
torque()
distTravelWithMIL()
fuelRate()

```
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ELM327v15
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char:  
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A0
[... 227 lines cut ...]
	Received char: N
	Received char: O
	Received char:  
	Received char: D
	Received char: A
	Received char: T
	Received char: A
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: NODATA
ELM responded with errror "NO DATA"
Expected response header: 415C
Response not detected
RPM: 1006
Consumo de combustible: 0
Millaje: 0
Torque: 4294967171
TemperaturaTomaDeAire: 16
TemperaturaRefrigerante: 27
TemperaturaAmbiente: 16
TemperaturaAceite: 4294967256

```

PowerBroker2, 2021-05-22:
You need to make sure that if an error occurs when querying a PID, you don't use the response value. That's why it's **super important** you follow the example code. Also, if you get "NO DATA" as a response from the ELM327 while the car is on and running, that particular PID is not supported.

legocrack1, 2021-05-22:
I have reviewed my code again and I can assure you that everything is fine, I do not receive errors during the requests, I have modified my code to follow the rules of your examples, but I must insist on the reading problems that I mentioned previously

legocrack1, 2021-05-22:
I verified each one of my querys and  the ones that I mentioned to you that have problems  are the only ones that are responding with "No Data" and the error "ELM NO DATA", I assume that in this case the car does not have those sensors or the library does not support that specific request, with which in my case there is no problem, I will adapt to the information that I can receive, I do not see any more problems on my part.

PowerBroker2, 2021-05-22:
>  I assume that in this case the car does not have those sensors or the library does not support that specific request

The library supports all PIDs that have corresponding functions (i.e. the ones you're trying to use). The "NO DATA" response is coming directly from the ELM327 and is an issue with either your car or ELM327.

legocrack1, 2021-05-22:
Thank you so much, everything is working now, I Will wait the deploy of the New versión to update the library

brxxxl, 2024-09-25:
Heyy, @PowerBroker2, did you change the PID processing code again? I have a problem and i think it's related to this issue. I can't get any readings from the request functions (it instead returns timeouts instead) and set all to float 0.0. The connection, however, was a success, and elm.begin() works fine.

This is my code:
`
void loop()
{
	if (button1.state == true)
	{
		if (mode < modeCount)
		{
			mode++;
		}
		else
		{
			mode = 0;
		}
		button1.state = false;
	}

	switch (mode)
	{
	case 0:
	{
#ifndef DEBUG_SKIP_CONNECTION
		rpm = myELM327.rpm();
#endif // DEBUG_SKIP_CONNECTION
		printNewToLine("RPM: ", 0, lcd, true);
		lcd.setCursor(5, 0);
		lcd.print(rpm);
		break;
	}
	case 1:
	{
#ifndef DEBUG_SKIP_CONNECTION
		kph = myELM327.kph();
#endif // DEBUG_SKIP_CONNECTION
		printNewToLine("KPH: ", 0, lcd, true);
		lcd.setCursor(5, 0);
		lcd.print(kph);
		break;
	}
	}
	delay(300);
}`

I don't have the serial logs right now so i can't show you what it looks like, but is really similar (if not the same) as OP's.

However, i tried changing the rpm() function, but as payBytes isn't defined anywhere, the code didn't build.

PowerBroker2, 2024-09-25:
Your code isn't doing the polling correctly. Take a look at the examples in the main branch/latest release to see how to do it properly

brxxxl, 2024-09-25:
Sorry, you mean like this?

```c
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

I have already did that. Forgot to mention my code is edited for now, to debug other parts (such as display and button interruptions), so i've removed this stuff (which wasn't working before) to test stuff.

What i mean is that, with this example code, the functions were not working probably (yet my ELM327 sends correct information over bluetooth to my smartphone, so that's not the problem). I guess the ESP32 adds some weird characters?

Anyways, i'll try and get the Serial logs so i can post here to see if it helps. Thanks for the response!

jimwhitelaw, 2024-09-26:
@brxxl In order for us to help you best, it is important that you provide the exact code you're running (not a modified/simplified version) and the serial logs with debugging enabled. That will provide the most information for diagnosis.

brxxxl, 2024-09-26:
Hey, @jimwhitelaw, thanks for the response. My current running code is the following:

```c
#include "BluetoothSerial.h"
#include "BluetoothSerial.cpp"
#include "ELMduino.h"
#include "Wire.h"
#include "LiquidCrystal_I2C.h"
#include <HardwareSerial.h>
#include "facilities.h"
#include "customCharacters.h"

#ifndef DEBUG_SKIP_CONNECTION
// #define DEBUG_SKIP_CONNECTION
#endif // DEBUG_SKIP_CONNECTION

uint8_t mac[6] = {0x1C, 0xA1, 0x35, 0x69, 0x8D, 0xC5};

button button1 = {19, false};

int modeCount = 1; // Number of modes - 1
float rpm = 0;
float kph = 0;

void IRAM_ATTR buttonPressed()
{
	if (button1.mode < modeCount)
	{
		button1.mode++;
	}
	else
	{
		button1.mode = 0;
	}
}

void setButtonInterrupts()
{
	pinMode(button1.pin, INPUT_PULLDOWN);
	attachInterrupt(button1.pin, buttonPressed, FALLING);
}

void setup()
{
	startDevices();
	DEBUG_PORT.println("Device started");
	DEBUG_PORT.println("Attempting to connect to ELM327...");
	printNewToLine("Connecting to", 0, lcd, true);
	printNewToLine("ELM327", 1, lcd, false);
	blinkingDotsAnimation(6, 1, lcd);

	attemptToConnect(mac); // Connect to the OBD scanner

	setButtonInterrupts();

	Serial.println("Connected to ELM327");
	delay(1000);
	Serial.println("Starting loop...");
	Serial.println("\n");
	Serial.println("\n");
	Serial.println("\n");
}

void loop()
{
	float tempRPM = myELM327.rpm();

	if (myELM327.nb_rx_state == ELM_SUCCESS)
	{
		rpm = (uint32_t)tempRPM;
[... 245 lines cut ...]
11:24:40.934 
11:24:40.934 
11:24:40.934 Service: 1
11:24:40.938 PID: 12
11:24:40.940 Normal length query detected
11:24:40.946 Query string: 010C1
11:24:40.949 Clearing input serial buffer
11:24:40.957 Sending the following command/query: 010C1
11:24:40.989 	Received char: S
11:24:40.994 	Received char: E
11:24:40.998 	Received char: A
11:24:41.003 	Received char: R
11:24:41.009 	Received char: C
11:24:41.013 	Received char: H
11:24:41.018 	Received char: I
11:24:41.023 	Received char: N
11:24:41.028 	Received char: G
11:24:41.033 	Received char: .
11:24:41.038 	Received char: .
11:24:41.043 	Received char: .
11:24:41.048 	Received char: \r
11:24:42.955 Timeout detected with overflow of 0ms
11:24:42.964 Received: SEARCHING...
11:24:42.970 ERROR: ELM_TIMEOUT
```

brxxxl, 2024-09-27:
@PowerBroker2 i've got the logs. Do you have any possible reasons it could be timing out? It works fine with my phone and also the ESP connects right and quick, so i really have no idea.

PowerBroker2, 2024-09-28:
I can't tell what's going wrong from that log, I need to see the response of more rpm queries. It's not uncommon to get "searching..." for the first query response. I'm interested to see what subsequent queries respond with

brxxxl, 2024-10-03:
Hey @PowerBroker2 @jimwhitelaw sorry for taking so long to respond, i was on a trip and couldn't get the logs for my new try.

Anyways, now i've done as you suggested, let the rpm queries continue querying, as you can see on this code (literally just like the example loop):

```
void loop()
{
	mainDisplayRoutine.execute(); // This is just a button checking routine, it should not interfere with the ELM parts, right?

	float tempRPM = myELM327.rpm();

	if (myELM327.nb_rx_state == ELM_SUCCESS)
	{
		rpm = (uint32_t)tempRPM;
		Serial.print("RPM: ");
		Serial.println(rpm);
	}
	else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
	{
		myELM327.printError();
	}
}
```

These were the logs i got:

```
Device started
14:50:49.959 Attempting to connect to ELM327...
14:50:55.232 Phase 1 - successful
14:50:55.797 Clearing input serial buffer
14:50:55.803 Sending the following command/query: AT D
14:50:55.815 	Received char: A
14:50:55.819 	Received char: T
14:50:55.824 	Received char: _
14:50:55.829 	Received char: D
14:50:55.834 	Received char: \r
14:50:55.840 	Received char: O
14:50:55.844 	Received char: K
14:50:55.849 	Received char: \r
14:50:55.855 	Received char: \r
14:50:55.860 	Received char: >
14:50:55.865 Delimiter found.
14:50:55.870 All chars received: ATDOK
14:50:55.968 Clearing input serial buffer
14:50:55.974 Sending the following command/query: AT Z
14:50:55.985 	Received char: A
14:50:55.990 	Received char: T
14:50:55.995 	Received char: _
14:50:56.000 	Received char: Z
14:50:56.005 	Received char: \r
14:50:56.010 	Received char: \r
14:50:56.016 	Received char: \r
14:50:56.021 	Received char: E
14:50:56.026 	Received char: L
14:50:56.031 	Received char: M
14:50:56.036 	Received char: 3
14:50:56.041 	Received char: 2
14:50:56.046 	Received char: 7
14:50:56.051 	Received char: _
14:50:56.055 	Received char: v
14:50:56.060 	Received char: 2
14:50:56.065 	Received char: .
14:50:56.070 	Received char: 1
14:50:56.075 	Received char: \r
14:50:56.080 	Received char: \r
14:50:56.086 	Received char: >
14:50:56.091 Delimiter found.
14:50:56.095 All chars received: ATZELM327v2.1
14:50:56.201 Clearing input serial buffer
[... 261 lines cut ...]
14:51:56.665 Normal length query detected
14:51:56.670 Query string: 010C1
14:51:56.675 Clearing input serial buffer
14:51:56.683 Sending the following command/query: 010C1
14:51:58.689 Timeout detected with overflow of 0ms
14:51:58.698 Received: 
14:51:58.701 ERROR: ELM_TIMEOUT
14:51:58.788 Service: 1
14:51:58.792 PID: 12
14:51:58.793 Normal length query detected
14:51:58.800 Query string: 010C1
14:51:58.806 Clearing input serial buffer
14:51:58.815 Sending the following command/query: 010C1
14:52:00.819 Timeout detected with overflow of 0ms
14:52:00.828 Received: 
14:52:00.831 ERROR: ELM_TIMEOUT
14:52:00.919 Service: 1
14:52:00.923 PID: 12
14:52:00.925 Normal length query detected
14:52:00.931 Query string: 010C1
14:52:00.937 Clearing input serial buffer
14:52:00.945 Sending the following command/query: 010C1
```

I stopped it after that since it just keeps repeating the same loop, "UNABLE TO CONNECT", (even if it DID in fact connect to the ELM as the phase 2 logs show). I really have no idea how to solve this.

brxxxl, 2024-10-03:
I tried to search for the protocol online but didn't find anything. However, i was able to connect to Car Scanner app and read stuff with ECU protocol: 5) ISO 14230-4 KWP (fast baud init, 10.4 kbaud), so i tested that with this code in the phase 2:

```c
	if (!myELM327.begin(ELM_PORT, true, 2000, 5))
	{
		Serial.println("Couldn't connect to OBD scanner - Phase 2");
		printNewToLCD("Phase 2 - Failed", lcd);
		delay(1000);

		#ifndef DEBUG_SKIP_CONNECTION
			restartESP();
		#endif // DEBUG_SKIP_CONNECTION
	}
	else
	{
		DEBUG_PORT.println("Phase 2 - successful");
		printNewToLine("Phase 2 - ", 0, lcd, true);
		printNewToLine("Successful", 1, lcd, false);
		delay(500);
		printNewToLine("Connected to", 0, lcd, true);
		printNewToLine("ELM327", 1, lcd, false);
	}
```

Results:

```
16:55:08.958  E  D D   !"j ґ"NE" Zj Z ґ"sZ *}PQ  K*T KIx ED֩* ) "  " )R$  Device started
16:55:11.057 Attempting to connect to ELM327...
16:55:13.361 Phase 1 - successful
16:55:13.930 Clearing input serial buffer
16:55:13.936 Sending the following command/query: AT D
16:55:13.946 	Received char: A
16:55:13.951 	Received char: T
16:55:13.956 	Received char: _
16:55:13.961 	Received char: D
16:55:13.966 	Received char: \r
16:55:13.971 	Received char: O
16:55:13.976 	Received char: K
16:55:13.981 	Received char: \r
16:55:13.986 	Received char: \r
16:55:13.991 	Received char: >
16:55:13.996 Delimiter found.
16:55:14.001 All chars received: ATDOK
16:55:14.101 Clearing input serial buffer
16:55:14.109 Sending the following command/query: AT Z
16:55:14.118 	Received char: A
16:55:14.124 	Received char: T
16:55:14.129 	Received char: _
16:55:14.133 	Received char: Z
16:55:14.138 	Received char: \r
16:55:14.143 	Received char: \r
16:55:14.148 	Received char: \r
16:55:14.153 	Received char: E
16:55:14.158 	Received char: L
16:55:14.163 	Received char: M
16:55:14.168 	Received char: 3
16:55:14.173 	Received char: 2
16:55:14.178 	Received char: 7
16:55:14.183 	Received char: _
16:55:14.188 	Received char: v
16:55:14.193 	Received char: 2
16:55:14.198 	Received char: .
16:55:14.203 	Received char: 1
16:55:14.208 	Received char: \r
16:55:14.213 	Received char: \r
16:55:14.218 	Received char: >
16:55:14.223 Delimiter found.
16:55:14.228 All chars received: ATZELM327v2.1
16:55:14.334 Clearing input serial buffer
[... 351 lines cut ...]
16:55:56.406 Normal length query detected
16:55:56.413 Query string: 010C1
16:55:56.419 Clearing input serial buffer
16:55:56.427 Sending the following command/query: 010C1
16:55:58.432 Timeout detected with overflow of 0ms
16:55:58.441 Received: 
16:55:58.444 ERROR: ELM_TIMEOUT
16:55:58.532 Service: 1
16:55:58.535 PID: 12
16:55:58.537 Normal length query detected
16:55:58.544 Query string: 010C1
16:55:58.550 Clearing input serial buffer
16:55:58.558 Sending the following command/query: 010C1
16:56:00.562 Timeout detected with overflow of 0ms
16:56:00.572 Received: 
16:56:00.575 ERROR: ELM_TIMEOUT
16:56:00.703 Service: 1
16:56:00.705 PID: 12
16:56:00.708 Normal length query detected
16:56:00.715 Query string: 010C1
16:56:00.720 Clearing input serial buffer
16:56:00.728 Sending the following command/query: 010C1
```

I also tried with protocols 3 and 5 but it just repeatedly timeouts