PowerBroker2/ELMduino #84: how can I get information for 0x34

nices0325, 2021-05-26:
hello powerbroker!
Thank for your awesome library.
I'm struggling to get O2 lambda equivalent from 0x34. Will this code work?

float ELM327::o2lamdasensor()
{
	if (queryPID(SERVICE_01, OXYGEN_SENSOR_1_C ))
		return conditionResponse(findResponse(), 2, 2.0/65536);

	return ELM_GENERAL_ERROR;
}

actually I didn't understand of the conditionResponse() function. I think It added recently ...

Waiting for your reply thanks!

PowerBroker2, 2021-05-26:
Yes, I added `conditionResponse()` as a fix for issues #81 and #83, where the user's ELM327's were sending extra trailing zeros and were throwing off my parsing algorithm. The function also allows you to specify a factor and bias.

There's a couple issues with your code:
- The response to that PID is 4 bytes long, not 2
- The PID's response is split into two parts - each 2 bytes long and are processed differently:
   - For the first two bytes: [image]
   - For the second two bytes: [image]
- I would not edit the library directly - use existing library functions to make your own `o2lamdasensor()` **_in your sketch_**

I hope this helps and gets you pointed in the right direction.

nices0325, 2021-05-27:
my goal is to use first two bytes!. so coded like this " return  conditionResponse ( findResponse (), 2 , 2.0 / 65536 ); " Can I get data from first two bytes?

PowerBroker2, 2021-05-27:
> my goal is to use first two bytes!

Ah, I didn't know.

Try this in your sketch (not the library):

```C++
float o2lamdasensor()
{
	if (myELM327.queryPID(SERVICE_01, OXYGEN_SENSOR_1_C ))
		return (myELM327.findResponse() >> (((myELM327.numPayChars / 2) - 2) * 8)) * (2.0 / 65536.0);
}
```

nices0325, 2021-05-28:
> > 내 목표는 처음 2 바이트를 사용하는 것입니다!
> 
> 아, 몰랐어요.
> 
> 스케치 (라이브러리 아님)에서 이것을 시도하십시오.
> 
> ```c++
> float  o2lamdasensor ()
> {
> 	if (myELM327. queryPID (SERVICE_01, OXYGEN_SENSOR_1_C))
> 		 return (myELM327. findResponse () >> (((myELM327. numPayChars / 2 ) -2 ) * 8 )) * ( 2.0 / 65536.0 );
> }
> ```

genius reply Thank you.. one more question plz...
I do need mafRate to calculate instant fuel consumption, but mafRate returns 0 as i query... I'm driving kia k7 2017 which has maf sensor..

PowerBroker2, 2021-05-28:
What do the debug statements say?

PowerBroker2, 2021-06-07:
No response, assuming you got it figured out

closed 2021-06-07 by PowerBroker2