PowerBroker2/ELMduino #85: how to use extended PIDs

nices0325, 2021-06-19:
Hello powerbroker! I have read previous issues related to using extended PIDs. However for me, It's a little bit difficult to understand.. I want to use BMS PIDs using SERVICE 0x22 ---> link https://github.com/iternio/ev-obd-pids/blob/main/hkmc/hkmc-2019-2021.csv

should I set header to response correct data? like this 
 void setup() {
  char command[20] = { '\0' };
  sprintf(command, SET_HEADER, "7E4");
  if (myELM327.sendCommand(command) != ELM_SUCCESS)
  {
    // deal with error
  }
  }
  
  void loop
  {
  if (myELM327.queryPID(34, 257))   // Kona PID = 22 01 01 = 34, 257
              {
                 Serial.print("Payload received: ");
                 for (int i=0; i<myELM327.recBytes; i++) Serial.print(myELM327.payload[i]);
                 Serial.println();
              }
   }
   
 It heard that It should be changed to change payloadLen 200 to parse the data.

As I Set Header to 7E4, Should I set Header to the other one to get data from SERVICE_01 ?? It's so confusing.. cause I'm newbie of programming...
 
 How can I use your findResponse function or conditionResponse from this raw data... Thanks for your library again!

PowerBroker2, 2021-06-19:
I'm not familiar with setting custom headers, but there's info on that on page 39 of the [ELM327 datasheet](https://www.elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf).

Did you try the code you posted? What exactly is your question?

nices0325, 2021-06-21:
> 커스텀 헤더 설정에 익숙하지 않지만 [ELM327 데이터 시트](https://www.elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf) 39 페이지에 정보가 있습니다.
> 
> 게시 한 코드를 사용해 보셨습니까? 질문이 정확히 무엇입니까?

Oh sorry. My point is that as I set header to 7E4 ( " AT SH 7E4 "), Can't I use the other default PIDs ( i.e. the Service_01 that I was using). Should I change header to the other one to use my default PIDs?

nices0325, 2021-06-21:
looking forward to your reply thanks again!

PowerBroker2, 2021-06-21:
That is mentioned in the datasheet I linked. Please read page 39 of the datasheet.

nices0325, 2021-06-22:
> 그것은 내가 연결 한 데이터 시트에 언급되어 있습니다. 데이터 시트 39 페이지를 읽으십시오.

I read that and "AT SH xxx " is valid on one time query.. Am I right? As I use standard PIDs, It doesn't need to set header to standard header .. right? Do I understand right??.. like this below
[image]

closed 2021-09-11 by PowerBroker2