PowerBroker2/ELMduino #86: Dificulty to conect and read  my car

nikolasjunqueira, 2021-06-24:
hey, fine? 
I'm having issue to conect and read my car, Renault Twingo 2001/2002 1.0 16v tha uses marelli iaw 5n ECU. i've tested with another car, Xsara Picasso 2004/2005, both uses ISO 14230-4 (fast init). In the Xsara i can conect and read perfectly but in twingo no, when i can conect with twingo i've got this serial log. a intresting detail is that Twingo only can be read by android phone using the app "EODB facile" maybe my car use a diferent pid? the code that i've use for test is the Elm32 WiFi example from library.
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Connecting to WiFi-OBDII
.....
Connected to Wifi
IP address: 
192.168.0.1
connected
Clearing input serial buffer
Sending the following command/query: AT D
All chars received: 
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT E0
All chars received: 
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: A
	Received char: T
	Received char:  
	Received char: Z
	Received char: \r
	Received char: \r
	Received char: \r
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char:  
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATZELM327v15
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: A
	Received char: T
	Received char:  
	Received char: A
	Received char: L
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATALOK
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: A
	Received char: T
	Received char:  
	Received char: T
	Received char: P
	Received char:  
	Received char: A
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATTPA0OK
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: C
	Received char: \r
	Received char: S
	Received char: E
	Received char: A
	Received char: R
	Received char: C
	Received char: H
	Received char: I
	Received char: N
	Received char: G
	Received char: .
	Received char: .
	Received char: .
	Received char: \r
All chars received: 010CSEARCHING
Timeout detected with overflow of 0ms
Received: 010CSEARCHING
ERROR: ELM_TIMEOUT
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
	Received char: S
	Received char: T
	Received char: O
	Received char: P
	Received char: P
	Received char: E
	Received char: D
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: STOPPED
ELM responded with errror "STOPPED"
Expected response header: 410C
Response not detected
Received: STOPPED
ERROR: ELM_STOPPED
Service: 1
PID: 12
Normal length query detected
Query string: 010C
Clearing input serial buffer
Sending the following command/query: 010C
`

PowerBroker2, 2021-06-24:
Try increasing the timeout threshold to something like 1min or so and see if that helps

nikolasjunqueira, 2021-06-24:
In the ELMduino.h? i'm gonna try this today. and about de dificulty to conect to wifi in my car?

PowerBroker2, 2021-06-24:
There should be a way to do it via the library's API. Not sure what you mean about the wifi in your car

nikolasjunqueira, 2021-06-24:
I made the change in this part of ELMduino.h
`	bool begin(Stream& stream, const bool& debug = false, const uint16_t& timeout = 60000, const char& protocol = '0', const uint16_t& payloadLen = 40);
	bool initializeELM(const char& protocol='0');
	void flushInputBuff();
	uint64_t findResponse();
	bool queryPID(uint8_t service, uint16_t pid);
	bool queryPID(char queryStr[]);
	int8_t sendCommand(const char *cmd);
	bool timeout();`
 i hope that works, abou the wifi i've noted that esp have difficulty to connect to wifi of ELM327 but only in twingo.

PowerBroker2, 2021-06-24:
You don't need to change the header file, you can simply pass the desired timeout in ms to the `begin()` function

nikolasjunqueira, 2021-06-24:
ok, now i understand, thanks and sorry for the noob question. i'm gona try this and test a another ELM327, a bluetooh one. i give a feedback soon

nikolasjunqueira, 2021-06-24:
I've made the modification in the code, but nothing change.
`#include <WiFi.h>
#include "ELMduino.h"

const char* ssid = "WiFi-OBDII";
const char* password = "";

//IP Adress of your ELM327 Dongle
IPAddress server(192, 168, 0, 10);
WiFiClient client;
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(115200);

  // Connecting to ELM327 WiFi
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_AP);
 // WiFi.begin(ssid);
  WiFi.begin(ssid, password); //Use this line if your ELM327 has a password protected WiFi

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to Wifi");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (client.connect(server, 35000))
    Serial.println("connected");
  else
  {
    Serial.println("connection failed");
    while(1);
  }

  myELM327.begin(client, true, 60000);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
    myELM327.printError();
}`
and the serial monitor return this:
`Connected to Wifi
IP address: 
192.168.0.1
connected
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: \r
	Received char: O
	Received char: K
[... 758 lines cut ...]
	Received char: 5
	Received char: 8
	Received char: 7
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: 410587
Expected response header: 4105
Single response detected
64-bit response: 
	responseByte_0: 135
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
95.00
95 C, 0s
ok!ok!Service: 1
PID: 5
`
any idea how i can make this work propretaly in my car?

PowerBroker2, 2021-06-24:
Since your cars are pre-2008, I can't guarantee this lib will work with them since OBD-2 became the enforced standard on pretty much all cars in 2008.

It seems, though, that the lib is interfacing with your Xsara Picasso just fine. For the other car, it's returning the wrong response header for some reason. For instance, when you send the PID query `0105`, the proper OBD-2 response would be `4105XX` where `4105` is the response header and `XX` is the data requested. This is the same for all standard PIDs. On your car, however, it is responding `7F0111`. This is totally different than what the lib was written to expect and I'm not even sure what the pattern is with this particular car. Again, I think this is simply because the car pre-dates 2008 and doesn't fully conform to the OBD-2 protocol.

nikolasjunqueira, 2021-06-24:
thanks for your attention! this car is so problematic with many scanners... so maybe this really doesn't work.

closed 2021-06-24 by nikolasjunqueira