PowerBroker2/ELMduino #89: Debug cannot be switched off

sixtybi, 2021-09-01:
I can't turn off serial debugging.
If I execute ELM_PORT.begin() with the second parameter disabled, I can't connect to OBD adapter.
If I set it to true, I can connect, everything works fine, but I get all the debug messages too.
Is there a way to disable debugging?
As a WA I have downgraded the package, but it would be great to have a solution with the latest version.
Thanks!

PowerBroker2, 2021-09-02:
Please post all the contextual details about your project including (but not limited to) the code you're using. Any additional data on error codes or anything else helps, too

sixtybi, 2021-09-05:
Here is my code:

#define SIM800L_IP5306_VERSION_20190610

/*
  OBD_Processor.ino  for SafeString V4.0.0+

  Example of using SafeString to process Car OnBoardData

  by Matthew Ford
  Copyright(c)2020 Forward Computing and Control Pty. Ltd.
  This example code is in the public domain.

  download and install the SafeString library V4.0.0+ from Arduino library manager
  or from www.forward.com.au/pfod/ArduinoProgramming/SafeString/index.html
*/

#include "ELMduino.h"

#include "SafeString.h"
// see https://github.com/PowerBroker2/ELMduino/issues/44#issuecomment-712617781
//Rawdata from OBD:
// some test data
char rawData[] = "7F22127F22127F22127F22127F221203E0:620101FFF7E71:FF8A00000000832:00240ED71713133:141713000010C14:22C12800008B005:00450B0000434B6:000019A80000187:1200200EE70D018:7B0000000003E8";

// where to store the results
struct dataFrames_struct {
  char frames[9][20]; // 9 frames each of 20chars
};

typedef struct dataFrames_struct dataFrames; // create a simple name for this type of data
dataFrames results; // this struct will hold the results

uint8_t address[6]  = {0x00, 0x04, 0x3E, 0x6F, 0x2B, 0x0C};

#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define ELM_PORT   SerialBT

ELM327 myELM327;

void setup() {
  // Open serial communications and wait a few seconds
  Serial.begin(115200);
  for (int i = 10; i > 0; i--) {
    Serial.print(' '); Serial.print(i);
    delay(500);
  }
  Serial.println();
  SafeString::setOutput(Serial);
    ELM_PORT.setPin("0000");         // OBDLink LX
    //  ELM_PORT.setPin("1234");         // Tibi
    ELM_PORT.begin("boost", false);
    delay(1000);
    if (!ELM_PORT.connect(address))   // Device name of iCar Vgate pro BT4.0 OBD adapter
    {
      Serial.println("Couldn't connect to OBD scanner - Phase 1");
      ESP.restart();
    }
    if (!myELM327.begin(ELM_PORT, '6')) { // select protocol '6'
      //   if (!myELM327.begin(ELM_PORT,true,2000)){ // select protocol '6'
      Serial.println("Couldn't connect to OBD scanner - Phase 2");
      delay(1000);
      ESP.restart();
    }
    Serial.println("Connected to ELM327");
    myELM327.PAYLOAD_LEN = 150;
//    myELM327.sendCommand("ATD");
[... 201 lines cut ...]
	Received char: >
Delimiter found
All chars received: ATTPA0OK
Connected to ELM327
Clearing input serial buffer
Sending the following command/query: AT SH 7E4
	Received char: A
	Received char: T
	Received char:  
	Received char: S
	Received char: H
	Received char:  
	Received char: 7
	Received char: E
	Received char: 4
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found
All chars received: ATSH7E4OK

So I can't disable debugging in the newest version (2.6.1)

sixtybi, 2021-09-05:
I have realized, after analyzing the source code, this should be changed not there, I mentioned earlier (it was suggested by someone in a discussion.)
So, to switched off debugging should be changed here:
myELM327.begin(ELM_PORT, true,2000,'6',200);
Second parameter should be set to false to disable debugging.
And the fifth parameter is maximum response length which caused a lot of trouble for me with my new obdlink lx adapter.

sixtybi, 2021-09-05:
I am closing this issue as I have figured it out why it was wrong for me.

closed 2021-09-05 by sixtybi