PowerBroker2/ELMduino #90: Semicolon disappeared in 2.5.5

sixtybi, 2021-09-05:
Since release 2.5.5, the output is different using the same code.
Finally I was able to debug it, I would like to share with you.

Until 2.5.4:
All chars received: 03D0:6101FFFFFFFF1:8526482648A3002:070E671E1B1D1D3:1B1A1D001AC00D4:BF0100008000035:3C0400033B57006:012C4B000122887:00AA9D240901718:0000000003E800

Since 2.5.5:
All chars received: 03D06101FFFFFFFF18526482648A3002190E631E1B1D1D31B1A1D001ABF0B4BF01000081000353C0400033B57006012C4B00012288700AA9DAF09017080000000003E800

Initialization, parameters, code, device are all the same.

I just wanted to open this issue to help others after upgrading the library, to find out why it behaves differently.

PowerBroker2, 2021-09-06:
This is probably due to this line I added to prevent certain issues other people were having:

https://github.com/PowerBroker2/ELMduino/blob/246d994b387da39c331c5926f03095213121bde0/src/ELMduino.cpp#L2428

sixtybi, 2021-09-07:
What will be the expected here?
Shall I rewrite my code, or will you add semicolon as supported character?
I have seen few projects which used your library and most of them relied on semicolon as a separator character.
So if they upgrade your library their app won't work.
So I would suggest to add semicolon as an allowed character.
What do you think?

PowerBroker2, 2021-09-07:
I didn't realize it was that big of a deal, but the fix is super simple. If the fix I made works, I'll tag out a new version. Thanks!

sixtybi, 2021-09-10:
Your fix is working, I have tested it. Thank you!

PowerBroker2, 2021-09-10:
Fixed in 2.6.2

closed 2021-09-10 by PowerBroker2