PowerBroker2/ELMduino #92: RPM overflow

MaxYoshida, 2021-10-06:
Thanks for the amazing library. Looks like a great potential!

I wired arduino to HC-05, which is paired to a good quality ELM327. I tested some parts of the code. Those are coolant temperature, RPM, vehicle speed, throttle, fuel level, engine load.

They all worked very nice except fuel level which always showed 0, but I think it could be an issue in my car itself as fuel level didn't show as well using Freematics OBDII and their library.

My issue is in RPM, it shows "ovf" and some weird random numbers between 54 - 110 while I am idle (about 720 RPM in reality). Increasing RPM also didn't work for me.

Any ideas why this happens?

Thanks in advance

PowerBroker2, 2021-10-06:
Can you run your sketch in debug mode and post the both the printout and the sketch itself? That will give me the best idea as to what is going on.

MaxYoshida, 2021-10-06:
Thank you for your response.

Sure. I attached the response I get while debug is on.

Code is basically same as hardware example provided with the library. I just changed baud rate to 9600. As I am using Arduino mega, I attached HC-05 to Serial3. Anyway I attached the code I used to get this debug. You will notice that I commented functions that works fine.

Other things I tried with the RPM: Using same ELM327 module, I connected it to 2 android apps and I see the RPM correct 100%.
I tried Freematics OBDII reader with the library to obtain RPM, works perfect.
I tried this arduino code: https://www.instructables.com/Arduino-OBD2-ELM327-I2C-LCD-HC05-Bluetooth/ (After modifying PID number and using correct equation from wiki which is ((256 * A +B) / 4)), it works fine as well.

Hope these help out in providing a better idea about the issue.

[res.txt](https://github.com/PowerBroker2/ELMduino/files/7294074/res.txt)
[code.txt](https://github.com/PowerBroker2/ELMduino/files/7294088/code.txt)

PowerBroker2, 2021-10-06:
Release [2.6.3](https://github.com/PowerBroker2/ELMduino/releases/tag/2.6.3) will now let you make that custom rpm function like this:

```C++
float rpm()
{
  if (myELM327.queryPID(SERVICE_01, ENGINE_RPM))
    return myELM327.conditionResponse(myELM327.findResponse(), 1, 256.0 / 4.0);
  
  return ELM_GENERAL_ERROR;
}
```

closed 2021-11-08 by PowerBroker2