PowerBroker2/ELMduino #95: ELM to ESP connection to arduino

braedenv25, 2021-11-24:
Amazing library man! It has been so useful to me and my buddies using it. I was wondering if it is possible for an Arduino to use outputs that the ESP is giving out in the serial monitor? and have you used any code like that in the past?

PowerBroker2, 2021-11-24:
Thank you so much! Yeah, it's totally possible to use that information. I actually designed the library originally to use both RPM and speed data to make a car heads up display. I call it ArduHUD - the code, PCB, and 3D print files are free: [https://github.com/PowerBroker2/ArduHUD](GitHub Link).

Other people have used data from their cars via this lib to do other projects, too.

braedenv25, 2021-11-24:
Thankyou for responding so quick! Yes I have actually used some of your code in my program thus far to connect the ELM and the ESP32. Where I got lost specifically was then figuring out how to have an Arduino Mega interpret the data coming through the serial monitor.

PowerBroker2, 2021-11-24:
You can use this example with the Mega and an HC-05: [https://github.com/PowerBroker2/ELMduino/blob/master/examples/Arduino_hardware_serial_test/Arduino_hardware_serial_test.ino](https://github.com/PowerBroker2/ELMduino/blob/master/examples/Arduino_hardware_serial_test/Arduino_hardware_serial_test.ino)

braedenv25, 2021-11-24:
We lucked out using the HC-05. We have only found success using the ESP32 and have a solid connection between that and the ELM. The HC-05 never got connected properly for us no matter how many times we tried. I was hoping there was a way the ESP could send the RPM i am getting from the ELM to the Arduino Mega

braedenv25, 2021-11-24:
I think if i upload some code to the arduino and the esp32 and hardwire them together i could have the arduino receive the real-time RPM the esp is getting from the elm

PowerBroker2, 2021-11-24:
In ArduHUD, I use [one of my other libraries](https://github.com/PowerBroker2/SerialTransfer) to transfer the RPM and speed data from the ESP32 to the other Arduino.

braedenv25, 2021-11-24:
wow man! that's incredible. i will try to implement that into my project. you seriously are a master at work here

braedenv25, 2021-11-24:
i am just a beginner in coding and i am trying my best to understand and you have been a great help. hopefully i can grast your other library as well

PowerBroker2, 2021-11-24:
Good luck!

closed 2021-11-24 by PowerBroker2