PowerBroker2/ELMduino #97: ESP32 w-ROOM not connect with ELM327 bluetooh.

Pilonetto, 2021-11-24:
Hi, I'm using your example code available [here](https://github.com/PowerBroker2/ELMduino/tree/master/examples/ESP32_Bluetooth_Serial).

Even though I don't have the ELM327 connected and not even any other bluetooth devices nearby, the function "ELM_PORT.connect("OBDII")" returns "true" after a while. Thus leaving the connection loop.

I'm using an ESP32 w-ROOM and ELM327 bluetooh.

`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;
LiquidCrystal_I2C lcd(0x3F, 16, 2);

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif
  lcd.begin();
  lcd.backlight();
  lcd.print("Conectando");
  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ArduHUD", true);

  while (true) {
    if (!ELM_PORT.connect("OBDII"))
    {
      DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
      delay(200);
    } else {
      break;
    }
  }

  lcd.setCursor(0, 0);
  lcd.print("Pareado     ");
  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  lcd.print("Connected to ELM327");
  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    lcd.setCursor(0, 0);
    lcd.print("RPM:               ");
    lcd.setCursor(0, 1);
    lcd.print(rpm);
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
    myELM327.printError();
}`

PowerBroker2, 2021-11-24:
To be clear, does the issue come from this loop?:

```C++
  while (true) {
    if (!ELM_PORT.connect("OBDII"))
    {
      DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
      delay(200);
    } else {
      break;
    }
  }
```

Pilonetto, 2021-11-24:
Sorry about the loop, I ended up leaving it there. To be clearer, even if the ODBII device exists, the connect function has no effect.

closed 2021-11-25 by Pilonetto