PowerBroker2/ELMduino #98: ELM responded with errror "NO DATA".  ESP32 WROOM

Pilonetto, 2021-11-25:
Hi, I'm trying to read my car's RPM (peugeot, if that helps). I was able to establish communication but I always get the ELM response responded with error "NO DATA".

I need guidance if there's a way to test all the protocols until I get an answer.

I will post the log and my code below.

CODE
`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;
LiquidCrystal_I2C lcd(0x3F, 16, 2);

uint8_t address[6]  = {0x00, 0x1d, 0xa5, 0x68, 0x98, 0x8b};

uint32_t rpm = 0;
bool connected;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif
  lcd.begin();
  lcd.backlight();
  lcd.print("Conectando");
  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ArduHUD", true);
  //ELM_PORT.setPin("1234");

  //connected = ELM_PORT.connect(address);
  connected = ELM_PORT.connect("OBDII");
  
  if(connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while(!ELM_PORT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
      lcd.setCursor(0, 0);
      lcd.print("Conectando");
    }
  }

  if (!myELM327.begin(ELM_PORT, true, 20000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  lcd.setCursor(0, 0);
  lcd.print("Connected");
  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  Serial.println(myELM327.status);
  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    lcd.setCursor(0, 0);
    lcd.print("RPM:               ");
    lcd.setCursor(0, 1);
    lcd.print(rpm);
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
    myELM327.printError();
}`

FULL LOG
[testeesp.txt](https://github.com/PowerBroker2/ELMduino/files/7603156/testeesp.txt)

Pilonetto, 2021-11-25:
Yes, the car is running. The vehicle is year 2014.

PowerBroker2, 2021-11-25:
Strange, not sure why your car doesn't want to give the ELM327 RPM data. The library seems to be working fine, however. Perhaps try another ELM327

Pilonetto, 2021-11-25:
Okay, I'll buy another one and run new tests. If you have one that I can recommend that works on as many cars and trucks as possible, I would be grateful. Thanks for your answers. :)

PowerBroker2, 2021-11-25:
It's tough since they're all Chinese clones, so I don't have any in particular to recommend

closed 2021-11-25 by PowerBroker2