PowerBroker2/ELMduino #99: How to get a list of all parameters of my car

binderth, 2021-12-01:
Hi there,

I put in a OBD II ELM327 device and used the Android app "Car Scanner" with my Kia EV6 and got all the information (and much much more!) I need in the App. In the list of "standard" PIDs I don't find the ones I need most:

* SoC of my car's battery
* Battery temperatures
* and a bit more EV-related stuff

How can I find out how to address those information coming out of the ELM327 to request it via ELMduino?

Thanks for your help!

PowerBroker2, 2021-12-01:
You'll have to do that research yourself - I don't know those off the top of my head. Here's the list of PIDs supported by the library currently. Searching car forums might be of some use to you.

https://github.com/PowerBroker2/ELMduino/blob/44084a25e4941ae5efabc8c1f2953239759e8706/src/ELMduino.h#L27-L140

closed 2021-12-01 by PowerBroker2

binderth, 2021-12-02:
Sorry, if it seemed, you should do the work for me! ;)
I'm thinking of a way to read out the available channels/PIDs from the car - if this is possible at all...?

PowerBroker2, 2021-12-02:
If you're using a standard PID, you can use the corresponding function in the library:

https://github.com/PowerBroker2/ELMduino/blob/44084a25e4941ae5efabc8c1f2953239759e8706/src/ELMduino.h#L314-L396

If you're using a non-standard PID, you can make your own function in your sketch:

```C++
float newThing()
{
  if (myELM327.queryPID(SERVICE_01, newPid)) 
    return myELM327.conditionResponse(myELM327.findResponse(), 4, 1, 0); // THESE ARGUMENTS MUST BE ADJUSTED FOR DIFFERENT PIDS!

  return ELM_GENERAL_ERROR;
}
```

Note that `myELM327.conditionResponse()` is defined as:

https://github.com/PowerBroker2/ELMduino/blob/44084a25e4941ae5efabc8c1f2953239759e8706/src/ELMduino.cpp#L356-L359