PowerBroker2/ELMduino #100: Receiving weird characters from my ELM327

satungen, 2021-12-19:
Hello,

Sorry in advance for the perhaps silly question; I am quite new to all of this!

Trying the Arduino Software Serial test example you have provided, I only get gibberish characters in return for the commands. 

> 02:59:03.419 -> Attempting to connect to ELM327...
> 02:59:03.756 -> Clearing input serial buffer
> 02:59:03.756 -> Sending the following command/query: AT D
> 02:59:03.756 -> 	Received char: ⸮
> 02:59:03.756 -> 	Received char: 
> 02:59:05.720 -> All chars received: 
> 02:59:05.720 -> Timeout detected with overflow of 0ms
> 02:59:05.863 -> Clearing input serial buffer
> 02:59:05.863 -> Sending the following command/query: AT Z
> 02:59:05.863 -> 	Received char: 
> 02:59:05.863 -> 	Received char: ⸮
> 02:59:05.863 -> 	Received char: ⸮
> 02:59:05.863 -> 	Received char: ⸮
> 02:59:05.863 -> 	Received char: ⸮
> 02:59:07.826 -> All chars received: 
> 02:59:07.826 -> Timeout detected with overflow of 0ms
> 02:59:07.922 -> Clearing input serial buffer
> 02:59:07.922 -> Sending the following command/query: AT E0
> 02:59:07.970 -> 	Received char: ⸮
> 02:59:07.970 -> 	Received char: 
> 02:59:07.970 -> 	Received char: 
> 02:59:07.970 -> 	Received char: 
> 02:59:09.935 -> All chars received: 
> 02:59:09.935 -> Timeout detected with overflow of 0ms
> 02:59:10.031 -> Clearing input serial buffer
> 02:59:10.031 -> Sending the following command/query: AT S0
> 02:59:10.031 -> 	Received char: ⸮
> 02:59:10.031 -> 	Received char: 
> 02:59:10.031 -> 	Received char: 
> 02:59:10.079 -> 	Received char: 
> 02:59:12.043 -> All chars received: 
> 02:59:12.043 -> Timeout detected with overflow of 0ms
> 02:59:12.139 -> Clearing input serial buffer
> 02:59:12.139 -> Sending the following command/query: AT AL
> 02:59:12.139 -> 	Received char: ⸮
> 02:59:12.139 -> 	Received char: 
> 02:59:12.139 -> 	Received char: 
> 02:59:12.139 -> 	Received char: 

I get the same gibberish when I try to send any of the commands through Serial communication. Do you have any idea what's happening?

Thank you in advance!

PowerBroker2, 2021-12-19:
It sounds like a baud mismatch between the Arduino/bluetooth module/ELM327. Are you using an ESP32 or an HC-05 for the bluetooth? Either way, try different bauds and see if you get better results.

satungen, 2021-12-19:
I am using wires for my serial connection.

You're absolutely right, 38400 baud rate solved my problem. Thanks for the quick response, and thanks for the great library! Happy holidays

satungen, 2021-12-23:
Sorry, again to bother. Probably not an issue with your library per se but I might as well ask since you seem helpful and knowledgable.

Running this code I get characters I can't seem to figure out why they appear. 
`#include <SoftwareSerial.h>
#include "ELMduino.h"
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1351.h>
#include <SPI.h>

SoftwareSerial mySerial(8, 9); // RX, TX
#define ELM_PORT mySerial
#define SCREEN_WIDTH  128
#define SCREEN_HEIGHT 128
#define RST_PIN  3
#define DC_PIN   4
#define CS_PIN   5
#define SCLK_PIN 6
#define MOSI_PIN 7
#define BLACK           0x0000
#define BLUE            0x001F
#define RED             0xF800
#define GREEN           0x07E0
#define CYAN            0x07FF
#define MAGENTA         0xF81F
#define YELLOW          0xFFE0  
#define WHITE           0xFFFF

ELM327 myELM327;
Adafruit_SSD1351 tft = Adafruit_SSD1351(SCREEN_WIDTH, SCREEN_HEIGHT, CS_PIN, DC_PIN, MOSI_PIN, SCLK_PIN, RST_PIN);

uint32_t rpm = 0;
int32_t kph = 0;
int errorcount = 0;

void setup()
{
  tft.begin();
  tft.fillScreen(BLACK);
  tft.setTextColor(WHITE, BLACK);
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  Serial.begin(38400);
  ELM_PORT.begin(38400);

  Serial.println("Attempting to connect to ELM327...");
  tft.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT, true, 5000))
  {
    Serial.println("Couldn't connect to OBD scanner");
    tft.println("Could not connect to OBD scanner");
    while (1);
  }

  Serial.println("Connected to ELM327");
  tft.println("Connected to ELM327");
  delay(500);
  tft.fillScreen(BLACK);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
[... 251 lines cut ...]
13:18:03.524 -> Sending the following command/query: 010C
13:18:03.571 -> 	Received char: Q
13:18:03.571 -> 	Received char: =
13:18:03.571 -> 	Received char: A
13:18:03.571 -> 	Received char: 
13:18:03.571 -> 	Received char: 
13:18:03.571 -> 	Received char: j
13:18:03.571 -> 	Received char: ⸮
13:18:03.756 -> 	Received char: S
13:18:03.756 -> 	Received char: E
13:18:03.756 -> 	Received char: A
13:18:03.756 -> 	Received char: R
13:18:03.805 -> 	Received char: C
13:18:03.805 -> 	Received char: H
13:18:03.805 -> 	Received char: I
13:18:03.805 -> 	Received char: N
13:18:03.805 -> 	Received char: G
13:18:03.805 -> 	Received char: .
13:18:03.805 -> 	Received char: .
13:18:03.805 -> 	Received char: .
13:18:03.805 -> 	Received char: \r
13:18:08.549 -> All chars received: QAjSEARCHING
13:18:08.549 -> Timeout detected with overflow of 0ms
`
Any clues as to why these extra characters are appearing?

PowerBroker2, 2021-12-23:
I'm not sure why these characters are showing up - probably an issue with the ELM327 (it's probably a cheap Chinese clone). The library will automatically ignore non-ascii characters, so it shouldn't be an issue as long as the data is being returned.

closed 2022-01-01 by PowerBroker2