PowerBroker2/ELMduino #101: Retrieving OXYGEN_SENSOR_1_C

lebowitz987, 2021-12-30:
First, thank you for creating this library!  I've been having great fun learning to use it.  I was particularly interested in a prior question about retrieving the lambda value from one (or more) of the oxygen sensors via the OBD reader.  You suggested using the following code:

`//
// Query the O2 sensor for the Air-Fuel Equivalence Ratio
//
float o2lamdasensor()
{
  float tempO2;
  if (myELM327.queryPID(SERVICE_01, OXYGEN_SENSOR_1_C ))
    return (myELM327.findResponse() >> (((myELM327.numPayChars / 2) - 2) * 8)) * (2.0 / 65536.0);
}

I added this code to my sketch and am calling it using the following:

`float tempO2 = o2lamdasensor();
    lcd.print("l: ");             // Print the O2 sensor air fuel ratio
    DEBUG_PORT.print("tempO2: "); DEBUG_PORT.println(tempO2);
    dtostrf(tempO2, 4, 2, lambda_str);
    lcd.print(lambda_str);```

When I execute the code, some of the returned values are in the range of 0.00 - 0.99, however quite a few are clearly overflow conditions which ultimately display a nonsense value (e.g. 8689).  Here are a couple of examples of the debug output from the code:

`15:39:56.357 -> Service: 1
15:39:56.357 -> PID: 52
15:39:56.357 -> Normal length query detected
15:39:56.357 -> Query string: 0134
15:39:56.357 -> Clearing input serial buffer
15:39:56.357 -> Sending the following command/query: 0134
15:39:56.428 -> 	Received char: 4
15:39:56.428 -> 	Received char: 1
15:39:56.428 -> 	Received char: 3
15:39:56.461 -> 	Received char: 4
15:39:56.461 -> 	Received char: 7
15:39:56.461 -> 	Received char: F
15:39:56.461 -> 	Received char: 1
15:39:56.461 -> 	Received char: 7
15:39:56.461 -> 	Received char: 7
15:39:56.461 -> 	Received char: F
15:39:56.461 -> 	Received char: F
15:39:56.461 -> 	Received char: 5
15:39:56.461 -> 	Received char: \r
15:39:56.461 -> 	Received char: \n
15:39:56.532 -> 	Received char: \r
15:39:56.532 -> 	Received char: \n
15:39:56.532 -> 	Received char: >
15:39:56.532 -> Delimiter found
15:39:56.532 -> All chars received: 41347F177FF5
15:39:56.532 -> Expected response header: 4134
15:39:56.532 -> Single response detected
15:39:56.532 -> 64-bit response: 
15:39:56.532 -> 	responseByte_0: 245
15:39:56.532 -> 	responseByte_1: 127
15:39:56.532 -> 	responseByte_2: 23
15:39:56.532 -> 	responseByte_3: 127
15:39:56.532 -> 	responseByte_4: 0
15:39:56.532 -> 	responseByte_5: 0
15:39:56.532 -> 	responseByte_6: 0
15:39:56.532 -> 	responseByte_7: 0
15:39:56.532 -> RPM: 641
15:39:56.566 -> tempO2: 0.99

15:40:00.235 -> Service: 1
15:40:00.235 -> PID: 52
15:40:00.235 -> Normal length query detected
15:40:00.235 -> Query string: 0134
15:40:00.235 -> Clearing input serial buffer
15:40:00.235 -> Sending the following command/query: 0134
15:40:00.335 -> 	Received char: 4
15:40:00.335 -> 	Received char: 1
15:40:00.335 -> 	Received char: 3
15:40:00.335 -> 	Received char: 4
15:40:00.335 -> 	Received char: 8
15:40:00.335 -> 	Received char: 0
15:40:00.335 -> 	Received char: 6
15:40:00.335 -> 	Received char: 7
15:40:00.335 -> 	Received char: 7
15:40:00.335 -> 	Received char: F
15:40:00.335 -> 	Received char: F
15:40:00.335 -> 	Received char: A
15:40:00.335 -> 	Received char: \r
15:40:00.335 -> 	Received char: \n
15:40:00.401 -> 	Received char: \r
15:40:00.401 -> 	Received char: \n
15:40:00.401 -> 	Received char: >
15:40:00.401 -> Delimiter found
15:40:00.401 -> All chars received: 413480677FFA
15:40:00.401 -> Expected response header: 4134
15:40:00.401 -> Single response detected
15:40:00.401 -> 64-bit response: 
15:40:00.401 -> 	responseByte_0: 250
15:40:00.401 -> 	responseByte_1: 127
15:40:00.401 -> 	responseByte_2: 103
15:40:00.435 -> 	responseByte_3: 128
15:40:00.435 -> 	responseByte_4: 255
15:40:00.435 -> 	responseByte_5: 255
15:40:00.435 -> 	responseByte_6: 255
15:40:00.435 -> 	responseByte_7: 255
15:40:00.435 -> RPM: 633
15:40:00.435 -> tempO2: ovf```

Unfortunately, my coding skills are decades old and more than a bit rusty.  Is there anything here that you can see that might help to identify the cause for this issue?

Thanks for your assistance!

PowerBroker2, 2021-12-30:
I haven't looked super close at your code, but note that you can use `conditionResponse()` to do some of the processing in your custom function:

https://github.com/PowerBroker2/ELMduino/blob/44084a25e4941ae5efabc8c1f2953239759e8706/src/ELMduino.cpp#L356-L359

Also, try removing all RPM pieces of code and stick just to O2 polling.

Lastly, do you think the values could be correct but `dtostrf()` is overflowing due to your other arguments?

lebowitz987, 2021-12-30:
The debug output is printing the float value.  I neglected to include the DEBUG_PORT.print() line in my code snippet.
The dtostrf() is formatting the float for use in an lcd.print so that isn't the issue.  I expect that the return value for the float should be from 0.0 - 1.0, and the fuller debug output seems to indicate that this range is pretty consistent, except when the "ovf" is returned.  How would that translate into a scaleFactor and bias for conditionResponse?

And just for good measure.... Happy new year!

lebowitz987, 2021-12-31:
I trimmed my code down to only query PID 52.  In doing so, I discovered what appears to be a case in which the response buffer isn't cleared at some point.  The buffer is returned with:

`09:17:24.623 -> All chars received: SEARCHING413482507FFF
09:17:24.665 -> Expected response header: 4134
09:17:24.665 -> Single response detected
09:17:24.665 -> 64-bit response: 
09:17:24.698 -> 	responseByte_0: 255
09:17:24.698 -> 	responseByte_1: 127
09:17:24.698 -> 	responseByte_2: 80
09:17:24.698 -> 	responseByte_3: 130
09:17:24.698 -> 	responseByte_4: 255
09:17:24.698 -> 	responseByte_5: 255
09:17:24.698 -> 	responseByte_6: 255
09:17:24.698 -> 	responseByte_7: 255
09:17:24.698 -> tempO2: ovf`

Notice the clear text "SEARCHING" within the response.   The float value returned results in an overflow and a panic condition that restarts the controller.  In my case, I'm using an ESP32.

I created 2 functions in which I tried variations of the call to PID 52.  One is a straight call to find findResponse() and the other is one in which I experimented with a call to conditionResponse().  I admit that I don't yet understand exactly what these calls are doing other than shifting the response buffer to the right by a number of bytes.

Here is the code I'm using to test:

`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

static char lambda_str[5];

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

  DEBUG_PORT.begin(115200);   // Setup serial output for debug
  ELM_PORT.begin("ArduHUD", true);  // INITIALIZE the Bluetooth connection

  // Attempt to connect to remote bluetooth source "OBDII"

  lcd.clear();                    // clear the LCD
  lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
  lcd.print("BT Search: OBDII");  // Print a connect attempt message
  DEBUG_PORT.println("BT Search: OBDII");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("BT: \"OBDII\" ");
    lcd.setCursor(0, 1);
    lcd.print("not found");         // Print a connect attempt message
    while (1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("Phase 2: FAIL");     // Print a connect attempt message
    while (1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  lcd.clear();                    // clear the LCD
  lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
  lcd.print("BT CONNECT: OBDII");     // Print a connect attempt message
}

void loop()
{

  float tempO2 = o2lamdasensor();

  if (myELM327.status == ELM_SUCCESS)
  {
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("l: ");             // Print the O2 sensor air fuel ratio
    DEBUG_PORT.print("tempO2: "); DEBUG_PORT.println(tempO2);
    dtostrf(tempO2, 4, 2, lambda_str);
    lcd.print(lambda_str);
  }
  else {
    myELM327.printError();
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("BT not found");
  }
}

//
// Query the O2 sensor for the Air-Fuel Equivalence Ratio
//
float o2lamdasensor()
{
  if (myELM327.queryPID(SERVICE_01, OXYGEN_SENSOR_1_C ))
    return (myELM327.findResponse() >> (((myELM327.numPayChars / 2) - 2) * 8)) * (
[backtrace.pdf](https://github.com/PowerBroker2/ELMduino/files/7796421/backtrace.pdf)
2.0 / 65536.0);
}

float o2lamdasensor2()
{
  if (myELM327.queryPID(SERVICE_01, OXYGEN_SENSOR_1_C ))
    return (myELM327.conditionResponse(myELM327.findResponse() >> (((myELM327.numPayChars / 2) - 2) * 8), 2, 1, 0));
}`

I'm attaching a file "backtrace" which contains the serial output from my last run of this sketch.  Any suggestions you might have would be much appreciated.

[backtrace.pdf](https://github.com/PowerBroker2/ELMduino/files/7796424/backtrace.pdf)
.

PowerBroker2, 2021-12-31:
Interesting... Thank you for posting the full code and backtrace.

As for the core panic, I'm not experienced enough with the ESP32 to help - I would suggest asking around on the [ESP32](https://esp32.com/) or [Arduino](https://forum.arduino.cc/) Forums and see what they say.

Core panic aside, you can still fix the data processing part: After researching the O2 sensor ratio/current PID I realized `conditionResponse()` won't be much of a help. This is why I didn't make a builtin function for it. It seems there are 4 bytes returned by this PID, where 2 bytes are for a ratio calculation and the last two bytes are for a current calculation in mA. To do this, you can use the following code (uncompiled and untested):

```C++
float ratio = (256 * myELM327.responseByte_0 + myELM327.responseByte_1) * (2.0 / 65536.0);
float current = (myELM327.responseByte_2 + (myELM327.responseByte_3 / 256.0)) - 128;
```

lebowitz987, 2021-12-31:
Thanks for the code suggestion.  I'll test those out and let you know how it goes.

I think that the panic is somehow related to the input buffer not being cleared before dropping the result of the PID 52 query into that address space.

lebowitz987, 2021-12-31:
Ok.  I took a stab at putting your line of code to calculate the ratio.  This particular experiment consistently returns a zero value.  I added a couple of debug lines to print responseByte_0 and responseByte_1, cast as float, though I'm not sure that I did that correctly.  Both of those values also appear as zero.

Here is the code:
`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

static char lambda_str[5];

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

  DEBUG_PORT.begin(115200);   // Setup serial output for debug
  ELM_PORT.begin("ArduHUD", true);  // INITIALIZE the Bluetooth connection

  // Attempt to connect to remote bluetooth source "OBDII"

  lcd.clear();                    // clear the LCD
  lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
  lcd.print("BT Search: OBDII");  // Print a connect attempt message
  DEBUG_PORT.println("BT Search: OBDII");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("BT: \"OBDII\" ");
    lcd.setCursor(0, 1);
    lcd.print("not found");         // Print a connect attempt message
    while (1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 2");
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("Phase 2: FAIL");     // Print a connect attempt message
    while (1);
  }

  DEBUG_PORT.println("Connected to OBDII");
  lcd.clear();                    // clear the LCD
  lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
  lcd.print("BT CONNECT: OBDII");     // Print a connect attempt message
}

void loop()
{

  float tempO2 = o2Ratio();
  DEBUG_PORT.print("tempO2: "); DEBUG_PORT.println(tempO2);

  if (myELM327.status == ELM_SUCCESS)
  {
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 1);            // Set the curor at char 0 on line 0
    lcd.print("L: ");             // Print the O2 sensor air fuel ratio
    dtostrf(tempO2, 4, 2, lambda_str);
    lcd.print(lambda_str);
  }
  else {
    myELM327.printError();
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("BT not found");
  }
}

//
// Query the O2 sensor for the Air-Fuel Equivalence Ratio
//
//float ratio = (256 * myELM327.responseByte_0 + myELM327.responseByte_1) * (2.0 / 65536.0);
//float current = (myELM327.responseByte_2 + (myELM327.responseByte_3 / 256.0)) - 128;
float o2Ratio()
{
  float ratio = 0.0;
  if (myELM327.queryPID(SERVICE_01, OXYGEN_SENSOR_1_C )) {
    ratio = (256 * myELM327.responseByte_0 + myELM327.responseByte_1) * (2.0 / 65536.0);
    DEBUG_PORT.print("responseByte_0 = "); DEBUG_PORT.println((float) myELM327.responseByte_0);
    DEBUG_PORT.print("responseByte_1 = "); DEBUG_PORT.println((float) myELM327.responseByte_1);
    DEBUG_PORT.print("ratio = ");
    DEBUG_PORT.println(ratio);
  }
  return ratio;
}`

The debug output is attached.  Thanks for your continued help.

[Ratio Serial Output.pdf](https://github.com/PowerBroker2/ELMduino/files/7796923/Ratio.Serial.Output.pdf)

PowerBroker2, 2021-12-31:
Ohhhh, my bad. I forgot that you're supposed to call `findResponse()` before using the response bytes. Like this:

```C++
myELM327.findResponse()
ratio = (256 * myELM327.responseByte_0 + myELM327.responseByte_1) * (2.0 / 65536.0);
```

lebowitz987, 2021-12-31:
Success!  Now to comprehend the data!  The values are all in the 1.9x range.

Thank you again for your help!

The test code as it stands:
`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

static char lambda_str[5];

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

  DEBUG_PORT.begin(115200);   // Setup serial output for debug
  ELM_PORT.begin("ArduHUD", true);  // INITIALIZE the Bluetooth connection

  // Attempt to connect to remote bluetooth source "OBDII"

  lcd.clear();                    // clear the LCD
  lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
  lcd.print("BT Search: OBDII");  // Print a connect attempt message
  DEBUG_PORT.println("BT Search: OBDII");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("BT: \"OBDII\" ");
    lcd.setCursor(0, 1);
    lcd.print("not found");         // Print a connect attempt message
    while (1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 2");
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("Phase 2: FAIL");     // Print a connect attempt message
    while (1);
  }

  DEBUG_PORT.println("Connected to OBDII");
  lcd.clear();                    // clear the LCD
  lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
  lcd.print("BT CONNECT: OBDII");     // Print a connect attempt message
}

void loop()
{

  float tempO2 = o2Ratio();
  DEBUG_PORT.print("tempO2: "); DEBUG_PORT.println(tempO2);

  if (myELM327.status == ELM_SUCCESS)
  {
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 1);            // Set the curor at char 0 on line 0
    lcd.print("L: ");             // Print the O2 sensor air fuel ratio
    dtostrf(tempO2, 4, 2, lambda_str);
    lcd.print(lambda_str);
  }
  else {
    myELM327.printError();
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("BT not found");
  }
}

//
// Query the O2 sensor for the Air-Fuel Equivalence Ratio
//
//myELM327.findResponse();
//float ratio = (256 * myELM327.responseByte_0 + myELM327.responseByte_1) * (2.0 / 65536.0);
//float current = (myELM327.responseByte_2 + (myELM327.responseByte_3 / 256.0)) - 128;

float o2Ratio()
{
  float ratio = 0.0;
  if (myELM327.queryPID(SERVICE_01, OXYGEN_SENSOR_1_C )) {
    myELM327.findResponse();
    ratio = (256 * myELM327.responseByte_0 + myELM327.responseByte_1) * (2.0 / 65536.0);
    DEBUG_PORT.print("responseByte_0 = "); DEBUG_PORT.println((float) myELM327.responseByte_0);
    DEBUG_PORT.print("responseByte_1 = "); DEBUG_PORT.println((float) myELM327.responseByte_1);
    DEBUG_PORT.print("ratio = ");
    DEBUG_PORT.println(ratio);
[Success.pdf](https://github.com/PowerBroker2/ELMduino/files/7797045/Success.pdf)

  }
  return ratio;
}`

The resulting debug data is attached.
[Success.pdf](https://github.com/PowerBroker2/ELMduino/files/7797046/Success.pdf)

closed 2021-12-31 by lebowitz987

PowerBroker2, 2021-12-31:
Nice!

Also, all values between 0 and 2 are valid for the ratio:

[image]

lebowitz987, 2021-12-31:
I saw that. I just need to now understand what the values really represent.
How closely related to lambda, I wonder?  I’ll dig into that this weekend.