PowerBroker2/ELMduino #104: ELM327 not Connected To ESP 32

Anwar0510, 2022-01-09:
Hello I'm Anwar

I'm using ESP 32 DevKit, Mini OBD-II as shown in this link (https://www.tokopedia.com/tokodibandung/super-mini-elm327-bluetooth-obd2-v2-1-automotive-test-tool-blue?whid =0), For the code I only use example ESP32_bluetoothserial and ESP32_test, Actually I use OBD-II to the DLC socket (Data Link Connector) of a motorcycle that already uses OBD-II. The motorcycle I use is Honda Vario (2021). The Mini OBD-II I'm using works fine if I plug it into the "Torque" app. But when I try to connect to the ESP 32 it can't connect. I see on serial monitor it only says "Couldn't connect to OBD scanner - Phase 1". I tried to change the baudrate from 115200 to 38400 also the same. I've also uncommented //SerialBT.setPin, nothing has changed. Can you provide a solution for my problem?
[image]

PowerBroker2, 2022-01-09:
Error on `Phase 1` means your ELM327's bluetooth device name is not `OBDII` like most are:

https://github.com/PowerBroker2/ELMduino/blob/cc4392ed8cf15b894e684eec68c49dafc6bf1ceb/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino#L27

You need to find what the device name is and replace it in your sketch.

Anwar0510, 2022-01-10:
Hello @PowerBroker2 , When I try OBDII via "Torque app" on my cellphone, it connects via bluetooth named "OBDII" as in the photo below, initially I have to enter PIN "1234", I've also tried uncommenting // SerialBT.setPin("1234"); . But the result is still the same, is it because of the low price OBDII?, Thank you..
[image]

PowerBroker2, 2022-01-10:
Probably because your phone is connecting to the ELM327 first. Try disabling bluetooth on your phone and try to get the ESP32 to connect.

Anwar0510, 2022-01-10:
But when I try to use the ESP32, the Bluetooth on my cellphone is off. Is it possible? But I will try your suggestion

Anwar0510, 2022-01-10:
I've tried unpairing the ELM327 with my cellphone, I even tried turning off my cellphone. I tried example ESP32_Bluetoothserial. Try uncommenting `SerialBT.setPIN("1234")` also i tried without uncommenting it. But the result is same as previous. 

Then I try to turn on my cellphone again, turn on bluetooth and pair it with "OBDII", give pin "1234" and pair it. Then I open the "Torque" app it works fine, the app shows data as usual (rpm, tps, etc). So where do you think I should improve?

Thank You @PowerBroker2

PowerBroker2, 2022-01-10:
Hmm, this is an odd problem. Are you sure it errors on Phase 1 and not Phase 2? Can you post a printout of the serial monitor?

If it really is a Phase 1 error, you might have to find the MAC address of your ELM327 and connect to it using that address instead of the name.

Also, is your ELM327 a standard Bluetooth device or a BLE device?

Anwar0510, 2022-01-10:
Halo @PowerBroker2, yes that's right my monitor serial shows couldn't connect phase 1, I attached my PrtScr here.

for finding MAC address my ELM327 how to do that??

I'm not sure if this is standard Bluetooth or BLE?, do you mean BLE (Bluetooth low energy)?, but I think it's BLE. But in my case, does it matter if it's standard Bluetooth or BLE?
[image]

PowerBroker2, 2022-01-10:
If it's BLE, that makes more sense. This library isn't compatible with BLE (at least I haven't tested it with BLE, nor do I know how to use BLE with the ESP32). I would suggest getting a standard bluetooth ELM327

Anwar0510, 2022-01-11:
You have never tried this library using BLE, for example, I can use BLE built in ESP32, and try using this library, what do you think the percentage of success is?

[https://randomnerdtutorials.com/esp32-bluetooth-low-energy-ble-arduino-ide/](ESP32BLE)

PowerBroker2, 2022-01-11:
You'll have to do a bit of work to get everything compatible, but I imagine it's possible. Only one way to find out, lol.

> can you suggest me standard Bluetooth ELM327?

[This might work](https://www.amazon.com/Bluetooth-Diagnostic-Interface-Scanner-Software/dp/B0953KN9WZ/ref=sr_1_5?crid=3R3O032FIWK6V&keywords=elm327&qid=1641935416&sprefix=elm327%2Caps%2C94&sr=8-5)

closed 2022-01-14 by PowerBroker2