PowerBroker2/ELMduino #105: ELMduino 3.0 possible defect

lebowitz987, 2022-01-09:
PB - While working on a problem with my own code this morning, I saw that you published version 3.0 of your library.  I updated my own copy and then read your notes about the calls now being non-blocking and how to structure my own code to know when a response had been received.  I made those updates and went to run a test.  My own code queries several PIDs and displays the output on a small LCD panel as well as using a small NeoPixel strip as a "shift light" to indicate the RPM range.  What I've found is that if multiple PIDs are queried, the test of "nb_rx_state" never returns ELM_SUCCESS.  What I suspect that as long as there is a request outstanding, which in my code, will likely ALWAYS be the case.  I'm going to create a version of my code that will query and check for the response of each PID in turn.  My current code and 2 example debug outputs are attached.  Apologies for the bits of slop in the code.  

`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>
#include <Adafruit_NeoPixel.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

uint32_t  rpm = 0;
uint32_t  oilTempC = 0;
uint32_t  maf = 0;
uint32_t  coolantTemp = 0;
static char out_str[5];
static float O2current;
static float O2ratio;
static float ltFuelTrim;
static float stFuelTrim;
static unsigned long currentTime;
static unsigned long prevTime;
static unsigned long deltaT;

// Which pin on the Arduino is connected to the NeoPixels?
// On a Trinket or Gemma we suggest changing this to 1:
#define LED_PIN   23

// Which pin is connected to the pot for the LED brightness?
//
#define LED_BRITE 34
const int LEDpotPin = LED_BRITE;

// How many NeoPixels are attached to the Arduino?
#define LED_COUNT 8

// Declare our NeoPixel strip object:
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
// Argument 1 = Number of pixels in NeoPixel strip
// Argument 2 = Arduino pin number (most are valid)
// Argument 3 = Pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  // take our first time stamp
  //
  currentTime = millis();
  prevTime = currentTime;

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

  strip.begin();            // INITIALIZE NeoPixel strip object (REQUIRED)
  // int potVal = analogRead(LEDpotPin);         // read the value of the pot
[... 151 lines cut ...]
// Rainbow-enhanced theater marquee. Pass delay time (in ms) between frames.
void theaterChaseRainbow(int wait) {
  int firstPixelHue = 0;     // First pixel starts at red (hue 0)
  for (int a = 0; a < 30; a++) { // Repeat 30 times...
    for (int b = 0; b < 3; b++) { //  'b' counts from 0 to 2...
      strip.clear();         //   Set all pixels in RAM to 0 (off)
      // 'c' counts up from 'b' to end of strip in increments of 3...
      for (int c = b; c < strip.numPixels(); c += 3) {
        // hue of pixel 'c' is offset by an amount to make one full
        // revolution of the color wheel (range 65536) along the length
        // of the strip (strip.numPixels() steps):
        int      hue   = firstPixelHue + c * 65536L / strip.numPixels();
        uint32_t color = strip.gamma32(strip.ColorHSV(hue)); // hue -> RGB
        strip.setPixelColor(c, color); // Set pixel 'c' to value 'color'
      }
      strip.show();                // Update strip with new contents
      delay(wait);                 // Pause for a moment
      firstPixelHue += 65536 / 90; // One cycle of color wheel over 90 frames
    }
  }
}
[rpm-only.pdf](https://github.com/PowerBroker2/ELMduino/f
[rpm-oilTemp.pdf](https://github.com/PowerBroker2/ELMduino/files/7835381/rpm-oilTemp.pdf)
iles/7835380/rpm-only.pdf)
`

PowerBroker2, 2022-01-09:
I'm happy to help, but please fix your code tags. [See here](https://youtu.be/ZjuSilc99SY)

lebowitz987, 2022-01-09:
Apologies.  I'm not sure what happened within the text of the code to do that.  I've appended the file to the original question as a .txt file.

lebowitz987, 2022-01-09:
I've refactored my code to call each PID function one at a time and to wait for the response before moving on.  For example:
`DEBUG_PORT.print("call rpm()\n");
  float tempRPM = myELM327.rpm();
  DEBUG_PORT.print("call rpm() complete\n");
  while (myELM327.nb_rx_state == ELM_GETTING_MSG ) {
    DEBUG_PORT.print("rpm == ELM_GETTING_MESSAGE\n");
    delay(10);
  }
  DEBUG_PORT.print("rpm != ELM_GETTING_MESSAGE\n");
  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    DEBUG_PORT.print("rpm ELM_SUCESSES\n");
    rpm = (uint32_t)tempRPM;
    (void) shiftLight(rpm);
    DEBUG_PORT.print("RPM: "); DEBUG_PORT.println(rpm);
    lcd.clear();                    // clear the LCD
    lcd.setCursor(0, 0);            // Set the curor at char 0 on line 0
    lcd.print("R: ");               // Print the RPM value
    lcd.print(rpm);
  }   else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    DEBUG_PORT.print("rpm != ELM_SUCESSES\n");
    myELM327.printError();
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("ELM327 error");
  }`

The resulting debug output indicates that the nb_rx_state contains the value ELM_GETTING_MSG for at least several seconds (as long as I was watching).  It doesn't appear to ever change state to ELM_SUCCESS.  I'm going to create a stripped down version to test this further.  The debug output is attached.
[rpm never returns value.pdf](https://github.com/PowerBroker2/ELMduino/files/7835708/rpm.never.returns.value.pdf)

PowerBroker2, 2022-01-09:
Oops, I'm sorry about the link in my first comment. I accidentally used the wrong URL - [I've since updated it](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting). Take a look at the link and post your latest sketch (complete - no snippets) in correct tags.

lebowitz987, 2022-01-10:
Ok.  Here is the original sketch.  I'm working on an experiment to see if I can implement what is in effect, a block for the call. 

```
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>
#include <Adafruit_NeoPixel.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

uint32_t  rpm = 0;
uint32_t  oilTempC = 0;
uint32_t  maf = 0;
uint32_t  coolantTemp = 0;
static char out_str[5];
static float O2current;
static float O2ratio;
static float ltFuelTrim;
static float stFuelTrim;
static unsigned long currentTime;
static unsigned long prevTime;
static unsigned long deltaT;

// Which pin on the Arduino is connected to the NeoPixels?
// On a Trinket or Gemma we suggest changing this to 1:
#define LED_PIN   23

// Which pin is connected to the pot for the LED brightness?
//
#define LED_BRITE 34
const int LEDpotPin = LED_BRITE;

// How many NeoPixels are attached to the Arduino?
#define LED_COUNT 8

// Declare our NeoPixel strip object:
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
// Argument 1 = Number of pixels in NeoPixel strip
// Argument 2 = Arduino pin number (most are valid)
// Argument 3 = Pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  // take our first time stamp
  //
  currentTime = millis();
  prevTime = currentTime;

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

  strip.begin();            // INITIALIZE NeoPixel strip object (REQUIRED)
[... 364 lines cut ...]
    return;
}

// Rainbow-enhanced theater marquee. Pass delay time (in ms) between frames.
void theaterChaseRainbow(int wait) {
  int firstPixelHue = 0;     // First pixel starts at red (hue 0)
  for (int a = 0; a < 30; a++) { // Repeat 30 times...
    for (int b = 0; b < 3; b++) { //  'b' counts from 0 to 2...
      strip.clear();         //   Set all pixels in RAM to 0 (off)
      // 'c' counts up from 'b' to end of strip in increments of 3...
      for (int c = b; c < strip.numPixels(); c += 3) {
        // hue of pixel 'c' is offset by an amount to make one full
        // revolution of the color wheel (range 65536) along the length
        // of the strip (strip.numPixels() steps):
        int      hue   = firstPixelHue + c * 65536L / strip.numPixels();
        uint32_t color = strip.gamma32(strip.ColorHSV(hue)); // hue -> RGB
        strip.setPixelColor(c, color); // Set pixel 'c' to value 'color'
      }
      strip.show();                // Update strip with new contents
      delay(wait);                 // Pause for a moment
      firstPixelHue += 65536 / 90; // One cycle of color wheel over 90 frames
    }
  }
}
```

PowerBroker2, 2022-01-10:
Using blocking while loops are sub-optimal for Arduino projects like this. It would probably be better to use [a state machine](https://www.elprocus.com/finite-state-machine-mealy-state-machine-and-moore-state-machine/). Either way, a simple fix should be below:

```C++
  DEBUG_PORT.print("call rpm()\n");
  float tempRPM = myELM327.rpm();
  delay(100);
  
  DEBUG_PORT.print("call rpm() complete\n");
  while (myELM327.nb_rx_state == ELM_GETTING_MSG ) {
    DEBUG_PORT.print("rpm == ELM_GETTING_MESSAGE\n");
    float tempRPM = myELM327.rpm(); // <----------------- NEW ---------------
    delay(100);
  }
```

lebowitz987, 2022-01-10:
I was thinking along those lines, but with a do until loop. I’ll post
something after I test it.

lebowitz987, 2022-01-10:
Ok.  I've created a version in which I loop until the rpm() returns with something other than ELM_GETTING_MESSAGE.  Now I need to create a version in which I can call the other functions as well.  Since I don't need those other values as often, I'll come up with a structure that only gets those at some interval of the execution of loop().  I had originally tested this with a delay() within the do-while loop, but this turned out to be unnecessary.  The latency to complete the do-while loop is on the order of 200 ms after the initial loop through.

I realize that this may not be optimal, but I don't yet see a way to query multiple PIDs without knowing which PID is being processed at that time.  Otherwise, I can see a condition in which nb_rx_state is set to ELM_SUCCESS for one PID but another (or several others) hasn't completed yet.  Perhaps a mechanism to return the status for each PID called so that they can be tested independently.

Here is my current rpm() only sketch (debug output is attached):

```
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>
#include <Adafruit_NeoPixel.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

uint32_t  rpm = 0;
uint32_t  oilTempC = 0;
static char out_str[5];
static float ltFuelTrim;
static unsigned long currentTime;
static unsigned long prevTime;
static unsigned long deltaT;
static unsigned callLoopBeginTime;
static unsigned callLoopEndTime;

// Which pin on the Arduino is connected to the NeoPixels?
// On a Trinket or Gemma we suggest changing this to 1:
#define LED_PIN   23

// Which pin is connected to the pot for the LED brightness?
//
#define LED_BRITE 34
const int LEDpotPin = LED_BRITE;

// How many NeoPixels are attached to the Arduino?
#define LED_COUNT 8

// Declare our NeoPixel strip object:
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
// Argument 1 = Number of pixels in NeoPixel strip
// Argument 2 = Arduino pin number (most are valid)
// Argument 3 = Pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  // take our first time stamp
  //
  currentTime = millis();
  prevTime = currentTime;

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

[... 128 lines cut ...]

// Rainbow-enhanced theater marquee. Pass delay time (in ms) between frames.
void theaterChaseRainbow(int wait) {
  int firstPixelHue = 0;     // First pixel starts at red (hue 0)
  for (int a = 0; a < 30; a++) { // Repeat 30 times...
    for (int b = 0; b < 3; b++) { //  'b' counts from 0 to 2...
      strip.clear();         //   Set all pixels in RAM to 0 (off)
      // 'c' counts up from 'b' to end of strip in increments of 3...
      for (int c = b; c < strip.numPixels(); c += 3) {
        // hue of pixel 'c' is offset by an amount to make one full
        // revolution of the color wheel (range 65536) along the length
        // of the strip (strip.numPixels() steps):
        int      hue   = firstPixelHue + c * 65536L / strip.numPixels();
        uint32_t color = strip.gamma32(strip.ColorHSV(hue)); // hue -> RGB
        strip.setPixelColor(c, color); // Set pixel 'c' to value 'color'
      }
      strip.show();                // Update strip with new contents
      delay(wait);                 // Pause for a moment
      firstPixelHue += 65536 / 90; // One cycle of color wheel over 90 frames
    }
  }
}
```

[rpm-only-success.pdf](https://github.com/PowerBroker2/ELMduino/files/7840069/rpm-only-success.pdf)

lebowitz987, 2022-01-10:
So here is what I've put together that gives me multiple PIDs.  I've setup a counter that cycles from 0-20.  The rpm PID is called each cycle through loop().  The other PIDs are called one in 20 cycles through loop().  I've done this for 2 reason.  One, is that the data in these other PIDs isn't as time sensitive as rpm and the other is that it reduces the latency through most of the loop() iterations to keep the latency for rpm() down.  Adding one additional call for another PID doesn't seem to add much latency, where calling all of the PIDs each cycle increases the latency enough that the shift light display (a row of LEDs) doesn't respond as quickly as it needs to.

```
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal.h>
#include <Adafruit_NeoPixel.h>

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

LiquidCrystal lcd(13, 12, 14, 27, 26, 25);

ELM327 myELM327;

uint32_t  rpm = 0;
uint32_t  oilTempC = 0;
static char out_str[5];
static float ltFuelTrim;
static unsigned long currentTime;
static unsigned long prevTime;
static unsigned long deltaT;
static unsigned long callLoopBeginTime;
static unsigned long callLoopEndTime;
static unsigned int loopCount;

// Which pin on the Arduino is connected to the NeoPixels?
// On a Trinket or Gemma we suggest changing this to 1:
#define LED_PIN   23

// How many NeoPixels are attached to the Arduino?
#define LED_COUNT 8

// Declare our NeoPixel strip object:
Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRB + NEO_KHZ800);
// Argument 1 = Number of pixels in NeoPixel strip
// Argument 2 = Arduino pin number (most are valid)
// Argument 3 = Pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
//   NEO_RGBW    Pixels are wired for RGBW bitstream (NeoPixel RGBW products)

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  //
  // set loopCount to zero before we enter loop()
  //
  loopCount = 0;
  
  //
  // take our first time stamp
  //
  currentTime = millis();
  prevTime = currentTime;

  lcd.begin(16, 2);         // INITIALIZE the LCD as 16 char by 2 lines
  lcd.clear();              // clear the LCD
  lcd.setCursor(0, 0);      // Set the curor at char 0 on line 0
  lcd.print("LCD Enabled"); // Show this text to let me know that LCD is up

  strip.begin();            // INITIALIZE NeoPixel strip object (REQUIRED)
  strip.setBrightness(15);  // Set BRIGHTNESS to about 1/5 (max = 255)
[... 189 lines cut ...]
    return;
}

// Rainbow-enhanced theater marquee. Pass delay time (in ms) between frames.
void theaterChaseRainbow(int wait) {
  int firstPixelHue = 0;     // First pixel starts at red (hue 0)
  for (int a = 0; a < 30; a++) { // Repeat 30 times...
    for (int b = 0; b < 3; b++) { //  'b' counts from 0 to 2...
      strip.clear();         //   Set all pixels in RAM to 0 (off)
      // 'c' counts up from 'b' to end of strip in increments of 3...
      for (int c = b; c < strip.numPixels(); c += 3) {
        // hue of pixel 'c' is offset by an amount to make one full
        // revolution of the color wheel (range 65536) along the length
        // of the strip (strip.numPixels() steps):
        int      hue   = firstPixelHue + c * 65536L / strip.numPixels();
        uint32_t color = strip.gamma32(strip.ColorHSV(hue)); // hue -> RGB
        strip.setPixelColor(c, color); // Set pixel 'c' to value 'color'
      }
      strip.show();                // Update strip with new contents
      delay(wait);                 // Pause for a moment
      firstPixelHue += 65536 / 90; // One cycle of color wheel over 90 frames
    }
  }
}
```

closed 2022-01-15 by PowerBroker2