PowerBroker2/ELMduino #106: Support for multi PID request

costeacostin, 2022-01-17:
Hello,

I really like your project, it's really nice.
One feature that I feel it's need it's the ability to send multiple requests in one request.
More info about this are present here: https://stackoverflow.com/questions/21334147/send-multiple-obd-commands-together-and-get-response-simultaneously
In my opinion it's needed because if you want for example to take speed and rpm there are needed 2 requests and this will impact a lot the speed of the code.

Great job with the repo, it's amazing!

PowerBroker2, 2022-01-17:
Do you mean have the ability to send a query string like this:?

```
010C010D
```

where `010C` is the query for RPM and `010D` is the query for kph? Or are you trying to send `010C` immediately followed by `010D` and parsing both responses at once?

Either way, I don't have a lot of time to dedicate to this library at the moment or near future and I have not tested multiple queries before.

Anwar0510, 2022-01-17:
I think he meant like this, send `010C` immediately followed by `010D` and parsing both responses at once.
can we just print the data on the serial monitor without printing  _responByte etc.?

PowerBroker2, 2022-01-17:
You can turn off debug mode and then process the data yourself and see what it takes to parse the data

Anwar0510, 2022-01-19:
Hello, thanks again for the reply. I want to retrieve engine load and throttle position data, is this method correct?
 `//Load`
  float tempLoad = myELM327.load();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    load = (uint32_t)tempLoad;
    Serial.print("Engine Load: "); Serial.println(load);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
    
    
  

> You can turn off debug mode and then process the data yourself and see what it takes to parse the data

 Did You mean i have turn off debug mode on .cpp file, or just turn off this `#define DEBUG_PORT Serial`?

Anwar0510, 2022-01-19:
First of all, sorry, didn't mean to spam. I just tried querying for three data, namely RPM, throttle and Timing Advance. With the code as below.
```
//querying for RPM
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();

  //querying for Throttle Position
  float tempThrottle = myELM327.throttle();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    throttle = (uint32_t)tempThrottle;
    Serial.print("Throttle: "); Serial.println(throttle);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();

    //querying for Timing Advance
  float tempTiming = myELM327.timingAdvance();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    sudutPengapian = (uint32_t)tempTiming;
    Serial.print("Timing Advance: "); Serial.println(sudutPengapian);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
```

 actually it has succeeded in requesting data, only there are some "spots" where the RPM value = 0 while for vehicles that use normally 1400 - 1600 rpm, as well as the throttle value the normal value for my throttle is 10 idle conditions, but why on some "spots" the value is 55, the same as Timing advance.

I sent 2 pictures of the spot I mean
[image]
[image]

closed 2023-03-02 by PowerBroker2