PowerBroker2/ELMduino #107: Motorcycle user querying Data

Anwar0510, 2022-01-20:
Hello @PowerBroker2, Thanks in advance And thanks again for this powerful library. The first time I tried this library, I couldn't connect ELM327 with ESP32 assuming the ELM327 I was using uses BLE. Then I tried to connect my ELM327 with the Bluetooth serial terminal app, it uses Bluetooth classic and I tried some AT commands, I also checked for compatibility using PID `01 00` and the response I got `41 00 F6 1C B0 01` which means my vehicle supports OBD II, and I checked using the torque app my vehicle uses ISO 14230-4 KWP fast.

My setup
ELM327 v1.5, Sumitomo socket, ESP32 Devkit, vehicles >> (Vario 2020) Motorcycles from Honda
Note: Sumitomo socket is for DLC motorcycles.

I've tried example  ` ESP32_Bluetooth_Serial.ino` successfully in querying the RPM data, but after I query for 3 data, namely rpm, throttle and Timing advance, I got odd results where sometimes there is a value of 0 at rpm, and very large values ​​in advance and throttle timing. Here is my codes for querying 3 data.

```
//my variables
uint8_t address[6] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xBA};
uint32_t rpm = 0;
uint32_t throttle = 0;
uint32_t sudutPengapian = 0; //for Timing advance

// void setup = example
void loop()
{
  //querying for RPM
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();

  //querying for Throttle Position
  float tempThrottle = myELM327.throttle();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    throttle = (uint32_t)tempThrottle;
    Serial.print("Throttle: "); Serial.println(throttle);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();

    //querying for Timing Advance
  float tempTiming = myELM327.timingAdvance();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    sudutPengapian = (uint32_t)tempTiming;
    Serial.print("Timing Advance: "); Serial.println(sudutPengapian);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();

}
```
So why am I getting a value of 0/high in some conditions?, Is there something wrong with the code I'm using?
Please help

PowerBroker2, 2022-01-20:
The library recently became non-blocking. Because of this, you should not query a new pid until you get either a valid answer or error from the previous pid. A good way to implement this is via a state machine with a switch statement (and a case for each pid to query)

PowerBroker2, 2022-01-20:
I'll probably add an example to the library soon that shows how to do this

Anwar0510, 2022-01-20:
> The library recently became non-blocking

 So, one PID query must be completed before querying for another PID? what does non-blocking mean like this?
 
 

> I'll probably add an example to the library soon that shows how to do this

Yes, many people who only understand the basics of programming like me need it, we will really appreciate it thanks.
Actually I take course Mechanical eng. , but i'm so curious about electrical

PowerBroker2, 2022-01-20:
> So, one PID query must be completed before querying for another PID? what does non-blocking mean like this?

It means that `myELM327.throttle()` (and other PID queries) will not wait for the ELM327 to respond or timeout. It instead queries the PID on the first call and requires subsequent calls to parse the ELM327 (or to decide a timeout or error has occurred). So, if you call `myELM327.throttle()` and immediately call `myELM327.rpm()`, you will be sending two PID queries without parsing any responses, giving you no data (or worse, bad data). The previous version of the library could handle this easier since `myELM327.throttle()` wouldn't even return until it finished parsing, an error occurred, or a timeout happened. This is why it was "blocking", which you don't want to do in Arduino code.

Anwar0510, 2022-01-20:
> The previous version of the library could handle this easier since `myELM327.throttle()` wouldn't even return until it finished parsing, an error occurred, or a timeout happened. 

So my statement "one PID query must be completed before querying for another PID" more precise to the previous version of the library?

Is it true that the purpose of the inclusion of this non-blocking for data transmission rate?

PowerBroker2, 2022-01-20:
> So my statement "one PID query must be completed before querying for another PID" more precise to the previous version of the library?

I'm not sure what you mean by that

> Is it true that the purpose of the inclusion of this non-blocking for data transmission rate?

Making the library non-blocking doesn't affect the transmission rate - it allows the Arduino to do other stuff while waiting for the ELM327 to respond. Non-blocking code makes Arduino programs more efficient.

MeCoolAus, 2022-01-24:
> I'll probably add an example to the library soon that shows how to do this

Hi @PowerBroker2, I would also definitely appreciate any example of the state machine principle, since it's not familiar to me. I happily updated your library, but "bricked" my latest code, unfortunately. Doing while cycle `while (myELM327.nb_rx_state == ELM_GETTING_MSG)` seems to not fix it at all, instead it locks the code inside the while cycle.

Thanks in advance.

PowerBroker2, 2022-01-25:
This is a good example: [https://github.com/patfelst/ELMduino/blob/8780d5fb05bf097b2e0d1d0268ca075110b51a36/examples/non_blocking_2/non%20blocking%20example%202.ino](https://github.com/patfelst/ELMduino/blob/8780d5fb05bf097b2e0d1d0268ca075110b51a36/examples/non_blocking_2/non%20blocking%20example%202.ino)

Anwar0510, 2022-01-26:
Hello, @PowerBroker2
Thanks for the examples provided, the library is very powerful. I didn't expect you to type thousands of lines in ELMduino.cpp.
Again, You are AMAZING!!!

closed 2022-01-30 by PowerBroker2