PowerBroker2/ELMduino #109: Error compilling

AekPu, 2022-01-26:
Arduino: 1.8.19 (Windows 10), Board: "Arduino Uno"

C:\Users\Aekkarad\Documents\Arduino\libraries\ELMDuino-3.0.0\src\ELMduino.cpp: In member function 'float ELM327::batteryVoltage()':

C:\Users\Aekkarad\Documents\Arduino\libraries\ELMDuino-3.0.0\src\ELMduino.cpp:2632:11: error: 'strtof' was not declared in this scope

    return strtof(payload, NULL);

           ^~~~~~

C:\Users\Aekkarad\Documents\Arduino\libraries\ELMDuino-3.0.0\src\ELMduino.cpp:2632:11: note: suggested alternative: 'strtok'

    return strtof(payload, NULL);

           ^~~~~~

           strtok

exit status 1

Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

PowerBroker2, 2022-01-26:
What happens if you replace all `strtof()` calls to `strtod()`? Does the error go away?

PowerBroker2, 2022-01-26:
No, if you go into ELMduino.cpp and replace all instances of `strtof` to `(float)strtod`, do you still get a compile time error?

AekPu, 2022-01-27:
It complied and No error

PowerBroker2, 2022-01-27:
Fixed in [3.0.1](https://github.com/PowerBroker2/ELMduino/releases/tag/3.0.1)

closed 2022-01-27 by PowerBroker2

AekPu, 2022-01-27:
for use , i must setting HC05 in AT MODE right?