PowerBroker2/ELMduino #110: ESP32 connects with BT ODBII Adapter but after Pairing with Phone not anymore?

gordonBimig, 2022-01-26:
Hello, 

I've configured a ESP32 to read DPF data from an Opel engine/ecu using the sketch from this project:
https://www.instructables.com/Opel-DPF-Indicator-Monitor/

At first it worked, but suddently not anymore. I assumed something is wrong with the BT Adapter, since another ESP32 I've programmed  (after encountered this issue ) with the same sketch did not work either, strangely. However, over an Smarthpone App like Torque or OPL Monitor I was able to communicate over the BT adapt, which by the was is called "OBDII" on the BT protocol and I certainly edited the sketch for the correct device name.

So I ordered another BT Adapter called "Veepeak" which seems to be more or less the same I used before. It also registers with the name "OBDII". So I din't have to change the sketch. So i immediately tried the originally programmed esp32 and it worked like a charm.

THEN .... i connected my Smartphone to the Adapter which was prompting me for a PIN which ist 1234 according to the manual. Also the apps were working fine with the BT Adapter, but ever since, the ESP32 won't work with the BT adapter anymore.
So I assumed, maybe only one device may be connected to the BT Adapter and thus I thought, maybe my Smartphone is interfeering, thus I disabled BT, which didn't help, and also shutdown the phone completely to be sure, but the ESP32 won't connect to the BT Adapter anymore. This is so weitd and I have no clue what is going on. 

I'm permanently getting the erro "OBD error 2" which occurs when this state is meet:

  if (!myELM327.begin(ELM_PORT))

So now I have 2 BT Adapters and 2 ESP32's which don't work anymore, after the worked initially (the Sketch/Code has not been changed, after the first upload).
  
  Does anybody has an idea what could be the problem here?

I read that ESP to BT Adapter via Elmduino library does not require a PIN, but the problems began after connecting to the BT Adapter via Smarthone and PIN, therefore I'm thinking something changed in the BT Adapter, which is causing the Adapter not to work without PIN/pairing anymore? Just my thought, not sure how much sense that makes. However, this is driving me crazy. Would be great if somebody would have a hint what's going on here.

PowerBroker2, 2022-01-26:
> I'm permanently getting the erro "OBD error 2" which occurs when this state is meet:

What is the full debug printout?

gordonBimig, 2022-01-27:
Hi,

good point :-)

not sure if this is the Debug output. I'm only getting this on the monitor output:

reconnecting..
         ERROR: ELM_TIMEOUT
Lost connection..
reconnecting..
         ERROR: ELM_TIMEOUT

and so on.

But this does not seem to be the debug output or is it? The scond parameter is set to true, so DebugMode should be on?

ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

MeCoolAus, 2022-01-29:
Hi @gordonBimig, I was having the same issue #70 here and found the solution [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/bt_remove_paired_devices/bt_remove_paired_devices.ino). It removes all paired devices from the ESP32 bt-device-list, so therefore ESP32 initializes brand new pairing with successful connection then.

gordonBimig, 2022-01-29:
@MeCoolAus 
thanks a tousand times. I still don't understand how this all fit's together, but the referenced sketch showed 2 paired devices and after deleting all I was able to connect again to the BT dongle! Thanks again for chiming in :-)

closed 2022-01-29 by PowerBroker2