PowerBroker2/ELMduino #111: Attempting to use library with Sparkfun OBD II UART board

jfield6242, 2022-01-28:
I am trying to get vehicle data and have attempted many libraries as well as parsing the data myself. Nothing seems to be consistent.
Using the hardware serial example, I successfully receive "Connected to ELM327" and then an infinite loop of Searching and Stopping. 
Any suggestions would be greatly appreciated. Sorry if the code is tagged incorrectly, I will also add the files. 
Unfortunately the log is only the second half because I had to switch terminals to save the log.

Hardware: 
-Arduino Mega
-Sparkfun OBD II UART board (uses STN1110 IC compatible with ELM327 command set)

Log: 
[log.log](https://github.com/PowerBroker2/ELMduino/files/7962471/log.log)

Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: S
	Received char: E
	Received char: A
	Received char: R
	Received char: C
	Received char: H
	Received char: I
	Received char: N
	Received char: G
	Received char: .
	Received char: .
	Received char: .
	Received char: \r
Timeout detected with overflow of 0ms
Received: SEARCHING...
ERROR: ELM_TIMEOUT
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: S
	Received char: T
	Received char: O
	Received char: P
	Received char: P
	Received char: E
	Received char: D
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Received: STOPPED
ERROR: ELM_STOPPED

Code: 
[OBDtest.txt](https://github.com/PowerBroker2/ELMduino/files/7962469/OBDtest.txt)

`#include <ELMduino.h>

#define ELM_PORT Serial3

ELM327 myELM327;

uint32_t vehicleRPM = 0; 

void setup() 
{
  Serial.begin(9600);
  ELM_PORT.begin(9600);

  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop() 
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    vehicleRPM = (uint32_t)tempRPM;
    Serial.print("RPM: "); 
    Serial.println(vehicleRPM);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
  {
    myELM327.printError();
  }
}`

PowerBroker2, 2022-01-28:
What is your car's make, model, and year?

It looks to me that the car's diagnostics computer doesn't want to give the ELM327 any data for some reason. If putting a 1 second-ish delay in the loop doesn't solve things, you might be SOL.

jfield6242, 2022-01-29:
2018 GMC Sierra. 
I will try on my 2003 Subaru Outback and also try a delay. Thank you!

PowerBroker2, 2022-01-29:
2003 might be worse - OBD2 wasn't made standard for all cars until 2008. Good luck, tho

jfield6242, 2022-01-29:
Works accurately and consistently on my Subaru. Thank you so much for the help and the library!!!

closed 2022-01-30 by PowerBroker2