PowerBroker2/ELMduino #118: ESP32 Bluetooth Mazda

maciekelga, 2022-04-01:
Hi,

I want to ask about code for BT connection.

I`m using ELM327 with BT, which name is "v-link". In the code i see only line where ESP is checking is connection is done.

` DEBUG_PORT.begin(115200);`
 ` //SerialBT.setPin("1234");`
  `ELM_PORT.begin("ArduHUD", true);`
 ` if (!ELM_PORT.connect("OBDII"))`
 ` {`
   ` DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");`
`    while(1);`
 ` }`

1. Is it shall be before `DEBUG_PORT` inserted `ELM_PORT.connect("v-link)` ??
2. What is `"ArduHUD"` it can/shall be changed?
3. I want to look into custom PID from Mazda6 for DPF regeneration

Name: DPF Regen
Short Name: DPF Regen
PID: 220380
Equation: A
Min: 0.0
Max: 100.0
Units: ON / OFF

Could You help me to implement it?

PowerBroker2, 2022-04-01:
> Is it shall be before DEBUG_PORT inserted ELM_PORT.connect("v-link) ??

I think that is correct - did you test it?

> What is "ArduHUD" it can/shall be changed?

ArduHUD is a personal project of mine that uses ELMduino. I'm not sure if I updated the source code in ArduHUD to be compatible with ELMduino, so be careful when using ArduHUD as an example of how to use ELMduino. It would be better to just use the examples in this library and go from there.

> I want to look into custom PID from Mazda6 for DPF regeneration

Custom PIDs are supported. The easiest way to do this is by using `processPID()`: https://github.com/PowerBroker2/ELMduino/blob/d2be43ce5424b9782e09ce1357bba42acda15ce6/src/ELMduino.cpp#L472-L514

In your case, you could do something like this:

```C++
byte dpfRegen = myELMduino.processPID(<<service number here>>, 220380, 1, 1, 1, 0);
```

You'll have to verify the service number, scale factor, and bias to get it to work properly.

maciekelga, 2022-04-01:
Thank You!

I will try today to connect to my v-link, I didn`t try yet.

I guess I should use following sketch right?
https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino

Regarding PID is difficult to say, I took it from here
https://www.mazdaclub.it//forum/viewtopic.php?f=52&t=23307&hilit=extra+pids

But for me only this for DPF regen is important. I want to make a LED which give me indication when DPF regeneration is started.
I have a Forscan Android app with BT ELM327 / Windows app and hardwire ELM327 as well. Is it any way to check with some BT sniffer to read all information about PID?

maciekelga, 2022-04-01:
I tried connect to V-LINK with following code as in attachment

> #include "BluetoothSerial.h"
> #include "ELMduino.h"
> 
> 
> BluetoothSerial SerialBT;
> #define ELM_PORT   SerialBT
> #define DEBUG_PORT Serial
> 
> 
> ELM327 myELM327;
> 
> 
> uint32_t rpm = 0;
> 
> 
> void setup()
> {
> #if LED_BUILTIN
>   pinMode(LED_BUILTIN, OUTPUT);
>   digitalWrite(LED_BUILTIN, LOW);
> #endif
> 
>   DEBUG_PORT.begin(115200);
>   SerialBT.setPin("1234");
>   ELM_PORT.begin("ArduHUD", true);
>   ELM_PORT.connect("V-LINK");
>   
>   if (!ELM_PORT.connect("V-LINK"))
>   {
>     DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
>     while(1);
>   }
>   Serial.println("Connected to ELM327");
> }
> 
> 
> void loop()
> {
>   float tempRPM = myELM327.rpm();
> 
>   if (myELM327.nb_rx_state == ELM_SUCCESS)
>   {
>     rpm = (uint32_t)tempRPM;
>     Serial.print("RPM: "); Serial.println(rpm);
>   }
>   else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
>     myELM327.printError();
> 
>     byte dpfRegen = myELM327.processPID(0, 220380, 1, 1, 1, 0);
>     Serial.print("DPF: "); Serial.println(dpfRegen);
> }

And I got readings as here
https://drive.google.com/file/d/1XxWVrLmUap_EMby-vURdi4Zeay2PIgda/view?usp=sharing

From what I know this PID is return only 0 or 1. Now 2 things
1. Is it possible to check if received data is correct? I mean is possible to check if query readings is correct from software point of view ? If contain all necessary data?
2.  I need to do this reading only 1 time per second, even 1 time per 2-3 second is fine, how to do it in the best way?

PowerBroker2, 2022-04-01:
I can't help with debugging your code unless there's a library issue/bug. You can use other forums for that help. As a side note, your service number shouldn't be 0. It probably should be 2, but you might want to ask in the mazda forum.

maciekelga, 2022-04-01:
Sure.

Service number is it the same as header?

In additional part of code which contain `myELM327.nb_rx_state` command, what is a purpose of this part?

Could You just help me how to read in right way command one time per second? Just use command waitms?

closed 2022-04-01 by PowerBroker2