PowerBroker2/ELMduino #121: Serial bt sketch + suggested removed paired devices for esp32 sketch

qqm55, 2022-04-08:
Here is the following code i combined of the two.

`#include "BluetoothSerial.h"
#include "ELMduino.h"

#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include"esp_gap_bt_api.h"
#include "esp_err.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

#define REMOVE_BONDED_DEVICES 1   // <- Set to 0 to view all bonded devices addresses, set to 1 to remove

#define PAIR_MAX_DEVICES 20
uint8_t pairedDeviceBtAddr[PAIR_MAX_DEVICES][6];
char bda_str[18];

bool initBluetooth()
{
  if(!btStart()) {
    Serial.println("Failed to initialize controller");
    return false;
  }
 
  if(esp_bluedroid_init() != ESP_OK) {
    Serial.println("Failed to initialize bluedroid");
    return false;
  }
 
  if(esp_bluedroid_enable() != ESP_OK) {
    Serial.println("Failed to enable bluedroid");
    return false;
  }
  return true;
}

char *bda2str(const uint8_t* bda, char *str, size_t size)
{
  if (bda == NULL || str == NULL || size < 18) {
    return NULL;
  }
  sprintf(str, "%02x:%02x:%02x:%02x:%02x:%02x",
          bda[0], bda[1], bda[2], bda[3], bda[4], bda[5]);
  return str;
}

void setup()
{

  Serial.begin(115200);
initBluetooth();
  Serial.print("ESP32 bluetooth address: "); Serial.println(bda2str(esp_bt_dev_get_address(), bda_str, 18));
  // Get the numbers of bonded/paired devices in the BT module
  int count = esp_bt_gap_get_bond_device_num();
  if(!count) {
    Serial.println("No bonded device found.");
  } else {
    Serial.print("Bonded device count: "); Serial.println(count);
    if(PAIR_MAX_DEVICES < count) {
      count = PAIR_MAX_DEVICES; 
      Serial.print("Reset bonded device count: "); Serial.println(count);
    }
    esp_err_t tError =  esp_bt_gap_get_bond_device_list(&count, pairedDeviceBtAddr);
    if(ESP_OK == tError) {
      for(int i = 0; i < count; i++) {
        Serial.print("Found bonded device # "); Serial.print(i); Serial.print(" -> ");
        Serial.println(bda2str(pairedDeviceBtAddr[i], bda_str, 18));     
        if(REMOVE_BONDED_DEVICES) {
          esp_err_t tError = esp_bt_gap_remove_bond_device(pairedDeviceBtAddr[i]);
          if(ESP_OK == tError) {
            Serial.print("Removed bonded device # "); 
          } else {
            Serial.print("Failed to remove bonded device # ");
          }
          Serial.println(i);
        }
      }        
    }
  }

  
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif
Serial.println("LED_BUILTIN");
  DEBUG_PORT.begin(115200);
  Serial.println("DEBUG_PORT");
  SerialBT.setPin("1234");
  Serial.println("CsetPin");
  ELM_PORT.begin("ArduHUD", true);
  Serial.println(" ELM_PORT");
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}`

**then this is the output it dies midway printing the esp32 bluetooth device address before cleaning**

output:
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:43:32.873 -> configsip: 188777542, SPIWP:0xee
01:43:32.873 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:43:32.873 -> mode:DIO, clock div:2
01:43:32.873 -> load:0x3fff0018,len:4
01:43:32.873 -> load:0x3fff001c,len:1216
01:43:32.873 -> ho 0 tail 12 room 4
01:43:32.873 -> load:0x40078000,len:10944
01:43:32.873 -> load:0x40080400,len:6360
01:43:32.873 -> entry 0x400806b4
01:43:33.711 -> ESP32`

**and i am able to mitigate this issue somewhat by commenting out setpin and debug_port.begin but then i will not be able to connect to my elm327 which has a pin and im not sure if the code even runs correctly without the debug port.** 

output 
`01:44:34.352 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
01:44:34.352 -> configsip: 188777542, SPIWP:0xee
01:44:34.352 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
01:44:34.352 -> mode:DIO, clock div:2
01:44:34.352 -> load:0x3fff0018,len:4
01:44:34.352 -> load:0x3fff001c,len:1216
01:44:34.352 -> ho 0 tail 12 room 4
01:44:34.352 -> load:0x40078000,len:10944
01:44:34.352 -> load:0x40080400,len:6360
01:44:34.399 -> entry 0x400806b4
01:44:35.240 -> ESP32 bluetooth address: 98:cd:ac:d0:72:3a
01:44:35.240 -> No bonded device found.
01:44:35.240 -> LED_BUILTIN
01:44:35.240 -> [I][BluetoothSerial.cpp:571] _init_bt(): device name set
01:44:35.240 ->  ELM_PORT
01:44:35.240 -> [I][BluetoothSerial.cpp:243] esp_spp_cb(): ESP_SPP_INIT_EVT
01:44:35.240 -> [I][BluetoothSerial.cpp:783] connect(): master : remoteName
01:44:35.240 -> [I][BluetoothSerial.cpp:422] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
01:44:55.727 -> [I][BluetoothSerial.cpp:422] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
01:45:07.248 -> Couldn't connect to OBD scanner - Phase 1
`

**any advice appreciated**

I have a bigger project using this library and it crashes when it hits debug and setpin, i have isolated it to just these pieces of code.

PowerBroker2, 2022-04-08:
Please reformat your issue post, then I'll take a look: [https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting)

qqm55, 2022-04-08:
Apologies for terrible formatting, fixed.

PowerBroker2, 2022-04-08:
Hmm, I never had to mess with setting a PIN or removing paired devices for my ESP32, so I don't know if I can help much. Is another part of your project using `Serial` (`DEBUG_PORT`)? Maybe there's a conflict there?

qqm55, 2022-04-08:
I have two types of elm327's, the genuine one i have requires a pin of 1234 to pair, I'm not sure if that functionality is built into the library making it unnecessary to set the pin with `SerialBT.setPin("1234"); `. I checked my main project and I do not call `Serial(DEBUG_PORT)` at all, Do you have any idea why I needed to erase bonded devices with my esp32 dev board (ESP-WROOM-32) for it to consistently pair, without the bt erase code it would refuse to pair?

PowerBroker2, 2022-04-08:
As long as there are no other bluetooth devices paried to either the ELM327 or ESP32, I don't know why you would need to erase the bonded devices. You might want to post this question in the Espressif and/or Arduino forums for better help.

qqm55, 2022-04-09:
I will try it on both my dev boards without the bt erase part that makes it crash, maybe it's not necessary and I was unable to pair due to other reasons in the past? Will update with findings and If successful I owe you a six pack or coffee for this library.

qqm55, 2022-04-09:
I tried it on a Open Smartwatch board, cleared bonded devices in a separate sketch, uploaded the code without the bt remove code and it worked flawlessly and repaired multiple times, Im not sure but in a pass revision on the other esp32 dev board I had to clear the code for it to pair, but now that Im thinking about it it could've been my android auto pairing with it and the bt clear code could've been unnecessary on the wroom board. 

I then tried it without the BT clear code in my main sketch and It would still crash with the inclusion of DEBUG port begin and setpin, so there probably is more to this then just this code I thought I isolated but I was able to confirm that DEBUG port begin and setpin are not necessary to pair and pull data from the elm327 (even if it has a pin of 1234) which im sure you knew already lol. 

So my project is working now! How can we get a private conversation going so I can get you that six pack and if we can chat some more technical things about your library?

PowerBroker2, 2022-04-09:
Glad you got it working! Knowing my library(ies) are useful to other people is repayment enough. As for technical details, I don't have much time and all I know about ELM327s is from the documentation ([datasheet](https://www.elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf), [AT commands](https://www.sparkfun.com/datasheets/Widgets/ELM327_AT_Commands.pdf), [PIDs](https://en.wikipedia.org/wiki/OBD-II_PIDs)). If you think the library should be edited, feel free to submit a PR

closed 2022-04-09 by PowerBroker2