PowerBroker2/ELMduino #123: ELM_TIMEOUT

zxrxth, 2022-04-26:
Hi, i tried this code that i got from https://forum.arduino.cc/t/elmduino-library-for-car-hacking/593482/16#msg4679253

Code:
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
#define DEBUG_PORT Serial

//String MACadd = "00:10:CC:4F:36:03"; //enter the ELM327 MAC address
uint8_t address[6] = {0x00, 0x10, 0xCC, 0x4F, 0x36, 0x03}; //enter the ELM327 MAC address after the 0x

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, LOW);
#endif

DEBUG_PORT.begin(115200);
//SerialBT.setPin("1234");
ELM_PORT.begin("ArduHUD", true);

if (!ELM_PORT.connect(address)) //"OBDII" replaced by "address"
{
DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
while (1);
}

if (!myELM327.begin(ELM_PORT,true, 2000, AUTOMATIC))
{
Serial.println("Couldn't connect to OBD scanner - Phase 2");
while (1);
}

Serial.println("Connected to ELM327");
}

void loop()
{
float tempRPM = myELM327.rpm();

if (myELM327.nb_rx_state == ELM_SUCCESS)
{
rpm = (uint32_t)tempRPM;
Serial.print("RPM: "); Serial.println(rpm);
}
else
{
myELM327.printError();
}
}

void printError()
{
Serial.print("Received: ");
for (byte i = 0; i < myELM327.recBytes; i++)
Serial.write(myELM327.payload[i]);
Serial.println();

if (myELM327.nb_rx_state == ELM_SUCCESS)
Serial.println(F("\tELM_SUCCESS"));
else if (myELM327.nb_rx_state == ELM_NO_RESPONSE)
Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
else if (myELM327.nb_rx_state == ELM_BUFFER_OVERFLOW)
Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
else if (myELM327.nb_rx_state == ELM_UNABLE_TO_CONNECT)
Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
else if (myELM327.nb_rx_state == ELM_NO_DATA)
Serial.println(F("\tERROR: ELM_NO_DATA"));
else if (myELM327.nb_rx_state == ELM_STOPPED)
Serial.println(F("\tERROR: ELM_STOPPED"));
else if (myELM327.nb_rx_state == ELM_TIMEOUT)
Serial.println(F("\tERROR: ELM_TIMEOUT"));
else if (myELM327.nb_rx_state == ELM_TIMEOUT)
Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

delay(100);
}

Problem:
ELM_TIMEOUT

Some description:
It was successfully connected based on output on the serial monitor and green light on ELM 327. But there is no blue light on ESP32, only red light which seems that the ESP is only turned on but there's no connectivity. Correct me if I'm wrong. But I'm not sure if this is related to the ELM_TIMEOUT. I also tried your suggested solution but nothing happened. When I did the AT command, nothing comes out too. Is there any error in the code? I appreciate your time in looking into it.

PowerBroker2, 2022-04-26:
[Please use markdown for code snippets](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting)

Also, the API for the library has changed a lot since that forum post. I would suggest looking at [the new API](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting) and [current examples](https://github.com/PowerBroker2/ELMduino/tree/master/examples)

zxrxth, 2022-04-27:
okay, I'll try the one you suggested. will update the outcome later! thank you

PowerBroker2, 2022-04-27:
Oops, that's my fault - I updated my comment, but here's the link for convenience: [https://github.com/PowerBroker2/ELMduino#list-of-supported-obd-pid-processing-functions](https://github.com/PowerBroker2/ELMduino#list-of-supported-obd-pid-processing-functions)

zxrxth, 2022-04-27:
thank you so much, I'll try the suggested example and library. will update the outcome later

zxrxth, 2022-04-27:
I've tried and it was successfully connected and the rpm value also came out! But it is not the same as the one displayed on the dashboard. If I may know, why is it? And another question is, what is the difference between List of Supported OBD PID Processing Functions and List of standard PIDs?

PowerBroker2, 2022-04-27:
It should be exactly the same - how off is it?

Some of the standard PIDs listed on Wikipedia are really difficult to implement with C++ functions, so those I refuse to support in this library. I can't remember off the top of my head which PIDs those are, but it's only a handful of obscure PIDs. You can still use the library's lower level functions to manually query/parse those unsupported PIDs if needed.

Are you talking about the debug prints? You can turn those off by deleting the second argument to the elmduino's `begin()` function.

zxrxth, 2022-04-28:
When I didn't press the gas pedal, the rpm value was 4294967295 -- I assumed it is the default value. 
Then I began to press the gas pedal, constantly on rpm 2 (by looking at the dashboard), but the rpm value on the serial monitor changes to 5, 12, and so on. The serial monitor was scrolling too fast I might miss the correct data though.

Oh, okay thank you for the clarification on the PIDs listed.

Oh no, not debug prints. I just want to understand the outputs like the char, for example, what it indicates -- explanation of the outputs.

PowerBroker2, 2022-04-28:
For the RPM calculations:

https://github.com/PowerBroker2/ELMduino/blob/d2be43ce5424b9782e09ce1357bba42acda15ce6/src/ELMduino.cpp#L812-L815

This means query the RPM PID, expect only 1 line of data, expect 2 bytes (`A` and `B`), and scale `(256*A + B)` by `1/4`. This follows the equation you linked to, so the car must be giving you bad data. If you run in debug mode using the example sketch, what does the debug printout look like?

zxrxth, 2022-04-28:
Sorry, I should run example sketch "esp32_test" or "esp32_bluetooth_serial" in debug mode?

PowerBroker2, 2022-04-28:
If you have an ESP32, use "esp32_bluetooth_serial"

zxrxth, 2022-04-28:
[https://drive.google.com/drive/folders/1NkvQRUfxsZzTSkubwmWPsWvwQJJ_X-T_?usp=sharing](url)

Here's the screenshot of debug mode and no debug mode. I edited the part where it connects using the name to using the address because I can't connect if I use the name.

PowerBroker2, 2022-04-28:
Please attach the picture using github markdown or copy and paste the text into your reply with surrounding triple backticks. For instance:

\```
Debug Text
\```

zxrxth, 2022-04-28:
Done, I'm really sorry for the inconvenience I've caused, quite new to github.

PowerBroker2, 2022-04-28:
For some reason your car sends too many bytes as a response to this (and possibly other) PID(s). This confuses ELMduino, resulting in ridiculous RPM data. There might be a way to patch the library, but I can't support such an effort anytime soon. You could do your own workaround by replacing

```C++
float tempRPM = myELM327.rpm();
```

with 

```C++
myELM327.rpm();
float tempRPM = (255*myELM327.responseByte_0 + myELM327.responseByte_1)/4.0;
```

Note, I may have the responseByte numbers mixed up - you can easily test and fix if needed

zxrxth, 2022-04-28:
I see, thank you I will try and keep you updated. If I may ask, is the formula the same for other PID(s) too?  For example, for float fuelLevel();.

PowerBroker2, 2022-04-28:
No, you can use this as a guide: [https://en.wikipedia.org/wiki/OBD-II_PIDs](https://en.wikipedia.org/wiki/OBD-II_PIDs)

zxrxth, 2022-04-28:
okay noted thank you so much.

zxrxth, 2022-04-28:
Hi, I've already tried your suggestion. When I pressed the gas pedal to 1 (on the dashboard), on the serial monitor it displayed 1000 (my car rpm is x1000). I think it was successful?

PowerBroker2, 2022-04-28:
I guess so

closed 2022-04-28 by PowerBroker2