PowerBroker2/ELMduino #124: CAN_ERROR OIL FUEL

zxrxth, 2022-05-09:
Hi, I've tried the code and I am able to retrieve data from OBDII such as rpm, engine coolant temperature, kph, and engine load. But when I tried to retrieve fuel-related data such as oil temperature, fuel type, fuel level, and fuel rate, I encounter CAN_ERROR as photo attached below. Is this has relation to the type of car that I'm currently using for testing? 
[image]

PowerBroker2, 2022-05-09:
Your car is generating that error. Either that data is unavailable on your car or something is wrong with your car's CAN bus/diagnostics electronics.

zxrxth, 2022-05-09:
I see, is there any way to check whether it is because of the unavailable data or because of the car's CAN bus?

PowerBroker2, 2022-05-09:
No idea

zxrxth, 2022-05-10:
Alright, thank you. I'd like to ask, before this I saw a question on one of the ELMduino-related issues asking whether we can retrieve a few outputs simultaneously but seems that I couldn't find it again. I remember that you reply to look for it at an attached link if I'm not mistaken. Could you share it again? Sorry for troubling.

closed 2022-05-11 by PowerBroker2