PowerBroker2/ELMduino #125: 02 Sensors missing

RootUser3885, 2022-05-23:
**Is your feature request related to a problem? Please describe.**
Yes, the O2 Sensors appear to be missing. The PIDs are there but never used.

**Describe the solution you'd like**
Add the functions to read O2 sensor voltage/current/ratio

**Describe alternatives you've considered**
I will add the feature in myself if I can

**Additional context**
I am trying to calculate the MPG using the O2 sensors to monitor the actual a/f ratio being burned. Since my car does not support a fuel flow sensor I must calculate the fuel use with the O2s, MAF, and Fuel trims.

PowerBroker2, 2022-05-24:
Oxygen sensor PIDs are not directly supported via the API because they always return 2 values instead of just 1. That being said, you can still access the oxygen sensor data by calling `processPID()` with the service number and PID you want. Then you can manually process the returned data by accessing `responseByte_0, responseByte_1, ...` etc.

RootUser3885, 2022-05-25:
Thank you for your reply. I will process the O2 data within my program. I forgot that the O2s report multiple values. Seems like an odd design by the OBD standard.

Although it seems like sensors that return multiple values should be split up into their separate values by the API. As the purpose of the library is to obscure that type of inconvenience from the programmer.

Thanks

closed 2022-05-25 by PowerBroker2

RootUser3885, 2022-05-25:
That's true. The library does indeed make processing those PIDs fairly easy. I've used this library in countless projects. Keep it up. Thanks.