PowerBroker2/ELMduino #128: Trying to get data but getting random values. (ESP32)

githubUnknownUser, 2022-06-05:
Good evening, @PowerBroker2! Congratulations on the amazing job you did here!!!

I'm trying to use the library but I'm facing some problems. First, let me give you a hint of my hardware setup:

2010 Honda -> ELM327 -> ESP32-WROVER-B (Bluetooth)

I tried the default example code but I couldn't get the esp to connect, so I took some code from the BluetoothSerial library and then worked. 

When I tried to use the default ELMduino initializer (Auto protocol detector) I got 'ERROR: ELM_TIMEOUT'.

When I set the protocol, the setup completes gratefully, but I'm getting random reads using 'myELM327.rpm()'.

I've tried some workarounds like the #123 or #92 but unfortunately, none of them worked for me.

Can you help me? I might be doing something wrong...

Here's my code and the Logs: (The Bluetooth connection is an absolute mess but I waiting for everything to workout first to clean it)

'
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
#define DEBUG_PORT Serial

#define BT_DISCOVER_TIME 10000
esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE;   // or ESP_SPP_SEC_ENCRYPT|ESP_SPP_SEC_AUTHENTICATE to request pincode confirmation
esp_spp_role_t role    = ESP_SPP_ROLE_SLAVE; // or ESP_SPP_ROLE_MASTER

ELM327 myELM327;

uint32_t rpm = 0;

void bluetoothConnect();

void setup() {
  bluetoothConnect();

  // while (!myELM327.begin(ELM_PORT, true, 2000, '4')) {
  while (!myELM327.begin(ELM_PORT, true, 2000)) {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
  }

  Serial.println("Connected to ELM327");
}

void loop() {
  float tempRPM = myELM327.rpm();
  
  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: ");
    Serial.println(rpm);
  } else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}

void bluetoothConnect() {
  Serial.begin(115200);
  if (!SerialBT.begin("ESP32test", true)) {
    Serial.println("========== serialBT failed!");
    abort();
  }

  Serial.println("Starting discoverAsync...");
  BTScanResults *btDeviceList = SerialBT.getScanResults(); // maybe accessing from different threads!
  if (SerialBT.discoverAsync([](BTAdvertisedDevice *pDevice) {
        // BTAdvertisedDeviceSet*set =
        // reinterpret_cast<BTAdvertisedDeviceSet*>(pDevice);
        // btDeviceList[pDevice->getAddress()] = * set;
        Serial.printf(">>>>>>>>>>>Found a new device asynchronously: %s\n", pDevice->toString().c_str());
      })) {
    delay(BT_DISCOVER_TIME);
    Serial.print("Stopping discoverAsync... ");
    SerialBT.discoverAsyncStop();
    Serial.println("discoverAsync stopped");
    delay(1000);

    if (btDeviceList->getCount() > 0) {
      BTAddress addr;
      int channel = 0;
      Serial.println("Found devices:");
      for (int i = 0; i < btDeviceList->getCount(); i++) {
        BTAdvertisedDevice *device = btDeviceList->getDevice(i);

        Serial.printf(" ----- %s  %s %d\n", device->getAddress().toString().c_str(), device->getName().c_str(), device->getRSSI());

        std::map<int, std::string> channels = SerialBT.getChannels(device->getAddress());

        Serial.printf("scanned for services, found %d\n", channels.size());

        for (auto const &entry : channels) {
          Serial.printf("     channel %d (%s)\n", entry.first, entry.second.c_str());
        }
        if (channels.size() > 0) {
          addr    = device->getAddress();
          channel = channels.begin()->first;
        }

        if (device->getName() == "OBDII") {
          Serial.println("OBDII Founded!");
          Serial.print("Add ->");
          Serial.println(device->getAddress());
          break;
        }
      }
      Serial.print("Before connect ->");
      Serial.println(addr);
      if (addr) {
        Serial.printf("connecting to %s - %d\n", addr.toString().c_str(), channel);
        SerialBT.connect(addr, channel, sec_mask, role);
      }
    } else {
      Serial.println("Didn't find any devices");
    }
  } else {
    Serial.println("Error on discoverAsync f.e. not workin after a \"connect\"");
  }
}
,
[hondaObd2.txt](https://github.com/PowerBroker2/ELMduino/files/8841042/hondaObd2.txt)
[LogAutoProtocol.txt](https://github.com/PowerBroker2/ELMduino/files/8841033/LogAutoProtocol.txt)
[LogSettingProtocol.txt](https://github.com/PowerBroker2/ELMduino/files/8841034/LogSettingProtocol.txt)

PowerBroker2, 2022-06-06:
What year is your car? If it was made before 2008 I can't guarantee this library will work with your car. If your car was made after 2008, your ELM327 is most likely faulty and should replace it with a new one. I've had good luck with [these units](https://www.amazon.com/dp/B005NLQAHS?psc=1&ref=ppx_yo2ov_dt_b_product_details)

githubUnknownUser, 2022-06-06:
My car is from 2010, But the stranger thing is that the module works fine with phone apps. I'm using the latest version of the library. I've noticed that the bytes 0 and 3 seem to carry the information... I don't know what to do

PowerBroker2, 2022-06-06:
If different bytes carry the info, you can use lower level functions of the library to directly query the PID and use the built-in response byte objects to process the data yourself in the sketch.

closed 2022-06-06 by PowerBroker2