PowerBroker2/ELMduino #131: Multiple PIDS not working

simpleC0de, 2022-07-03:
Hey there, so I got an OLED-Display connected and I'm using an HC-05.
With a single query everything is working fine, but as soon as I implement a switch statement, as suggested in the examples directory,
nothing is printed to the display, nor is any data being printed on the serial port.

This is the code currently not working:
If code highlighting aint working: https://pastebin.com/nT0MKjV7
`
#include <SoftwareSerial.h>
#include "ELMduino.h"
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SH1106.h>
#define OLED_RESET 4
Adafruit_SH1106 display(OLED_RESET);
SoftwareSerial mySerial(2, 3); // RX, TX
#define ELM_PORT mySerial

uint32_t indx = 1;

ELM327 myELM327;

uint32_t oiltemp = 0;
uint32_t coolant = 0;
uint32_t inair = 0;

void setup()
{
  display.begin(SH1106_SWITCHCAPVCC, 0x3C); // initialize with the I2C addr 0x3D (for the 128x64)
  display.display();
  Serial.begin(9600);
  ELM_PORT.begin(9600);
  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    while (1);

  }
}

void loop()
{
  display.setCursor(0, 0);
  display.clearDisplay();
  display.setTextSize(1);
  display.setTextColor(WHITE);

  switch (indx)
  {
    case 0:
      {
        float _oiltemp = myELM327.oilTemp();

        if (myELM327.nb_rx_state == ELM_SUCCESS)
        {
          oiltemp = (uint32_t) _oiltemp;
                indx = 1;
        }
        else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
        {
          myELM327.printError();
                indx = 1;
        }
          
        break;
      }

    case 1:
      {
        float _coolant = myELM327.engineCoolantTemp();

        if (myELM327.nb_rx_state == ELM_SUCCESS)
        {
          coolant = (uint32_t) _coolant;
        indx = 2;
        }
        else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
         {
          myELM327.printError();
                indx = 2;
         }
         

        break;
      }

    case 2:
      {
        float _intakeair = myELM327.intakeAirTemp();

        if (myELM327.nb_rx_state == ELM_SUCCESS)
        {
          inair = (uint32_t) _intakeair;
                  indx = 0;
        }
        else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
          {
            myELM327.printError();
                  indx = 0;
          }

        break;
      }
      default:{
        indx = 0;
        break;
      }
  }

  

  display.print("Oil: ");
  display.println(oiltemp);
  display.print("Coolant: ");
  display.println(coolant);
  display.print("InAir: ");
  display.println(inair);
  display.display();
}
`

I'd really appreciate any kind of help or just a tip what I'm doing wrong here.
As said above, if I'm removing the switch and just doing the coolant for example, it is working and displaying all three.

Thanks..

PowerBroker2, 2022-07-03:
Getting rid of the "else if" statements might help

simpleC0de, 2022-07-03:
> Getting rid of the "else if" statements might help

I‘ll try tomorrow and give feedback, thanks for your fast reply!

closed 2022-09-22 by PowerBroker2