PowerBroker2/ELMduino #133: How to know if processPID has finished

seb30, 2022-07-17:
Hey folks,

I bought a TTGO T-Display to observe the DPF burn progress in my car. So I'm using the `processPID` command to obtain the results for my CustomPIDs (DPF soot, DPF mileage, DPF burn activity, ...). I've read, that this function must be called several times until a response was sent. But how do I know whether I got a response or I still have to wait (a valid result could be 0 as well)?

So I've changed the library as following:

from

`float processPID(const uint8_t& service, const uint16_t& pid, const uint8_t& num_responses, const uint8_t& numExpectedBytes, const float& scaleFactor = 1, const float& bias = 0);`

to

`float processPID(const uint8_t& service, const uint16_t& pid, const uint8_t& num_responses, const uint8_t& numExpectedBytes, const float& scaleFactor = 1, const float& bias = 0, bool *done = NULL);`

and the cpp

```
float ELM327::processPID(const uint8_t& service, const uint16_t& pid, const uint8_t& num_responses, const uint8_t& numExpectedBytes, const float& scaleFactor, const float& bias, bool *done)
{
	if (done) *done = false;

	if (nb_query_state == SEND_COMMAND)
	{
		queryPID(service, pid, num_responses);
		nb_query_state = WAITING_RESP;
	}
	else if (nb_query_state == WAITING_RESP)
	{
		get_response();
		if (nb_rx_state == ELM_SUCCESS)
		{
			if (done) *done = true;

			nb_query_state = SEND_COMMAND; // Reset the query state machine for next command
			return conditionResponse(findResponse(), numExpectedBytes, scaleFactor, bias);
		}
		else if (nb_rx_state != ELM_GETTING_MSG)
		{
			if (done) *done = true;

			nb_query_state = SEND_COMMAND; // Error or timeout, so reset the query state machine for next command
		}
	}
	return 0.0;
}
```

Do you think that this is a reasonable way to solve the issue?

seb30, 2022-07-17:
Ah ... I checked the [multiple PIDs example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/multiple_pids/multiple_pids.ino#L62) and there you are doing nearly the same, but outside the core function of the library.

PowerBroker2, 2022-07-17:
I would not change the library if I were you. I would suggest using the normal `processPID()` function in the same manner as the sketch you linked and everything should work. That is, assuming your arguments to `processPID()` are accurate.

PowerBroker2, 2022-07-18:
You should iteratively call `rpm = myELM327.rpm();` (or any PID processing function of your choice) as fast as possible until `myELM327.nb_rx_state == ELM_SUCCESS`. Do not use `delay()` in your code.

PowerBroker2, 2022-07-18:
I'm saying I wouldn't change the lib so it does the check for you, I'd do the check in the sketch like the example

maciekelga, 2022-07-18:
> You should iteratively call `rpm = myELM327.rpm();` (or any PID processing function of your choice) as fast as possible until `myELM327.nb_rx_state == ELM_SUCCESS`. Do not use `delay()` in your code.
> 
> Also, there is no double checking - if you get a response from the vehicle it is assumed to be accurate.

Thank You. I will use switch case to read multiple PID.

closed 2022-07-27 by seb30