PowerBroker2/ELMduino #134: Problems with connecting and ELM_TIMEOUT

AlexCobalt78, 2022-07-20:
Hello.
Please I need help. This is the code that I uploaded to my arduino Nano.

`#include <SoftwareSerial.h>
#include "ELMduino.h"
#include <FastLED.h>

#define NUM_LEDS 60
#define DATA_PIN 4
CRGB leds[NUM_LEDS];

SoftwareSerial mySerial(2, 3);
#define ELM_PORT mySerial
ELM327 myELM327;
uint32_t rpm = 0;

void setup()
{
  
  Serial.begin(38400);
  ELM_PORT.begin(38400);
  delay(200);
  myELM327.begin(mySerial);
  delay(200);
  if (!myELM327.begin(ELM_PORT, '6'))
  {
    Serial.println("Couldnt connect!");
    while (1);
  }
  Serial.println("Connection Success");
  delay(5000);
  FastLED.addLeds<WS2811, DATA_PIN, RGB>(leds, NUM_LEDS);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
    uint32_t mapRPM = map(rpm, 0, 3000, 0, 30);
    for (int i = 0; i < mapRPM; i = i + 1) {
      int sensorValue = analogRead(A0);
      int mapSensore = map(sensorValue, 0, 1023, 0, 255);
      leds[i] = CHSV(mapSensore, 255, 255);
      FastLED.show();
    }
    delay(100);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
  {
    myELM327.printError();
    Serial.println("Couldnt connect!");
    Serial.println(myELM327.nb_rx_state);
    delay(2000);
  }
}`

Im trying to connect to ELM327 with bluetooth with HC-05. I manage to connect and read a couple of lines of data and than i get this error.

`7
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
Timeout detected with overflow of 0ms
Received: 
ERROR: ELM_TIMEOUT`

Here is a whole output on serial monitor.
[Timeout.txt](https://github.com/PowerBroker2/ELMduino/files/9154631/Timeout.txt)
You can see in this file that i got 1 read for my rpm, sometimes I get couple more, but it always ends soon after with error mentioned above. I really need help with this. I see that error ELM_TIMEOUT has been mentioned in other issues, but mine is kinda different. I manage to get some data(rpm), but than it sudennly stops and I can't figure why.

PowerBroker2, 2022-07-20:
I've never seen this happen before. Perhaps your OBDII scanner is defective, maybe your car was made before 2008 (I can't guarantee cars made before then will work with this lib), and maybe your OBDII/car doesn't like how the ELM327 gets configured. I suggest posting in the Arduino forum or perhaps a forum dealing with OBDII/ELM327s specifically.

Also, I'm not sure what's going on here with the `'6'` parameter:

```C++
myELM327.begin(ELM_PORT, '6')
```

AlexCobalt78, 2022-07-20:
What would be correct command for begin function in my case? I picked this structure from one project that was using your library, but it was 2 years ago. What ERROR: ELM_TIMEOUT even means?

7
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
Timeout detected with overflow of 0ms
Received:
ERROR: ELM_TIMEOUT

Can you maybe direct me where to look for explanation of all this lines and what they mean? I should mention that I tried this code with 3 different OBDs, and every one of them has same error. They are all cheap ones from AliExpress and 2 of them are version ELM327 v1.5 and third is ELM327 v2.1 (His error output is the one I sent).

PowerBroker2, 2022-07-22:
I would suggest using `myELM327.begin(ELM_PORT, true, 5000)` unless there is a specific protocol (i.e. CAN vs K-line, etc) you'd like to use. Check out the API in the header for more info:
https://github.com/PowerBroker2/ELMduino/blob/11040a2e33b8234b0a7a70c44b96048db438ed57/src/ELMduino.cpp#L6-L26

`ERROR: ELM_TIMEOUT` means that the ELM327/OBDII scanner did not respond to the PID query at all. Basically, the OBDII scanner has gone completely silent.

The debug printouts you're referencing are fairly self explanatory provided you know a little about OBDII queries/responses.

AlexCobalt78, 2022-07-23:
I know my car is using protocol 6

[image]

 so can I than use begin command like this?
`myELM327.begin(ELM_PORT, true, 5000, '6')`
I checked it in some datasheet and confirmed after connecting to ELM327 via torque application on mobile phone.

[image]
It's strange that torque can easily read all kind of data, but when i try with my arduino nano and hc-05 it doesn't work.
Could it be that something is wrong with HC-05 module? What would be right commands for configuring HC-05 in AT mode for connecting to ELM327 module with your library? Thanks for any kind of help

PowerBroker2, 2022-07-23:
I wish I had a nickel every time someone mentions that frickin app in issues for this library. Since it's not open source I can't figure out what configuration it uses. Like I said previously, you need to ask this question in either the Arduino or OBDII online forums for better help. I can't replicate your issue with my Arduino/car pair.

closed 2022-07-23 by PowerBroker2