PowerBroker2/ELMduino #135: ESP32-S3 BLE Support?

PlastiBots, 2022-08-12:
Hi. I found that ESP32-S3's don't support Bluetooth Serial.  Any chance of getting this working using BLE?  Thx.

PowerBroker2, 2022-08-12:
I'm not familiar with BLE. If you (or anyone else) knows how to interface with ELM327s over BLE, I would be happy to roll those functions into the lib. Otherwise, I don't know how to do it.

PlastiBots, 2022-08-13:
Ok thx.

closed 2022-08-13 by PlastiBots

patfelst, 2022-08-13:
This guy has implented a BLE interface for ELM, he uses it for a gear shift light. The code is spread across many source files, so looks pretty involved to separate out the bits you'd need for this library.
https://github.com/janmannsbart23/OBDShiftIndicator

PlastiBots, 2022-08-13:
Thx. Will have a look.  I’m working with a VGate iCar Pro ODBII reader that’s BLE.  Was able to poll via an ESP32 using BluetoothSerial, but I need to use an ESP32-s3 which does not support BT over serial, so trying to figure out how to talk to the VGate using BLE and am green with my understanding of Services and Characteristics and not even sure if I can parse the VGate that way…