PowerBroker2/ELMduino #137: Error compiling for board ESP32C3 Dev Module

saagcs, 2022-08-17:
Hello, I am compiling a code that I used in an ESP32 Dev, to an ESP32 C3, but it does not allow me to compile, it shows me the following error, and I think it has to do with bluetooth:

Code:

```
//Import i2C LCD libraries
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <LiquidCrystal_I2C.h>

int sda_pin = 04; // GPIO16 as I2C SDA
int scl_pin = 05; // GPIO17 as I2C SCL

// set the LCD number of columns and rows
int lcdColumns = 16;
int lcdRows = 2;
// set LCD address, number of columns and rows
// if you don't know your display address, run an I2C scanner sketch
LiquidCrystal_I2C lcd(0x27, lcdColumns, lcdRows);

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

void setup()
{

  Wire.setPins(sda_pin, scl_pin); // Set the I2C pins before begin
  Wire.begin(); // join i2c bus (address optional for master)

#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  // initialize LCD
  lcd.init();
  // turn on LCD backlight
  lcd.backlight();

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop() 
{
  static int pidCounter = 0;
  
  switch (pidCounter)
  {
    case 0:
      {
        float motor = myELM327.engineCoolantTemp();
[... 51 lines cut ...]
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:945: undefined reference to `esp_bt_gap_start_discovery'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `.LVL458':
C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:265: undefined reference to `esp_bt_gap_set_scan_mode'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `esp_spp_cb':
C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:271: undefined reference to `esp_spp_start_srv'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:283: undefined reference to `esp_spp_disconnect'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:364: undefined reference to `esp_spp_connect'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:391: undefined reference to `esp_spp_disconnect'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `.L0 ':
C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:126: undefined reference to `esp_bt_gap_resolve_eir_data'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `get_name_from_eir':
C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:130: undefined reference to `esp_bt_gap_resolve_eir_data'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `.L0 ':
C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:456: undefined reference to `esp_bt_gap_cancel_discovery'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: libraries\BluetoothSerial\BluetoothSerial.cpp.o: in function `esp_bt_gap_cb':
C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:455: undefined reference to `esp_spp_start_discovery'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:522: undefined reference to `esp_bt_gap_pin_reply'
c:/users/saagc/appdata/local/arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\saagc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\BluetoothSerial\src/BluetoothSerial.cpp:530: undefined reference to `esp_bt_gap_ssp_confirm_reply'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32C3 Dev Module.

```

I can hardly find examples to use libraries in the ESP32 C3, thanks for the help

PowerBroker2, 2022-08-17:
Looks like a bug in BluetoothSerial.h

closed 2022-08-17 by PowerBroker2

patfelst, 2022-08-17:
The ESP32-C3 doesn't have bluetooth classic, only BLE.

saagcs, 2022-08-17:
Hello, is there an example with BLE?