PowerBroker2/ELMduino #140: Speed up OBDII connection and non blocking

filmchaser1, 2022-08-22:
Hi, Great work so far...I have and esp32 using your latest updated library, it connects to the ELM327 but takes about 14 seconds. And it blocks any other code from running. If no BLE is available it blocks for almost 2 minutes before code can run. 

Ideally, this would be a non blocking attempt to connect so I can run other code in the meantime. If that is not possible, then speeding up the connection time and reducing the timeout would help.
1.  Can this connection process be written non blocking? 
2. I thought about commenting out a few protocols, but is there a better way to make the bluetooth "connected" result faster with a known and previously connected device?
3. Where is the "timeout" code for when it cannot find a device? I saw it in the .cpp file but did not see where the timeframe variable or how many attempts it makes was set. How can I set a shorter connect attempt time before it moves on to my other code?
4. Why are you using while(1); in the setup that seems to indicate an infinite loop? Did you originally design this so that nothing works until it connects to the OBDII? I would imagine alot of people wanting to use this would like a state for example
if (no_device_present == true) {
//do.this.instead
}

Thank you.

PowerBroker2, 2022-08-23:
**Note** BLE is not currently supported by this library

1. Maybe, but I doubt it
2. I'm not aware of any configuration settings that speed up BT connects
3. The header file has default parameters for all functions listed in the function declarations. The declaration for `begin()` has the `timeout` parameter, which is what you're looking for: https://github.com/PowerBroker2/ELMduino/blob/fb24b93bee86b2d299dec27883f70587a1e2d01f/src/ELMduino.h#L310 The timeout is then set as a class attribute in `begin()` and subsequently used throughout the class functions
4. For my examples, if you can't connect to the ELM327, there's no point in continuing the example, so I put a halt command in. In a real application, you might want to do things differently

patfelst, 2022-08-23:
For point (2), assuming you're using bluetooth classic (not BLE), Specifying your ELM's bluetooth MAC address instead of advertised name (e.g. "OBDII") significantly reduces connection time. When you specify the advertised name, the ESP32 does a bluetooth scan then finds the one with the OBDII name and connects to that (even if you've previously paired it). You can see this if you turn on ESP32 verbose logging.

filmchaser1, 2022-08-26:
Thanks for the replies. I changed the question to say Bluetooth instead of BLE, to avoid confusion. The questions are still applicable.
1. My elm327 bricked so I had to get another one. I tried my code with the new one and it works fine. BUT changing ("OBDII") to ("00:...mac address") does not connect. I have tried with pin commented out and left uncommented. I do know I have the correct mac address for the new elm327. I must be missing something?
2. I noticed today that the latest version of code here is different than the Arduino library. Are there significant changes? 

I didn't realize when asking my questions that we were talking about two different example code. Thanks.

PowerBroker2, 2022-08-26:
My Libraries Manager is showing v3.0.2 which is the latest. Changes from v3.0.1 to latest is [here](https://github.com/PowerBroker2/ELMduino/compare/3.0.2...master)

filmchaser1, 2022-08-26:
I have 3.02 installed, however, the example file from that library install looks different than the example you have on the page here https://github.com/PowerBroker2/ELMduino also, multiple PIDS example on here is not in the arduino library version, and has a different setup and has  so I wasn't sure which was the latest. 

In any case, am I missing something with replacing the "OBDII" with "mac address"? It's not connecting, so I'm curious why not. I didn't see any other instructions for connecting via mac address. I'm using the colon format not dashes.

PowerBroker2, 2022-08-26:
Interesting, not sure why your version doesn't have the correct updates.

I've never used the MAC address to connect - are you sure you have the correct address? Maybe [this](https://techtutorialsx.com/2018/03/09/esp32-arduino-getting-the-bluetooth-device-address/) will be of some use

filmchaser1, 2022-08-26:
Not sure either. It's same on both my mac and PC, I installed the library on both last week.
Thanks, I'm confident it is the right mac address, I looked at it on my laptop to see what the mac of advertised name OBDII was. I also verified that the esp32 was advertising as ArduHUD per the example sketch.

patfelst, 2022-08-26:
Your mac address needs to be in an array, have a look at this example in the espressif bluetooth classic library 
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino

Essentially
```c++
uint8_t address[6]  = {0xAA, 0xBB, 0xCC, 0x11, 0x22, 0x33}; // For mac address "AA:BB:CC:11:22:33"
```
and then
```c++
  // connect(address) is fast (upto 10 secs max), connect(name) is slow (upto 30 secs max) as it needs
  // to resolve name to address first, but it allows to connect to different devices with the same name.
  // Set CoreDebugLevel to Info to view devices bluetooth address and device names
  connected = SerialBT.connect(address);
```