PowerBroker2/ELMduino #141: ISO_15765_11_BIT_500_KBAUD - findResponse() function not parses correctly the buffered data

AlexMichno93, 2022-09-03:
Hello all,

I am strugling for a while to read correctly the response recived by the car using ISO_15765_11_BIT_500_KBAUD.
Previously I was using ISO_9141_5_BAUD_INIT  on another car and everythink was perfect.
With ISO 15765 I got always 0 for each request. 

Logs below:

Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: \r
	Received char: 4
	Received char: 1
	Received char: _
	Received char: 0
	Received char: C
	Received char: _
	Received char: 0
	Received char: B
	Received char: _
	Received char: B
	Received char: 1
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 010C1410C0BB1000000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0

rpm = 0

You could see that chars are received but latter final response is always 0. I was playing a bit with findResponse() function but I gave up. 
Do You have any idea what could be the issue?

Thank You.
Alex.

PowerBroker2, 2022-09-03:
Hmm, looks like your ELM327 is responding with too many chars than expected. I'll have to look into this and see if there's a straightforward fix

AlexMichno93, 2022-09-04:
Allright, thanks! I am looking forward for any hint to help me resolve it :)

PowerBroker2, 2022-09-04:
I updated `conditionResponse()` so that if the ELM327 responds with extra bytes, the function tests to see if the response has trailing or padding zeros. Based on that test, it should respond with the correct data (thus fixing your issue):

https://github.com/PowerBroker2/ELMduino/blob/7cd508077cb56796f7c78c4fc526c1de36d9ecfe/src/ELMduino.cpp#L358-L406

This update is on the main branch, can you verify it fixes things for you before I tag out a new version? Thanks!

AlexMichno93, 2022-09-05:
Yeah, absolutly. Will test it tmrw and let You know. Thanks

PowerBroker2, 2022-09-05:
Now that I think about it, since the ELM327 responds with more hex chars than can fit in a uint64_t, I don't think this fix works. I'll have to try a different approach...

AlexMichno93, 2022-09-07:
I had opportunity to verify last solution but it seems it is still not working.
Please see log below:
Service: 1
PID: 4
Normal length query detected
Query string: 01041
Clearing input serial buffer
Sending the following command/query: 01041
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: 4
	Received char: 1
	Received char: \r
	Received char: 4
	Received char: 1
	Received char: _
	Received char: 0
	Received char: 4
	Received char: _
	Received char: 7
	Received char: 1
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: 0
	Received char: 0
	Received char: _
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 0104141047100000000
Expected response header: 4104
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
engineLoad()
0

I also checked other PIDs but all seems to be the same.

PowerBroker2, 2022-09-22:
I think I may have finally fixed the issue here with my recent update to the main branch. Could you give it another test to verify? Thanks!

CraigH78, 2023-01-21:
I'm having a similar problem with the same protocol, RPM always reads 884.  Here is what I can get from the serial panel, for some reason I can't select all and copy, only copies whats visible.

Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: A
	Received char: 9
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410C1A90
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 144
	responseByte_1: 26
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 884
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: 9
	Received char: D
	Received char: C
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410C19DC
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 220
	responseByte_1: 25
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 884
Service: 1
PID: 12
Normal length query detected

PowerBroker2, 2023-01-24:
@CraigH78 Did you test the main branch? It's different than the latest release

CraigH78, 2023-01-24:
> @CraigH78 Did you test the main branch? It's different than the latest release

I downloaded the zip from the main page (https://github.com/PowerBroker2/ELMduino).

alkonosst, 2023-01-27:
Hello @PowerBroker2 , hope you are doing great. I have the same issue than @CraigH78: RPM and all other queries are returning bad values and also don't change at all (_RPM doesn't change even when I'm pushing the gas pedal_).

I'm using **ESP32** and **main branch** (last commit).

I modified `ELMduino.cpp` to put some logs:

- `findResponse()`:
```.cpp
response = 0;
for (uint8_t i = 0; i < numPayChars; i++)
{
	uint8_t payloadIndex = firstDatum + i;
	uint8_t bitsOffset = 4 * (numPayChars - i - 1);
	response = response | (ctoi(payload[payloadIndex]) << bitsOffset);
}

log_d("response: %08X", (uint32_t)response);
```

- `conditionResponse()`:
```.cpp
uint64_t leadingResponse = 0;
for (uint8_t i = 0; i < numExpectedBytes; i++)
{
	uint8_t payloadIndex = PAYLOAD_LEN - numPayChars + i;
	uint8_t bitsOffset   = 4 * (numExpectedBytes - i - 1);

	leadingResponse |= (ctoi(payload[payloadIndex]) << bitsOffset);
}

log_d("leadingResponse: %08X", (uint32_t)leadingResponse);

uint64_t laggingResponse = 0;
for (uint8_t i = 0; i < numExpectedBytes; i++)
{
	uint8_t payloadIndex = PAYLOAD_LEN - numExpectedBytes + i;
	uint8_t bitsOffset   = 4 * (numExpectedBytes - i - 1);

	laggingResponse |= (ctoi(payload[payloadIndex]) << bitsOffset);
}

log_d("laggingResponse: %08X", (uint32_t)laggingResponse);

if (leadingResponse > laggingResponse) {
	float leading = ((float)leadingResponse * scaleFactor) + bias;
	log_d("leading response: %08X - %.2f", leading, leading);
	return leading;
}

float lagging = ((float)laggingResponse * scaleFactor) + bias;
log_d("lagging response: %08X - %.2f", lagging, lagging);
return lagging;
```

To read the RPM value, I'm using `rpm()` method in my main code:
```.cpp
case OBD2State::RPM:
{
  float rpm = obd2.rpm();

  if (obd2.nb_rx_state == ELM_SUCCESS) {
	log_i("RPM: %.0f rpm", rpm);
	obd2_state = OBD2State::STANDBY;
  }

  else if (obd2.nb_rx_state != ELM_GETTING_MSG) {
	obd2.printError();
	obd2_state = OBD2State::STANDBY;
  }
} break;
```

My terminal looks like this:
```bash
[ 10231][I][main.cpp:88] usbRead(): -> r
[ 10231][D][main.cpp:142] usbRead(): Querying RPM
Service: 1
PID: 12
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
        Received char: S
        Received char: E
        Received char: A
        Received char: R
        Received char: C
        Received char: H
        Received char: I
        Received char: N
        Received char: G
        Received char: .
        Received char: .
        Received char: .
        Received char: \r
        Received char: 4
        Received char: 1
        Received char: 0
        Received char: C
        Received char: 0
        Received char: D
        Received char: 2
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: SEARCHING...410C0D2D
Expected response header: 410C
Single response detected
[ 13547][D][ELMduino.cpp:2571] findResponse(): response: 00000D2D
64-bit response:
        responseByte_0: 45
        responseByte_1: 13
        responseByte_2: 0
        responseByte_3: 0
        responseByte_4: 0
        responseByte_5: 0
        responseByte_6: 0
        responseByte_7: 0
[ 13571][D][ELMduino.cpp:414] conditionResponse(): leadingResponse: 00000DD0
[ 13573][D][ELMduino.cpp:425] conditionResponse(): laggingResponse: 00000DD0
[ 13581][D][ELMduino.cpp:434] conditionResponse(): lagging response: 00000000 - 884.00      
[ 13589][I][main.cpp:254] processOBD2(): RPM: 884 rpm
```

As you can see, the variable `response` has the correct data (`0D2D`), but `laggingResponse` doesn't. I tested by pushing the gas pedal to different RPM levels, like 1000, 2000 and 3000, and the variable `response` always is correct. But `laggingResponse` (_variable used as `rpm()` return value_) is never correct.

I'll continue testing. Maybe this information helps you.

alkonosst, 2023-01-27:
@PowerBroker2 I changed this line:
https://github.com/PowerBroker2/ELMduino/blob/eabbb1a2e8e82d8fed1b758c0d893bc46efbe171/src/ELMduino.cpp#L393

to:
```cpp
else if ((numExpectedBytes * 2) == numPayChars) {
```

And now it's working. That's because, as I understand, the variable `numPayChars` represents a nibble. Example:
- `response`: 410C0D41
- `recBytes`: 8 (410C0D41)
- `numPayChars`: 4 (0D41)

So, when `rpm()` is called, the function `processPID()` would need the parameter `numExpectedBytes` to be 4.

There is another problem. When I try to use functions `supportedPIDs_xx_xx()`, there is a bad cast (_float to uint32_t_), which gives me a bad number. See my terminal (_additional logs inserted_):

```bash
[229977][I][main.cpp:88] usbRead(): -> 1
[229977][D][main.cpp:118] usbRead(): Querying Supported PIDs 1-20
Service: 1
PID: 0
Normal length query detected
Query string: 01001
Clearing input serial buffer
Sending the following command/query: 01001
        Received char: 4
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: B
        Received char: E
        Received char: 3
        Received char: E
        Received char: B
        Received char: 8
        Received char: 1
        Received char: 1
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 4100BE3EB811
Expected response header: 4100
[230124][D][ELMduino.cpp:2538] findResponse(): firstHeadIndex: 0
[230129][D][ELMduino.cpp:2539] findResponse(): secondHeadIndex: 4294967295
[230137][D][ELMduino.cpp:2548] findResponse(): firstDatum: 4
Single response detected
[230154][D][ELMduino.cpp:2569] findResponse(): recBytes: 12
[230154][D][ELMduino.cpp:2570] findResponse(): numPayChars: 8
[230158][D][ELMduino.cpp:2579] findResponse(): p_idx: 4 - offset: 28 - val: B
[230166][D][ELMduino.cpp:2579] findResponse(): p_idx: 5 - offset: 24 - val: E
[230174][D][ELMduino.cpp:2579] findResponse(): p_idx: 6 - offset: 20 - val: 3
[230182][D][ELMduino.cpp:2579] findResponse(): p_idx: 7 - offset: 16 - val: E
[230189][D][ELMduino.cpp:2579] findResponse(): p_idx: 8 - offset: 12 - val: B
[230197][D][ELMduino.cpp:2579] findResponse(): p_idx: 9 - offset: 8 - val: 8
[230205][D][ELMduino.cpp:2579] findResponse(): p_idx: 10 - offset: 4 - val: 1
[230213][D][ELMduino.cpp:2579] findResponse(): p_idx: 11 - offset: 0 - val: 1
[230221][D][ELMduino.cpp:2582] findResponse(): response: BE3EB811
64-bit response: 
        responseByte_0: 17
        responseByte_1: 184
        responseByte_2: 62
        responseByte_3: 190
        responseByte_4: 255
        responseByte_5: 255
        responseByte_6: 255
        responseByte_7: 255
[230248][D][ELMduino.cpp:395] conditionResponse(): numExpectedBytes == numPayChars: 00000000 - 1844674407376.00
[230255][I][main.cpp:186] processOBD2(): Supported PIDs 01-20: 0xFFFFFFFF
```

PowerBroker2, 2023-01-27:
@alkonosst, thanks for the fix, I was not looking forward to relearning how the lib works to address this issue again, lol. As for the `supportedPIDs_xx_xx()` error - what is the expected output in your case? (I've never actually used this functionality myself)

alkonosst, 2023-01-27:
@PowerBroker2 in the last example, `supportedPIDs_01_20()` should return `BE3EB811` (_that's the 'response' variable value_), which contains the bit mask for the first 32 PID values. But actually returns `FFFFFFFF`. I think it is because you are casting a `float` to `uint32_t`, and the value changes because of that conversion:
https://github.com/PowerBroker2/ELMduino/blob/eabbb1a2e8e82d8fed1b758c0d893bc46efbe171/src/ELMduino.cpp#L581-L584

So far I've read `float` PIDs without issue, like `rpm()`, `fuelLevel()` and `throttle()`.

CraigH78, 2023-01-28:
@alkonosst & @PowerBroker2 - Thank you both, that did the trick!

ruffoAraujo, 2023-04-20:
@PowerBroker2 Hello all,

I'm having the same problem as @CraigH78, I made the bluetooth connection with my OBDII ELM327, I put the protocol used by my car, and I only get a rpm measurement of 884. So, I tried other protocols and, some I didn't get measurements and others I got the same result for rpm. I'm using the latest version of the library (3.0.2). I'm a student and I'm doing a final project in engineering, and I'm developing this project and I grabbed that part. From what I understood in the previous comments, this problem is in the library and I couldn't fix it. Can you help me please?

I'm uploading my full code here, thanks!

#include <ELMduino.h>
#include <BluetoothSerial.h>

// esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE; 
// esp_spp_role_t role = ESP_SPP_ROLE_SLAVE;

BluetoothSerial SerialBT;
ELM327 myELM327;
//MAC = "00:10:cc:4f:36:03";
uint8_t MAC[6]  = {0x00, 0x10, 0xcc, 0x4f, 0x36, 0x03};
bool connectOBD;
int channel = 0;
uint32_t rpm = 0;

void setup() 
{
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  SerialBT.begin("ESP32", true);    
  //connectOBD = SerialBT.connect(MAC, channel, sec_mask, role);
  connectOBD = SerialBT.connect(MAC);
  if (connectOBD)
  {
    Serial.println("OBD - Phase 1");
    if (myELM327.begin(SerialBT,false,2000,ISO_9141_5_BAUD_INIT))
    {
      Serial.println("OBD - Phase 2");
      digitalWrite(LED_BUILTIN, HIGH);
    }
  }     
}

void loop()
{
  float tempRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}

PowerBroker2, 2023-04-20:
Try using what's on the main branch instead of the release

ruffoAraujo, 2023-04-20:
@PowerBroker2 Sorry but I didn't understand what you meant by that, could you explain? do you think the problem is in my code?

ruffoAraujo, 2023-04-20:
@PowerBroker2 I was able to understand what you said, and my answer is yes, I downloaded the library directly from the github site, from the main branch. I redid the entire library installation process and did the test again today and the rpm is still 884.
My monitor serial is:

Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410C0000
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 884

ruffoAraujo, 2023-04-22:
@PowerBroker2 Hello all, I managed to solve the problem. The file ELMduino.cpp that is in the main branch is without the changes made by @alkonosst. So I made the same changes he made and it worked. Thanks!

Black-Adam-jr, 2023-04-23:
Hello guys. Hope everybody doing great! 
The 
      `else if ((numExpectedBytes * 2) == numPayChars) {`

solves the problem for me if I want to get just one pid information. RPM or KPH in my case.

The code is below.
```
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>
#include "EEPROM.h"
TFT_eSPI tft = TFT_eSPI(); 

//#define USE_NAME
const char *pin = "1234";
#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif
BluetoothSerial SerialBT;
ELM327 myELM327;
int displayNum=0;
#define ELM_PORT SerialBT
uint32_t targetTime = 0; 
uint16_t PAYLOAD_LEN;

//int PAYLOAD_LEN;
uint32_t rpm = 0;
uint32_t kph = 0;
uint32_t ECT = -1;
uint32_t oil = -1;
float fuelLevel = 0; // in %
  //float tempBATTERY = myELM327.queryPID(SERVICE_01,CONTROL_MODULE_VOLTAGE);
  //float tempCOOLANT = myELM327.queryPID(SERVICE_01,ENGINE_COOLANT_TEMP);
  //float tempINTAKE = myELM327.queryPID(SERVICE_01,INTAKE_AIR_TEMP);
  //float tempOIL = myELM327.queryPID(SERVICE_01,ENGINE_OIL_TEMP);
  //float tempLOAD= myELM327.queryPID(SERVICE_01,ENGINE_LOAD);

//void(* resetFunc) (void) = 0;

#ifdef USE_NAME
  String slaveName = "OBDII"; 
#else
  String MACadd = "66:1e:31:f3:c1:d5"; // This only for printing
  //uint8_t address[6]  = {0x75, 0x57, 0x14, 0x21, 0x04, 0xf6};
  //uint8_t address[6]  = {0x66, 0x1e, 0x31, 0xf3, 0xc1, 0xd5};
  ////uint8_t address[6]  = {0x00, 0x10, 0xcc, 0x4f, 0x36, 0x03};
  uint8_t address[6]  = {0x00, 0x1d, 0xa5, 0x69, 0x45, 0x45};
  //uint8_t address[6]  = {0x87, 0x78, 0x0f, 0x05, 0x91, 0xdb}; 
  //uint8_t address[6]  = {0x39, 0xd3, 0xa4, 0x9c, 0xea, 0x57};
#endif

String myName = "ESP32-BT-Master";

void setup()
{
bool connected;
  tft.init();
   tft.setTextColor(random(0xFFFF));
   tft.drawString("Connecting to OBD-II",  10, 10, 4);
   tft.setTextFont(4);
   tft.setTextColor(TFT_WHITE, TFT_BLACK);
   
   tft.setRotation(1);
   tft.fillScreen(TFT_BLACK);

   

   targetTime = millis() + 1000;
[... 522 lines cut ...]
18:20:53.732 -> 	Received char: >
18:20:53.732 -> Delimiter found.
18:20:53.732 -> All chars received: 410C0BDC
18:20:53.732 -> Expected response header: 410C
18:20:53.732 -> Single response detected
18:20:53.732 -> 64-bit response: 
18:20:53.732 -> 	responseByte_0: 220
18:20:53.732 -> 	responseByte_1: 11
18:20:53.732 -> 	responseByte_2: 0
18:20:53.732 -> 	responseByte_3: 0
18:20:53.732 -> 	responseByte_4: 0
18:20:53.732 -> 	responseByte_5: 0
18:20:53.732 -> 	responseByte_6: 0
18:20:53.732 -> 	responseByte_7: 0
18:20:53.732 -> Speed: 208
18:20:53.764 -> Service: 1
18:20:53.764 -> PID: 12
18:20:53.764 -> Normal length query detected
18:20:53.764 -> Query string: 010C1
18:20:53.764 -> Clearing input serial buffer
18:20:53.764 -> Sending the following command/query: 010C1
```

Any advice please. Maybe somthing wrong with code getting pid information sequence.
Thank you.

Black-Adam-jr, 2023-04-24:
.....UPDATE......
I somehow solved the previous problem by editing my loop.
```
void loop()
{
  
  switch (obd_state)
  {
    case ENG_RPM:
    {
      rpm = myELM327.rpm();
      
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        Serial.print("rpm: ");
        Serial.println(rpm);
        tft.setCursor(5, 60); 
        tft.print("RPM: "); tft.print(rpm);
        obd_state = SPEED;
      }
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {
        myELM327.printError();
        obd_state = SPEED;
      }
      
      break;
    }
    
    case SPEED:
    {
      kph = myELM327.kph();
      
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        Serial.print("kph: ");
        Serial.println(kph);
        tft.setCursor(5, 30); 
        tft.print("SPEED: "); tft.print(kph);
        tft.setCursor(150, 30);
        tft.println(" kph ");
        
        obd_state = ENG_RPM;
      }
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {
        myELM327.printError();
        obd_state = ENG_RPM;
      }
      
      break;
    }
    
  }
} 
```
Almost everything working good, but the next problem is getting wrong header than was expected about every 5-10 seconds.
And rpm returns to 884 and kph returning 884.

```
11:45:53.034 -> Normal length query detected
11:45:53.034 -> Query string: 010D1
11:45:53.034 -> Clearing input serial buffer
11:45:53.034 -> Sending the following command/query: 010D1
11:45:53.068 -> 	Received char: 4
11:45:53.068 -> 	Received char: 1
11:45:53.068 -> 	Received char: 0
11:45:53.068 -> 	Received char: C
11:45:53.068 -> 	Received char: 3
11:45:53.068 -> 	Received char: 4
11:45:53.068 -> 	Received char: 2
11:45:53.068 -> 	Received char: 0
11:45:53.068 -> 	Received char: \r
11:45:53.068 -> 	Received char: \r
11:45:53.068 -> 	Received char: >
11:45:53.068 -> Delimiter found.
11:45:53.068 -> All chars received: 410C3420
11:45:53.068 -> Expected response header: 410D
11:45:53.068 -> Response not detected
11:45:53.068 -> kph: 208
11:45:53.068 -> Service: 1
11:45:53.068 -> PID: 12
```

ruffoAraujo, 2023-05-19:
Hello guys. Hope everybody doing great!
The
else if ((numExpectedBytes * 2) == numPayChars) solve my problem to.

But now, i discovered other other problems for my case. My car uses the '3' protocol, and how long to use other functions unfortunately the return is ELM_NO_DATA.

I'm sending all the data I'm able to read:

void loop()
{
  switch(decisao)
  {
    case 1:      // Velocidade (km/h)
    {
      valorLido = myELM327.kph();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        kph = valorLido;
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 2:      // Posição do acelerador em %
    {
      valorLido = myELM327.throttle();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        throttle = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 3:      // Temperatura do ar de admissão em C
    {
      valorLido = myELM327.intakeAirTemp();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        intakeAirTemp = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 4:      // Temperatura atual do líquido de arrefecimento
    {
      valorLido = myELM327.engineCoolantTemp();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        engineCoolantTemp = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 5:      // RPM
    {
      valorLido = myELM327.rpm();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        rpm = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 6:      // Voltagem da bateria 
    {
      valorLido = myELM327.batteryVoltage();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        batteryVoltage = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 7:      // Carga do motor
    {
      valorLido = myELM327.engineLoad();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        engineLoad = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 8: // Encontra a porcentagem de compensação de combustivel
    {
      valorLido = myELM327.longTermFuelTrimBank_1();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        shortTermFuelTrimBank_1 = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 9: // Pressão de combustivel
    {
      valorLido = myELM327.manifoldPressure();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao++;
        reset = 0;
        if (valorLido != 0)
        {
        manifoldPressure = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
    case 10: // Avanço de temporização em graus antes do ponto morto superior (TDC)
    {
      valorLido = myELM327.timingAdvance();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        decisao = 1;
        reset = 0;
        if (valorLido != 0)
        {
        timingAdvance = valorLido;
        }
      }
      else if (myELM327.nb_rx_state == ELM_NO_DATA)
        reset++;
      break;
    }
  }
}

Does anyone have any idea how I can read more data from my vehicle?

PowerBroker2, 2023-05-19:
`NO_DATA` means that the car's diagnostics computer doesn't have data for that PID at that protocol. Maybe try using the default library settings or using a different protocol?

ruffoAraujo, 2023-07-16:
I'm using the default library settings, protocol 3. I don't know why I couldn't read more data, but for the project I'm doing, the data I could read is enough. So thanks a lot for the help guys

commit eabbb1a2 (2022-09-22): Second attempt to fix #141
closed 2023-12-21 by PowerBroker2