PowerBroker2/ELMduino #142: How to connect to wired ELM OBD reader to arduino

NickSantini32, 2022-09-23:
Hello,

This may be a rudimentary question but I am struggling to find any documentation on how to solder a wired reader to an Arduino such that it would function with this library. I assume we would set our tx and rx pins and use software serial, but again, I'm not sure how to wire up the reader to the board. 

I know the author mentioned it in [this](https://forum.arduino.cc/t/car-heads-up-display/576252/3) build thread but I don't think he ever mentioned exactly how it was wired or why he switched to Bluetooth other than convenience. Also I noticed that he used fuses on his wires from to the reader to the board, if I use a wired reader will I need to use these and if so, where?

For some extra background info, I want to use a wired OBD reader because I plan to pair my Arduino with my phone such that I can access the data via an app. I read that the Bluetooth protocol only supports one device at a time hence the necessity for a wired reader. 

Thank you for any info helping me with this!

NickSantini32, 2022-09-23:
Just an idle thought, could the best solution be to buy an OBD to USB reader and then just solder the 4 USB wires to the board? (although not sure which would be rx and which would be tx)

PowerBroker2, 2022-09-23:
BLUF: Yes, only two devices can talk over a single bluetooth connection, however, you can add another bluetooth module to communicate separately with the ELM327. You'll have two bluetooth connections open at the same time. I'd suggest this approach.

When I originally started my HUD project a couple of years ago I tried to plug into the vehicle CAN port directly via a wired solution and quickly learned this was a bad idea. Instead I decided to use an ELM327 to do the CAN interface for me so I can simply grab the bluetooth data and process it for the HUD. That's why I switched to wireless.

If you really, really want to do wired: It's hard to find a wired ELM327 module, but if you use the schematic in the datasheet, you should be able to connect the the RS232 port available on pins 17 and 18. Note that this is NOT a UART port, but a RS-232 one, you'll probably need an interface chip (i.e. a MAX232) between your Arduino and ELM327.

PowerBroker2, 2022-09-23:
> Just an idle thought, could the best solution be to buy an OBD to USB reader and then just solder the 4 USB wires to the board? (although not sure which wires would be rx and which would be tx)

You'll need a USB host controller (very tricky to use - I've tried) if you use a USB OBD2 scanner

NickSantini32, 2022-09-23:
Thank you so much for the quick reply! I didn't realize that you could just have two Bluetooth modules to connect to two devices. Based on the experience you shared that sounds like the only option that is remotely within the scope of what I am capable of doing.

Thank you again, I look forward to using your library for my next project!

closed 2022-09-23 by NickSantini32