PowerBroker2/ELMduino #143: ESP32 connection issues

enslinm123, 2022-09-28:
Hey PowerBroker,

First off, it's an amazing feat of a project you've taken on making interfacing with a vehicles OBD easier for the noobs like myself.

Onto the issue. So I acquired one of those el'cheapo chinese bluetooth elm327's and it's working great with the torque android app. My initial thoughts were to identify the RX and TX pins of the bluetooth chip and intercept them with an Arduino Nano, make a 3D Printed enclosure and install it permanently into the car instead of looking at my mobile the whole time.

Did some tinkering and eventually gave up after a few months where I'm not able to identify the RX and TX pins with my logic analyzer. Mostly due to my clamps being too large and I cannot identify the actual BT chip or find smaller probes locally in my country (I am suspecting it's communicating over some sort of SPI interface). 

Anyways down to the issue at hand, I decided to get myself an ESP32 Dev Kit V1 board and I've been trying to just get it connected to the ELM327 to display the rpm or engine coolant temperature (want to try both respectively before moving over to a multi PID setup and displaying it on either a OLED or LCD screen)

So, when I upload the ESP32_test sketch I am able to successfully connect using the MAC and pin (mine prompts for the pin when I connect with my android device). All the AT commands that I've run return values in HEX, for instance AT RV returns the current battery voltage so there is definitely communication between my ESP32, ELM327 and the car. 

Here's the funniest part, the moment I upload the ESP32_Bluetooth_Serial sketch, all I see are timeouts and unable to connect messages so I'm at a loss for words and already pulled out a lot of hair, lol.

When I upload the paired devices sketch I can see the paired device. If I upload the ESP32_test sketch it connects successfully again. I've even removed it from the paired devices list, paired it again with the ESP32_test sketch, ran a couple AT commands to verify the connection, removed the pairing again, tried pairing by means of the ESP32_Bluetooth_Serial sketch and get the same errors. When it's paired via the ESP32_test sketch and I run the ESP32_Bluetooth_Serial sketch, the same errors come up.

When I set my logging to Information on the Arduino IDE, I can see all other bluetooth devices in the vicinity however it will not pair with the ELM327.

Am I missing something?

PowerBroker2, 2022-09-28:
I've never had to use the MAC address or pin on my ESP32s to connect, so I'm not sure if you missed anything there. However, based on your other debugging experiences, I doubt this is the issue.

Can you post (with proper markdown tags) the debug printouts?

enslinm123, 2022-09-30:
Hey PB,

Thank you for the response and apologies for only responding now.

Typical Murphy situation below, lol

Below is the terminal output when I uploaded the ESP32_test sketch only commenting out the following two lines to let the sketch compile and upload:
```
pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH);
```
Terminal output with a few AT commands after a manual reset:
[image]

This morning, I uploaded my BT Clear sketch (pasted near the bottom of this post) containing the "#define REMOVE_BONDED_DEVICES 1" to remove all pairings, afterwards I uploaded the ESP32_Bluetooth_Serial sketch (only changin the BT name of the ESP32 to "ESP32test" instead of "ArduHUD") and below is my terminal output when powering my ELM327 on the bench with GND (chassis ground on the ODB Port) and 5v (basically just to power it's Bluetooth chip) having it connect on "OBDII" without the pin and MAC address, but it seems fine now for some reason. I received connection errors on the bench as well as when connected to the car leaving the sketch default as well as change it to connect to the MAC address, previously I was not seeing any "OK" returns from the ELM327 in the terminal, not even a ELM327 version response. 

I am suspecting now that changing the ESP32 BT Name as per below is what resolved my connection issue:
```
  //ELM_PORT.begin("ArduHUD", true);
  ELM_PORT.begin("ESP32test", true);
```

Terminal output below from bench powered ELM327 (The ESP32 detects my neighbor's Xoami MiBox, lol)
Just a side note, the detected MAC Address: "aa:bb:cc:11:22:33" is the correct MAC for my ELM327, this is where one can see the 10 second delay in connecting to the BT Name instead of the Address.

Terminal Debug output:
```[     3][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   498][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   498][I][BluetoothSerial.cpp:685] _init_bt(): device name set
[   500][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
[   508][I][BluetoothSerial.cpp:938] connect(): master : remoteName
[   509][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   527][I][BluetoothSerial.cpp:507] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT started
[  3295][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  3295][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  3299][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  3307][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 196
[  3345][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  3345][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  3349][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  3357][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 194
[  3366][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  3372][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  3380][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  3387][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 186
[  3627][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  3627][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  3631][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  3639][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 189
[  4126][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4126][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  4130][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  4138][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 185
[  4254][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4254][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: 6c:0d:c4:06:3b:c4
[  4258][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x280424
[  4266][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 160
[  4273][I][BluetoothSerial.cpp:435] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : MIBOX4 nown : 11
[  4556][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4556][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  4560][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  4568][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 195
[  4854][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4854][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: 6c:0d:c4:06:3b:c4
[  4859][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x280424
[  4866][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 166
[  4874][I][BluetoothSerial.cpp:435] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : MIBOX4 nown : 11
[  5126][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  5126][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[... cut ...]
All that is left now is to go and connect it up to the car and hope for the best, I will report back once I have connected it to the car with some more terminal debug outputs, thank you for trying to assist me with my random issue that went away mysteriously.

Just a random question, which displays have you successfully been able to get working with your ESP32's ?
I have a Waveshare 1.5" RGB OLED I want to use and plan on using the Adafruit SSD1351 Library, hoping that will work lol

Regards,
Martin

enslinm123, 2022-09-30:
So a quick update. 

Managed to get a quick gap and tried connecting to the car, it connected successfully without any issues, I can kick myself for not capturing the terminal output, but was in such a rush to test it quickly before my boss noticed I was gone 😝 and for my own sanity sake. I even power cycled and it connected again and again without fail, I'm amazed that it just keeps on working now 👏 

Also managed to connect my TTP223 and 1.5" 65K Display over SPI successfully and able to display some text on it. Had some issues with the Adafruit library so I found the SSD13XX library from Sumotoy [here on github](https://github.com/sumotoy/SSD_13XX) 👌 it works great and is super easy as well. My next challenge will be to port my sketch to a larger 2" IPS LCD that I have coming in the mail.

Thank you for the help PB, no idea what resolved my issue but hey I'm not complaining 😄 💪 

Just sorting out a bug in a switch statement I want to use for multiple pages/screens where I have a summary page/screen with about 4 PID's actively displaying, then use the TTP223 to flip between pages/screens for a more detailed view of each PID and round robin back to the summary page.

enslinm123, 2022-09-30:
The celebrations were short lived 🤣 

I am able to successfully connect to my car without problems (my car uses ISO 9141-2 or AT SP3, auto tends to cause problems for some reason) and executes the AT commands from the library successfully every single time.

So I went ahead and coded the PID's that I want to display on my OLED screen with a plain ol "switch" statement in my main loop where I then call the switch cases defined below the main loop and made a quick summary page displaying 4 PID's (Will post a picture in the next post as it's on my mobile), but this is where I am stuck now, more on that later...

The TTP223 works fantastic going through the pages, had a slight hiccup on the terminal output where it only loads a line when I touch the TTP223. Fixed that issue (as you will see in my code below in the comments) by using a couple of "If" statements instead of an "else if" statement (it worked flawlessly on my Ardu Nano on another project I had with the else if statement).

So the issue I am seeing now, no matter if I have the car running or just the IGN switched on, none of my PID's are displaying any data on the OLED display nor on the serial terminal. As you can see from the terminal output below, I get connected to the ELM327, the AT commands from the library go through and then it displays nothing further, only shows RPM as 0.

If I connect with my android BT Serial app and execute the AT commands manually, they all work and I also receive the correct HEX back (photo to follow in the next post).

I would appreciate your assistance with this as I am getting frustrated now 😭 

Here is my terminal debug output:

```
[   741][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   741][I][BluetoothSerial.cpp:685] _init_bt(): device name set
[   744][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
[   769][I][BluetoothSerial.cpp:938] connect(): master : remoteName
[   772][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   823][I][BluetoothSerial.cpp:507] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT started
[  4028][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4029][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  4042][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  4064][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 182
[  4378][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4378][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  4392][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  4414][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 193
[  4768][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4768][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  4782][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  4804][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 195
[  4859][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  4859][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  4873][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  4896][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 196
[  5448][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  5448][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  5462][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  5484][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 196
[  5506][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  5530][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  5553][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  5576][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 197
[  5811][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  5811][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  5824][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  5846][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 190
[  5868][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  5892][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  5915][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  5938][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 192
[  8781][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  8781][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[  8794][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[  8816][D][BluetoothSerial.cpp:477] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_RSSI 198
[  8901][I][BluetoothSerial.cpp:423] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT properties=3
[  8901][I][BluetoothSerial.cpp:426] esp_bt_gap_cb(): Scanned device: aa:bb:cc:11:22:33
[... cut ...]
[  9564][D][BluetoothSerial.cpp:467] esp_bt_gap_cb(): ESP_BT_GAP_DEV_PROP_COD 0x1f00
[... 628 lines cut ...]
    tft.print(kph);
    tft.print(" KPH");
   }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
  {
    myELM327.printError();
   }
}
  
void getMAF(){
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    maf = myELM327.mafRate();
    Serial.print("maf: ");
    Serial.println(maf);
    tft.setCursor(10, 60);
    tft.print(maf);
    tft.print(" g/s");
   }
else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
{
  myELM327.printError();
 }
}
```

enslinm123, 2022-09-30:
Output of the android bt serial terminal. I forgot to set Carriage Return options, but it should display better if I change it.

[image]

This is my quick and dirty Summary page:

[image]

Showing the next page by pressing the TTP223. It changes quicker then I could touch it 🤣 

[image]

PowerBroker2, 2022-09-30:
I'm having a hard time understanding what's going on here, but I have a few suggestions:

1. Make sure you finish parsing the last response from the ELM327 before requesting a new PID
2. "NO DATA" usually means the car doesn't have that data available to give to the ELM327 in the first place
3. Verify you can get each PID **individually** (i.e. request one PID and don't request the others, but cycle through all PIDs you want to use)
4. Issues are mainly for bugs/feature requests. You'll probably get more in-depth/faster help with project styled questions on the Arduino forum

PlastiBots, 2023-01-09:
@enslinm123, not sure if you solved your issue, but having a look at your code, I think one issue is you are making the request for the value after ELM_SUCCESS (   iat = myELM327.intakeAirTemp(); is inside this).  However, I don't think you will get a success as there's nothing being checked.  Look at the examples provided as you have to request the PID outside of this. Also, per PowerBroker2's note, only do one at a time.

```
void getIAT(){
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    iat = myELM327.intakeAirTemp();
    Serial.print("iat: ");
    Serial.println(iat);
    tft.setTextColor(WHITE, BLACK);    
    tft.setTextScale(1);      
    tft.setCursor(10, 60);
    tft.print(iat);
    tft.print(" C");
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
  {
    myELM327.printError();
   }
}
```

enslinm123, 2023-01-13:
> @enslinm123, not sure if you solved your issue, but having a look at your code, I think one issue is you are making the request for the value after ELM_SUCCESS ( iat = myELM327.intakeAirTemp(); is [...]

Hey bud,

No I have not solved this issue and gave up and put everything back in my parts box for now, lol.

What I was trying to do is take the multiple PID's example (as the lib is non blocking) and adapt it for my purposes where I have a LCD displaying multiple items on a single screen as well as have the ability to flip between screens for a more detailed view of each PID. 

I had a quick glance at my sketch again and I think I know where the problem lies. Comparing to the [multiple PID](https://github.com/PowerBroker2/ELMduino/blob/master/examples/multiple_pids/multiple_pids.ino) example I need to adapt my code to reflect the correct flow as per below copied from the example and switch the obd states. 

I'm still learning how to work with cases and reading up a bit on it and glancing at my sketch I can see a lot of improvements I need to make, lol.

Here's the excerpt from the example I'm referring to:
```
typedef enum { ENG_RPM,
              SPEED } obd_pid_states;
obd_pid_states obd_state = ENG_RPM;

float rpm = 0;
float mph = 0;
```

I honestly think I was trying to over complicate things too soon instead of growing the project bit by bit. I did however have slight success with a single PID for rpm to display it on the serial monitor, but it was not updating on my LCD after which I decided to pack it away to look into again at another stage.

But thank you for pointing that out, I'm definitely going to look into it again and comb through my sketch with a fine tooth comb and start fresh from the ground up.

closed 2023-01-18 by enslinm123