PowerBroker2/ELMduino #145: Support for non-standard PIDs

petkop12, 2022-10-24:
Hello, 
Congratulations for this good project.
I use it for Toyota Corolla.
I found some PIDs that are specific to Toyota and I want to use them.

Is there currently a function in this library that I can set the PID and get the result?

I check some PIDs in Torque. And get result successfully. But I don't know how to use Your library.
In Torque add this fields:

Name:    Fuel Remain
Mode and PID:    2129
OBD Header:       7C0
Equation 1:    A/2, %
Equation 2:    ((A*13.2)/100), liters

I don't know how many bytes are returned as information.

PowerBroker2, 2022-10-24:
The easiest thing to do for simple PIDs is call `processPID()` repeatedly until `nb_rx_state == ELM_SUCCESS` using the PID data you just posted.

A more manual approach would require you to use `queryPID()` to "ask" the ELM327 for the PID data, then use `get_response()` to get the raw ELM327 response packet, then `findResponse()` to parse out the data from the packet, and (optionally) use `conditionResponse()` to apply a given "equation" to the data.

PowerBroker2, 2022-10-24:
You sure the `fuelLevel()` function won't work for you?

PowerBroker2, 2022-10-27:
`NO_DATA` means your car doesn't support that PID.

> For my second test, I try this code with queryPID :

Please see [this comment](https://github.com/PowerBroker2/ELMduino/issues/145#issuecomment-1289350326) on how to do custom PID queries.

petkop12, 2022-10-28:
When I try to read PID 2129 with Torque - I can get result. I see maybe true results. I am only comparing with the fuel shown on my dash board. And in Torque I can get result. But when enter correct header - 7c0, when try with another - No Data.

When I use esp32 and function    myELM327.fuel_remain()   - result is    NoData
[image]

When I use esp32 and functions  queryPID and   findResponse:
    if (myELM327.queryPID("2129")) {
        burn = myELM327.findResponse();    
        Serial.print("Test queryPID : "); Serial.println(burn);

I get result:    Received: 0080:6129000080001:00320000000000

But I don't know how to understand this answer.  Can You support me?

I see that You use exact the same PID in library:       processPID(21, 29, 1, 1 );

petkop12, 2022-11-07:
Can You support me to understand result?
Is this possible result ? 
0080:6129000080001:00320000000000

Part1:
0080
What is this?

Part2: 
6129000080001
this maybe a result of PID. But what is this:     "000080001"

Part3:
00320000000000
Is this result?
Is possible  "32"  to be useful result?
How can decode    "00320000000000"  ?

PowerBroker2, 2023-12-21:
Stale issue and might be fixed anyway with the recent changes to main branch

closed 2023-12-21 by PowerBroker2