PowerBroker2/ELMduino #146: ERROR: ELM_TIMEOUT - ELM327 v2.1 - 1 board

robsonsilvaandrade, 2022-10-27:
I used sample code but I get no response from OBD. When I use the ELM Terminal App on my smartphone and send 01, it returns normally. How should I proceed?

Follow Arduino Terminal debug and ELM Terminal App.

[image]

ARDUINO IDE TERMINAL
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: A
	Received char: T
	Received char: _
	Received char: D
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATDOK
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: A
	Received char: T
	Received char: _
	Received char: Z
	Received char: \r
	Received char: \r
	Received char: \r
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char: _
	Received char: v
	Received char: 2
	Received char: .
	Received char: 1
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATZELM327v2.1
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char: _
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
Timeout detected with overflow of 0ms
Received: 
ERROR: ELM_TIMEOUT
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
Timeout detected with overflow of 0ms
Received: 
ERROR: ELM_TIMEOUT

PowerBroker2, 2022-10-27:
The car/ELM327 is not responding to the PID query, so the lib times-out. Try using a [serial pass through](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino) to see if you can successfully do manual PID queries _with the Arduino_.

robsonsilvaandrade, 2022-10-29:
I followed your directions using the code indicated and got the following responses when manually submitting PIDs 010C and 0105: NO DATA

PowerBroker2, 2022-10-29:
If you get `NO DATA` that means the ELM327 couldn't get the data from your car for some reason. Maybe try a different ELM327

robsonsilvaandrade, 2022-10-29:
But the same ELM327 responds normally when I use the App Terminal on the smartphone.  Look:

[image]

robsonsilvaandrade, 2022-11-01:
How should I proceed?

jimwhitelaw, 2023-12-12:
I know it's late (a year later!), but for others who may encounter this issue, I have a couple comments.

- The App Terminal example in the image above shows the "SEARCHING..." response. What's happening is that the ELM327 device is detecting the correct OBDII protocol with the vehicle's ECU (which can be a slow process) and then eventually returning with a response code.
- In the code and output example, the program is timing out while waiting for that initial protocol detection. This timeout issue is resolved in later versions of ELMduino, so updating the lib to v 3.1.0 or later should help.

closed 2023-12-21 by PowerBroker2