PowerBroker2/ELMduino #148: Why am I getting wrong return values from the software, but the raw values are correct.

Paulus4711, 2022-11-13:
**Hello,
Congratulations and many thanks for this good project.**

I ask for your help!

I use the simple example ESP32 over Bluetooth.
ESP32_Bluetooth_Serial.ino
_________________________________________________________________________________________________________________________________________
```
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;
uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }
  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }
  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```
_______________________________________________________________________________________________________________________________________

**The output looks like this:**

```
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: F
	Received char: A
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410C0FA0
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 160
	responseByte_1: 15
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 884
```
_______________________________________________________________________________________________________________________________________
"884" for the speed is unfortunately wrong.
**1000 rpm are right for engine speed .**

This results from the raw values:
Single response detected
64-bit response:
responseByte_0: **160**  **"B"**
responseByte_1: **15**   **"A"**
responseByte_3: 0
responseByte_4: 0
.......
see calculation: https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_00

formula  **( 256 x A + B ) /4**

### ( 256 x 15 + 160) / 4 = 1000 It's right
**Please where is the problem, what am I doing wrong, or is it the code?**
 _______________________________________________________________________________________________________________________________________
Unfortunately, the mistakes are repeated, e.g. at the coolant temperature engineCoolantTemp();

Service: 1
**PID: 5**
Normal length query detected
Query string: 01051
Clearing input serial buffer
Sending the following command/query: 01051
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 5
	Received char: 7
	Received char: 8
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410578
Expected response header: 4105
Single response detected
64-bit response: 
	responseByte_0: 120
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
**engineCoolantTemp: 168**

responseByte_0: 120  **"A"**

formula:    A -40
**120 - 40 = 80 Are right for  engineCoolantTemp**
### 
### Thank you very much for your help

[image]

[image]

PowerBroker2, 2022-11-13:
Your concerns are valid. Another issue is the query string. It's saying `Query string: 010C1`, but it should be `Query string: 010C`. Not sure what's going wrong there. The interesting thing is that the library seems to be parsing the ELM327's A and B bytes correctly (as you pointed out), but for whatever reason is failing to combine those values, probably in the `conditionResponse()` function. I've relooked through the related functions and can't find any bugs that might cause this issue. You might need to do some more debugging to find out what is happening in the `findResponse()` and `conditionResponse()` functions. My guess is that `numPayChars` is equal to 3 when it should be 2. This seems probable because your ELM327 seems to be returning extra carriage returns (although the lib should be throwing them out) plus for the engine temp response has at least one extra response char.

Again, I'd suggest poking around the `findResponse()` and `conditionResponse()` functions to see what's really going on. You might also want to check out the main branch that has some updates since v3.0.2 that might fix things for you and would appreciate the feedback.

Also, thank you so much for your in depth explanation of the issue!

Paulus4711, 2022-11-14:
**Thanks for the support.**

It's not the ELM327, I tried three different ones and always got the same calculation errors.

I switched to the ELMDuinoVersion 3.0.1 library and the output was correct.

______________________________________________________________________________________________________________________________

```
ELMDuino in Version 3.0.1 

Service: 1
PID: 5
Normal length query detected
Query string: 01051
Clearing input serial buffer
Sending the following command/query: 01051
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 5
	Received char: 7
	Received char: 8
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410578
Expected response header: 4105
Single response detected
64-bit response: 
	responseByte_0: 120
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
engineCoolantTemp:  80
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: F
	Received char: A
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410C0FA0
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 160
	responseByte_1: 15
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
RPM: 1000
Leerlauf, schnelle PIDs abfragen bei 173.986 s
Service: 1
PID: 5
Normal length query detected
Query string: 01051
Clearing input serial buffer
Sending the following command/query: 01051
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 5
	Received char: 7
	Received char: 8
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410578
Expected response header: 4105
Single response detected
64-bit response: 
	responseByte_0: 120
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
engineCoolantTemp: 80
```
______________________________________________________________________________________________________________________________

After switching back to the ELMDuinoVersion 3.0.2 library, the output **was ALSO correct.**

The error was probably due to my incorrect **_ELMduino-master.zip_** library.

The thing with the Query string: 010C**1** , but it has remained

```
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1

```
### Thank you very much and keep up the good work, it gives people a lot of JOY

PowerBroker2, 2022-11-14:
Glad it's working now!

closed 2022-11-14 by PowerBroker2

patfelst, 2022-11-14:
Hi guys, the extra "1" at the end of the query string was added by me when I converted the library to non-blocking (although it's unrelated to the non-blocking code, just thought it was worth including at the same time).

The extra "1" speeds up responses from the OBD dongle significantly as it means the dongle will not wait around for additional responses, it will only wait for 1, and then return before waiting another timeout period. If you're sampling PIDs at full speed, you'll achieve much higher frame rate with this on.

Note you can set the number of responses expected as the 3rd parameter in the `ProcessPID()` function, e.g. to query RPM
```C++
float ELM327::rpm()
{
	return processPID(SERVICE_01, ENGINE_RPM, 1, 2, 1.0 / 4.0);
}
```
This is commented in the code at https://github.com/PowerBroker2/ELMduino/blob/eabbb1a2e8e82d8fed1b758c0d893bc46efbe171/src/ELMduino.cpp#L466

We probably should modify the code such that if this is set to zero, then the code does not append the "1" so the behaviour is as per the old library behaviour.

PowerBroker2, 2022-11-14:
Ah, thank you for the explanation!

> We probably should modify the code such that if this is set to zero, then the code does not append the "1" so the behaviour is as per the old library behaviour.

Yes, that would be a good idea