PowerBroker2/ELMduino #150: Continuously Receiving "NO DATA" from both Car and ECUsim 2000

John-A-Bonilla, 2022-11-29:
Hi, thank you for making such an awesome library! I have been trying to use it to talk to both my ECUsim 2000 and my car with seemingly no luck. I am using an ESP32 along with an amazon Wi-Fi ELM327 adapter. I know that the ESP32 is making a connection with the ELM327 but it does not seem to be able to communicate with the ECUsim 2000 as the light "link" does not work. I have been able to use this setup in the past with a different code but I am unsure of what is causing this issue. If you have any recommendations as to what I should try please let me know!

PowerBroker2, 2022-11-29:
Sorry, I've never used a Wi-Fi OBD-II scanner, so I don't have anything for you. You might want to ask in the Arduino forum or perhaps some car hacking forums

John-A-Bonilla, 2022-11-29:
> Sorry, I've never used a Wi-Fi OBD-II scanner, so I don't have anything for you. You might want to ask in the Arduino forum or perhaps some car hacking forums

Just curious has this program centered more around the Bluetooth OBD scanner? If so I have one that I can try.

PowerBroker2, 2022-11-29:
Yeah, I've only used the standard bluetooth scanners (not BLE!)

closed 2023-03-02 by PowerBroker2