PowerBroker2/ELMduino #151: >>>ERROR REQUESTING PID-ELM327 WITH ESP32<<<

Atorres9506, 2022-12-03:
HELLO. Your work is excellent, I am using your library in a HUD project for a vehicle. at the beginning I used an arduino one WITH AN elm327 v2.1 and it worked perfectly, but I need to implement it with an ESP32 and I got a little problem.
when connecting the ELM327 using "BluetoothSerial.h" the communication is established but when requesting the PID I get the response "Timeout detected with overflow of 0ms" and "ERROR: ELM_TIMEOUT"
I read all the forums related to this problem but I can't find a solution.
However, I did a test by connecting both devices to my laptop's bluetooth and internalizing the COM ports to check the data with a serial analysis software and surprise, it started to work correctly, I don't know the cause.
I attach the serial monitor data in both cases and the code used.
[image]
`#include "BluetoothSerial.h"
#include "ELMduino.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;
String MACadd = "66:1E:32:1E:39:69"; // elm address
uint8_t address[6]  = {0x66, 0x1E, 0x32, 0x1E, 0x39, 0x69};
void setup()
{

  DEBUG_PORT.begin(115200);
  
  ELM_PORT.begin("ArduHUD", true);
  SerialBT.setPin("1234");
  if (!ELM_PORT.connect(address))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}`

[terminal_2.txt](https://github.com/PowerBroker2/ELMduino/files/10145138/terminal_2.txt)
[terminal_ERROR.txt](https://github.com/PowerBroker2/ELMduino/files/10145139/terminal_ERROR.txt)

I appreciate your prompt response.

Atorres9506, 2022-12-03:
The problem lies in the direct communication between the ELM327 and the ESP32.
I don't know the cause

jimwhitelaw, 2023-11-23:
Yes, I know this is old now, but I've encountered the same thing, and I believe I know the cause.

According to the ELM documentation (https://www.elmelectronics.com/DSheets/ELM327DSH.pdf  page 34), when setting the protocol to 0 (AUTO) the ELM327 doesn't actually initiate the protocol search until the first OBD command is sent. You can see that in your log file - the AT TP A0 command is sent, with response OK and message "connected to ELM327", but the ELM -> ECU connection is not actually established yet. So when you send your first OBD query "010C1", the response is "SEARCHING..." and then it times out. The fix in your case is to set a longer timeout (remember that the value you set is quartered for some reason, so set it 4x longer than you actually want). 

I believe that the underlying issue is that initializeELM() is returning true (connected) before it has actually established a connection, it has only set the protocol to AUTO. In order to initiate & complete the protocol detection process, an OBD command needs to be sent with a long timeout to allow the protocol detection to take place. Only after that has completed, should the AT DP command be sent to find the actual protocol that is in use.

closed 2023-11-28 by PowerBroker2