PowerBroker2/ELMduino #152: Wonky fuel rate readings on ESP32-Pico V4

Cynicroute, 2023-01-13:
Hi everyone. First of all, kudos @PowerBroker2 for this brilliant library. It's simplified even getting started with OBD-2 for a complete novice like me.

I've been working on grabbing readings from a generic ELM327 on a Pico V4 (battery, fuel level, fuel rate). While I can connect reliably, getting readings is inconsistent. Aside from the inconsistency, the one reading that comes back wonky every time is the fuel rate. I suspect there's a divisor that i'm missing. Thoughts?

```
#include <BluetoothSerial.h>
#include <ELMduino.h>

BluetoothSerial     SerialBT;
#define ELM_PORT    SerialBT
#define DEBUG_PORT  Serial
//#define REMOVE_BONDED_DEVICES 1   // Set 0 to disable

ELM327 myELM327;

uint32_t rpm = 0;
uint32_t volts = 0;
uint32_t litres = 0;
uint32_t lph = 0;

// For future 'if last' statement
//float lastVolts;
//float lastLitres;
//float lastLPH;

void setup()  {
  
  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("OBDII", true);  //true enables debug mode
  delay(500);
  SerialBT.setPin("1234");
  
  DEBUG_PORT.println("Initializing...");
  
  if (!ELM_PORT.connect("OBDII"))
    {
      DEBUG_PORT.println("Stage 1 failed. Could not connect to ELM Device.");
      while(1);
    }

  if (!myELM327.begin(ELM_PORT, true, 2000))
    {
      Serial.println("Stage 2 failed. Could not connect to OBD-II.");
      while (1);
    }

  Serial.println("Connected to ELM327. Reading Data...");
  
}

void loop() {
  
  //--- RPM
  float getRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS)
    {
      rpm = (uint32_t)getRPM;
      Serial.print("Revs: "); 
      Serial.print(rpm);
      Serial.print(" RPM");
    }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
  delay(100);

  //--- BATTERY VOLTAGE
  float getVolts = myELM327.batteryVoltage();
  if (myELM327.nb_rx_state == ELM_SUCCESS)
    {
      volts = (uint32_t)getVolts;  
[... 311 lines cut ...]
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
Fuel Consumption: 297 L/hService: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: 7
	Received char: 7
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410C1770
Expected response header: 410C
Single response detected
```

PowerBroker2, 2023-01-13:
Glad you found the library useful! From a quick look I see you're calling multiple queries one right after the other before guaranteeing you fully parsed the responses of the initial query. Since this lib is asynchronous/non-blocking, if you don't do this check, it'll be like asking someone several questions in a conversation without giving them a chance to respond. This probably confuses which pid gets parsed to what variable (although I can't guarantee that's the only issue).

Take a look at [this example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/multiple_pids/multiple_pids.ino) that shows how to query multiple PIDs in a single sketch using a switch statement

Cynicroute, 2023-01-13:
Makes sense. I'll give it a whirl and see what comes out.

Thank you!

Cynicroute, 2023-01-13:
Well, I updated the sketch and it sort of works. It's pulling RPM, Volts, and Litres, but not L/h. I even threw in a small delay to slow things down a bit.

```
#include <BluetoothSerial.h>
#include <ELMduino.h>

BluetoothSerial     SerialBT;
#define ELM_PORT    SerialBT
#define DEBUG_PORT  Serial
//#define REMOVE_BONDED_DEVICES 1   // Set 0 to disable

ELM327 myELM327;

typedef enum { ENG_RPM, BATT_V, FUEL_L, FUEL_R }
    obd_pid_states;
    obd_pid_states obd_state = ENG_RPM;

uint32_t rpm = 0;
uint32_t volts = 0;
uint32_t litres = 0;
uint32_t lph = 0;

// For future 'if last' statement
//float lastVolts;
//float lastLitres;
//float lastLPH;

void setup()  {
  
  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("OBDII", true);  //true enables debug mode
  delay(500);
  SerialBT.setPin("1234");
  
  DEBUG_PORT.println("Initializing...");
  
  if (!ELM_PORT.connect("OBDII"))
    {
      DEBUG_PORT.println("Stage 1 failed. Could not connect to ELM Device.");
      while(1);
    }

  if (!myELM327.begin(ELM_PORT, true, 2000))
    {
      Serial.println("Stage 2 failed. Could not connect to OBD-II.");
      while (1);
    }

  Serial.println("Connected to ELM327. Reading Data...");
  
}

void loop() {

switch (obd_state)
{
    case ENG_RPM:
    {  //--- RPM
    float getRPM = myELM327.rpm();
    if (myELM327.nb_rx_state == ELM_SUCCESS)
        {
        rpm = (uint32_t)getRPM;
        Serial.print("Revs: "); 
        Serial.print(rpm);
        Serial.print(" RPM");
        obd_state = BATT_V;
        }
    else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
        {
        myELM327.printError();
[... 252 lines cut ...]
	responseByte_6: 0
	responseByte_7: 0
Fuel Tank: 14 %Service: 1
PID: 94
Normal length query detected
Query string: 015E1
Clearing input serial buffer
Sending the following command/query: 015E1
	Received char: N
	Received char: O
	Received char: _
	Received char: D
	Received char: A
	Received char: T
	Received char: A
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: NODATA
ELM responded with errror "NO DATA"
Received: NODATA
ERROR: ELM_NO_DATA
Service: 1
```

PowerBroker2, 2023-01-13:
If the ELM327 responds with `NO DATA` it means for some reason your car's diagnostics computer does not support that PID

Cynicroute, 2023-01-14:
Strange. I must be calling on the wrong PID. According to the ELM app on my phone, it's capturing 'calculated instant fuel rate' in L/h. (2.4 at idle... seems reasonable). I have some digging to do.

closed 2023-12-21 by PowerBroker2