PowerBroker2/ELMduino #153: connecting arduino to elm by wire without bluetooth

cercelmarius, 2023-01-29:
Hello!
super cool library. it helped me a lot to find out what I was trying to do through strings and mathematical calculations..
I'm a beginner.. but with the help of google and youtube I managed to make this sketch perfectly functional.

my request would be if someone can help me modify the code to be functional through rx tx connection with wires from elm to arduino without the need for a buetooth connection.
the bluetooth connection sometimes gives me an error and does not connect to elm.
thanks!
----------------------------------------------------------------------------------------------
#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial Bluetooth(10, 11);        //rx tx   uno
//SoftwareSerial Bluetooth(52, 53);      //rx tx mega2560
ELM327 myELM; 

int readRPM() //...........................................................turatie
{
  int tempRPM = myELM.rpm();
  while (myELM.nb_rx_state == ELM_GETTING_MSG) {
    tempRPM = myELM.rpm();
  }
  if (myELM.nb_rx_state == ELM_SUCCESS)
  {
    return tempRPM;
  }
  else
    return 0;
}

int readSpeed()  //.........................................................viteza  
{
  int tempSpeed = myELM.kph();    //conteaza ce este dupa myELM
  while (myELM.nb_rx_state == ELM_GETTING_MSG) {
    tempSpeed = myELM.kph();
  }
  if (myELM.nb_rx_state == ELM_SUCCESS)
  {
    return tempSpeed;
  }
  else
    return 0;
}

void setup() {
//initializare ELM si Bluetooth
    Serial.begin(115200);
    Bluetooth.begin(9600);
    myELM.begin(Bluetooth, true, 2000);  //protocol automat
    //myELM.begin(Bluetooth, true, 2000, ISO_9141_5_BAUD_INIT);  //protocol specificat
}
void loop() {
 // int RPM = readRPM();
  int Viteza = readSpeed();

  Serial.println();
 // Serial.print("RPM: ");
 // Serial.println(RPM);
  Serial.print("Viteza: ");
  Serial.println(Viteza);
  Serial.println(); 
}
----------------------------------------------------------------------------------------------

PowerBroker2, 2023-01-29:
Should work as-is (assuming the wiring and baud rate are correct)

cercelmarius, 2023-01-31:
thank you for the answer.
I will try in the next few days and I will come back with news.
the assembly is already in the integrated cluster and you will have to disassemble it
[https://youtu.be/KIz3xUZbsLk](url)

cercelmarius, 2023-02-08:
Hello!
I'm back with news

I connected the elm by wire to the arduino.
I removed the bluetooth module of elm, and bluetooth  connection between elm and hc-05

I soldered wires from rx and tx elm board to rx and tx arduino board
I flashed the same sketch above which is functional through the hc-05 connection.

I received the answer below by following the serial monitor.
I changed the baud rate to serial.begin. tried with 19200, 9600, 115200.
however, I received the same answers

---------------------------------------------------------------------------
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: ⸮
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: ⸮
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT ST 00
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 0ms
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 1ms
Setting protocol via AT SP %c did not work
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
Timeout detected with overflow of 0ms

RPM: 0

Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
	Received char: ⸮
-----------------------------------------------------------------------------------------------------------------

PowerBroker2, 2023-02-08:
Glad it works!

closed 2023-02-08 by PowerBroker2