PowerBroker2/ELMduino #154: Did not get right Data from ELM327 - "Long query detected"

ReelHiJacker, 2023-02-10:
I'm building a DPF Monitor based on ESP32 for my Opel Insignia with the help of this code https://www.instructables.com/Opel-DPF-Indicator-Monitor/
Because it did not work for me, i tried building a test sketch. When i'm using only rpm Data, i'm getting the right information. But when i add the parameter "soot level", i did not get any values.

So i checked Serial Monitor and this is what i get:
After
`Connected to ELM327`
this is what Serial Monitor shows:
`Service: 1`
`PID: 12`
`Normal length query detected`
`Query string: 010C1`
`Clearing input serial buffer`
`Sending the following command/query: 010C1`
`Service: 34`
`PID: 13162`
`Long query detected`
`Query string: 22336A1`
`Clearing input serial buffer`
`Sending the following command/query: 22336A0`
`	Received char: S`
`Service: 34`
`PID: 13162`
`Long query detected`
`Query string: 22336A1`
`Clearing input serial buffer`
`Sending the following command/query: 22336A0`
`	Received char: \r`

and then it get stucked an repeats over and over
`Service: 34`
`PID: 13162`
`Long query detected`
`Query string: 22336A1`
`Clearing input serial buffer`
`Sending the following command/query: 22336A0`
`	Received char: \r`

So my next attempt was to check if the pid is correct, so i used your test sketch an this is what i get in serial monitor for the soot PID
`22336A`
`62336A4C`

So this seems to be correct, 4C is the right data what i want to see.  So ELM327 is working.

And now i'm stucked and have no idea what is wrong...

And here is my code of course. 
I tried many things, and read every issue here in github, but don't know what is the problem. Hope you can help me. I also tried different baud rates, it changes nothing...

```

#include "BluetoothSerial.h"
#include "ELMduino.h"

#include <TFT_eSPI.h>
#include <SPI.h>
#include "Free_Fonts.h"
#include "opel.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

uint8_t address[6] = {0x13, 0xE0, 0x2F, 0x8D, 0x52, 0xF3};

ELM327 myELM327;

int32_t rpm = -1;
int32_t soot = -1;

TFT_eSPI tft = TFT_eSPI(135, 240);

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  tft.init();tft.setRotation(3);
  tft.setSwapBytes(true);
  tft.pushImage(0, 0, 240, 135, opel);
  delay(1000);

  if (!ELM_PORT.connect(address))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
      tft.fillScreen(TFT_BLACK);
      tft.setTextColor(TFT_WHITE, TFT_BLACK);
      tft.setCursor(10,40);
      tft.setTextFont(4);
      tft.println("OBD error 1");     
      delay(2000);
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
      tft.fillScreen(TFT_BLACK);
      tft.setTextColor(TFT_WHITE, TFT_BLACK);
      tft.setCursor(10,40);
      tft.setTextFont(4);
      tft.println("OBD Error 2"); 
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop() {

  float tempRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (int32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
      printError();

  if (myELM327.queryPID(34,13162))
  {
    int32_t tempSoot = myELM327.findResponse();
    
    if (myELM327.nb_rx_state == ELM_SUCCESS)
    {
      soot = tempSoot;
      Serial.print("DPF soot: "); Serial.println(soot);
    }
    else
      printError();
  }

      oneScreen(rpm, soot);
      delay(1000);

}

//void oneScreen (uint8_t dpfSoot, uint16_t dpfKm, uint8_t dpfBurn, uint8_t fuel, uint16_t rpm){
void oneScreen (uint16_t rpm, uint8_t dpfSoot){

      if (dpfSoot > 130) {dpfSoot = 0;}
//      if (dpfKm > 999) {dpfKm = 0;}
//      if (dpfBurn > 11) {dpfBurn = 0;}
//      if (fuel > 100) {fuel = 0;}
      if (rpm > 7000) {rpm = 0;}
      
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);

  tft.setFreeFont(NULL);

  tft.drawString("DPF:", 10, 5, 4);
  tft.drawString(String(dpfSoot), 130, 5, 4);
        
//  tft.drawString("DPF km :", 10, 31, 4);
//  tft.drawString(String(dpfKm), 130, 31, 4);
          
//  tft.drawString("Treibstoff :", 10, 57, 4);
//  tft.drawString(String(fuel), 130, 57, 4);
        
  tft.drawString("RPM :", 10, 83, 4);
  tft.drawString(String(rpm), 130, 83, 4);

//  tft.drawString(String(dpfBurn), 220, 125, 1);
  
}

void printError()
{
  if (myELM327.nb_rx_state == ELM_SUCCESS) 
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.nb_rx_state == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.nb_rx_state == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.nb_rx_state == ELM_GARBAGE)
    Serial.println(F("\tERROR: ELM_GARBAGE"));
  else if (myELM327.nb_rx_state == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.nb_rx_state == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.nb_rx_state == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.nb_rx_state == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.nb_rx_state == ELM_GENERAL_ERROR)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR")); 

  delay(500);
}

```

seb30, 2023-02-10:
I'm also driving a B20DTH and using successfully this lib on a ESP32. Check out my repo, maybe it helps:

https://github.com/seb30/DPFBurn

ReelHiJacker, 2023-02-10:
@seb30 Thanky you, i will try this... it looks a way better than my old lib... ;-)

seb30, 2023-02-10:
My last change was adding a buzzer to acoustically notify about beginning and end of a regen.

ReelHiJacker, 2023-02-11:
@seb30 Your lib is working perfect, thank you very much. I tried it the first time when i was driving home and exact the moment i wan't to turn of the car it starts burning dpf free :-) So very usefull as i expected. Thank you very much for sharing!!

closed 2023-02-11 by ReelHiJacker

seb30, 2023-02-11:
@ReelHiJacker So, this was just in time. I'm glad that I could help you.