PowerBroker2/ELMduino #156: Data refresh rate problem

DanteCecchetti, 2023-02-13:
I have a problem with the data refresh rate. I am using the code that you put on the home page of the library

#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial mySerial(2, 3); // RX, TX
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(38400);
  mySerial.begin(38400);
  myELM327.begin(mySerial);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}

It turns out that my HC05 module connects perfectly to the obd2 but the rpm data is refreshed at a very varied rate, normally every 2 seconds, but at times it takes 4 seconds and at others only 1. Do you have any idea how to make the communication be smoother and faster, because my goal is to do a light shift, but to be able to do that I need more data speed. Honestly, I would like to achieve a refresh rate similar to what I get with my cell phone using the torque lite application.
What do you recommend I do?

PowerBroker2, 2023-02-13:
Maybe using a different OBD2 scanner will help. Sometimes the chinese knock offs are difficult to get working right. Otherwise, could you attach a full log of all debug printouts during one of these test runs?

DanteCecchetti, 2023-02-13:
Ok, now I try again and I send you the full log of all debugging prints, but one question, how do I enable that feature? I'm sorry for asking you, I started doing the tests yesterday and it's difficult for me to adapt to the library and understand 100% all the commands it does. And one last question, did you manage to have fluid communication between a hc 05 module and an obd? and how fluid was it? Because I am analyzing the possibility of getting a scanner that can connect it directly to the arduino without the need to use bluetooth to increase speed.

PowerBroker2, 2023-02-14:
I use a Teensy 4.1 with an HC-05 connected via classic BT to [this OBD2 scanner](https://www.amazon.com/gp/product/B005NLQAHS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) and it works great.

I'd suggest using a new OBD2 scanner.

You might also want to try [this test sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino). **NOTE**: This test sketch implies your ESP32 has a built in BT receiver - does yours have this?

DanteCecchetti, 2023-02-15:
Hello, thanks for answering. I don't use an esp32, right now I'm trying to use an arduino with a hc 05 module. Also, in a few days I'm going to change my obd for one similar to the one in this video:

https://www.youtube.com/watch?v=y1XJ7fCBMLY&list=WL&index=2&ab_channel=NOOBStyle%E8%8F%9C%E9%B3%A5%E9%A2%A8%E6%A0%BC

I want to see if it can really be done as shown. Otherwise, as a last option, I will try to buy the obd that you sent me, but I find it difficult because in my country it is not available, if you know of any other obd model that you know that has worked, it would be very useful for me, so I know which of all I can get in my country.
Thanks again for answering

DanteCecchetti, 2023-03-26:
Hello, there I was able to buy the obd that you used to create the library and it really was a very abrupt change, from only having 1 or 2 data every 2 seconds to having 21 data approximately every 1 second. However, I wanted to ask you if it has ever happened to you that the elm stops giving data for a period of time (I have attached the data obtained). Do you have any idea how I can fix it?
And I also wanted to ask you if the configuration that you put in the esp32 example is enough for it to work, since I am currently doing the test with an arduino and a hc 05 but I would like to make the final prototype with an esp32 and honestly I never programmed one and I don't know what parameters I have to modify in the bluetooth configuration.

15:30:36.856 -> RPM: 853
15:30:36.903 -> RPM: 852
15:30:36.950 -> RPM: 851
15:30:36.997 -> RPM: 854
15:30:37.044 -> RPM: 848
15:30:39.012 -> Received: 410C0D41
15:30:39.012 -> ERROR: ELM_TIMEOUT
15:30:39.153 -> RPM: 830
15:30:41.168 -> Received: 410C0CFB
15:30:41.168 -> ERROR: ELM_TIMEOUT
15:30:41.308 -> RPM: 832
15:30:41.355 -> RPM: 833
15:30:41.402 -> RPM: 838
15:30:41.449 -> RPM: 840
15:30:41.496 -> RPM: 837
15:30:41.542 -> RPM: 843
15:30:41.576 -> RPM: 842
15:30:41.622 -> RPM: 837
15:30:41.669 -> RPM: 835
15:30:41.716 -> RPM: 835
15:30:41.716 -> RPM: 834
15:30:41.810 -> RPM: 834
15:30:41.857 -> RPM: 840
15:30:41.904 -> RPM: 836
15:30:41.950 -> RPM: 833
15:30:41.997 -> RPM: 0
15:30:42.044 -> RPM: 831
15:30:42.091 -> RPM: 837
15:30:42.138 -> RPM: 842
15:30:42.185 -> RPM: 841
15:30:42.231 -> RPM: 841
15:30:42.278 -> RPM: 844
15:30:42.278 -> RPM: 841
15:30:42.325 -> RPM: 836
15:30:42.372 -> RPM: 840
15:30:42.419 -> RPM: 845
15:30:42.466 -> RPM: 844
15:30:42.513 -> RPM: 844
15:30:42.559 -> RPM: 843
15:30:42.606 -> RPM: 842
15:30:42.653 -> RPM: 844
15:30:42.700 -> RPM: 847
15:30:42.747 -> RPM: 843
15:30:42.794 -> RPM: 839
15:30:42.841 -> RPM: 841
15:30:42.841 -> RPM: 840
15:30:42.888 -> RPM: 837
15:30:42.935 -> RPM: 838
15:30:42.981 -> RPM: 844
15:30:43.028 -> RPM: 846
15:30:43.075 -> RPM: 847
15:30:43.122 -> RPM: 854
15:30:43.169 -> RPM: 850
15:30:43.216 -> RPM: 846
15:30:43.263 -> RPM: 846
15:30:43.309 -> RPM: 846
15:30:43.356 -> RPM: 850
15:30:43.403 -> RPM: 0
15:30:43.403 -> RPM: 851
15:30:43.450 -> RPM: 848
15:30:43.497 -> RPM: 847
15:30:43.544 -> RPM: 851
15:30:43.590 -> RPM: 847
15:30:43.638 -> RPM: 843
15:30:43.684 -> RPM: 843
15:30:43.684 -> RPM: 844
15:30:43.731 -> RPM: 842
15:30:43.778 -> RPM: 840
15:30:43.872 -> RPM: 843
15:30:43.919 -> RPM: 840
15:30:43.919 -> RPM: 833
15:30:43.966 -> RPM: 834
15:30:44.012 -> RPM: 834
15:30:44.059 -> RPM: 833
15:30:44.106 -> RPM: 834
15:30:44.153 -> RPM: 834
15:30:44.153 -> RPM: 832
15:30:44.247 -> RPM: 832
15:30:44.294 -> RPM: 837
15:30:44.341 -> RPM: 840
15:30:44.387 -> RPM: 844
15:30:44.434 -> RPM: 848
15:30:44.481 -> RPM: 847
15:30:44.528 -> RPM: 846
15:30:44.575 -> RPM: 845
15:30:44.622 -> RPM: 840
15:30:44.669 -> RPM: 842
15:30:44.715 -> RPM: 843
15:30:44.762 -> RPM: 840
15:30:44.762 -> RPM: 833
15:30:44.856 -> RPM: 841
15:30:44.903 -> RPM: 839
15:30:44.903 -> RPM: 837
15:30:44.950 -> RPM: 841
15:30:45.043 -> RPM: 846
15:30:45.043 -> RPM: 845
15:30:45.090 -> RPM: 845
15:30:45.137 -> RPM: 847
15:30:45.184 -> RPM: 847
15:30:47.198 -> Received: 410C0D25
15:30:47.198 -> ERROR: ELM_TIMEOUT
15:30:47.338 -> RPM: 840
15:30:47.385 -> RPM: 845
15:30:47.432 -> RPM: 851
15:30:47.479 -> RPM: 852
15:30:47.525 -> RPM: 853
15:30:47.573 -> RPM: 855
15:30:47.573 -> RPM: 853
15:30:47.666 -> RPM: 849
15:30:47.713 -> RPM: 849
15:30:47.760 -> RPM: 975
15:30:47.760 -> RPM: 844
15:30:47.854 -> RPM: 842

PowerBroker2, 2023-03-26:
I don't think I've ever had any serious issues with timeouts when using an HC-05. In terms of using an ESP32, you shouldn't need any advanced configuration - the example sketch should be enough.

DanteCecchetti, 2023-03-26:
So you don't know any solution or recommendation for that problem?

PowerBroker2, 2023-03-26:
Nope. Maybe if I see a detailed debug printout for a query that time'd out. Even then, it might just be a hiccup with the scanner or the car

ScarNomad, 2023-04-17:
Hello P_B and Dante,
I managed to get everything working correctly with an ESP32 and a bluetooth connection, although I too am suffering from low data refresh rate.
I was wondering if, by changing the ELM327.begin function to something like ELM327.begin(ELM_PORT, false, 2000, 0, 4, 4) so by reducing the data timeout to 4ms and the payload lenght to 4bytes (my longest pid request should be rpm so a 4 bit response should be the maximum i.e. 41 10 and 2bytes of actual data) i could speed up the refresh rate.
Thanks for the amazing work on the lib!

patfelst, 2023-04-18:
Hi @ScarNomad, changing the payload length is just an internal character buffer in the ESP32 so shortening that won't help. But you can try not sending the "allow long messages" `AT AL` command (see code below where I've commented this out). The library doesn't request any PIDs that return any more than 4 bytes anyway. This might prevent the ELM waiting around for additional bytes.

Its been a while since I optimised my setup so I can't remember every detail, but I achieved 20Hz frame rate reading about 5 PIDs in a loop using a bluetooth ELM327 clone. My settings are below.

You can also try the more aggressive auto adaptive timeout. The timeout is how long the ELM will wait for a response from the car before it determines a NO_DATA response. The ELM dongle by default by uses the standard adaptive timeout setting of `AT1` (and this library doesn't change it). `AT2` is a more aggressive version of AT1.

If `AT2` doesn't seem to help, then yes as per your code above try playing with the timeout value yourself, its the last parameter in `ELM327.begin()`. You'll need to add back in the `SET_TIMEOUT_TO_H_X_4MS, dataTimeout / 4)`, and probably put the adaptive timing back to AT1 by deleting the code that sets `ADAPTIVE_TIMING_AUTO_2`. But without sophisticated timing gear, you'll find it difficult to get a perfect timeout and you could miss receiving data, best to use auto adaptive timeouts as per above.

And set reception of headers off, they are not needed. So to do all of this, try the following by editing the library source code to the following:

```C++
bool ELM327::initializeELM(const char &protocol, const byte &dataTimeout)
{
	char command[10] = {'\0'};
	connected = false;

	sendCommand_Blocking(SET_ALL_TO_DEFAULTS);
	delay(100);

	sendCommand_Blocking(RESET_ALL);
	delay(100);

	sendCommand_Blocking(ECHO_OFF);
	delay(100);

	sendCommand_Blocking(PRINTING_SPACES_OFF);
	delay(100);

	// sendCommand_Blocking(ALLOW_LONG_MESSAGES);
	// delay(100);

	sendCommand_Blocking(ADAPTIVE_TIMING_AUTO_2); // ATAT1 = default timeout. ATAT2 = aggressive timeout
	delay(100);

	sendCommand_Blocking(HEADERS_OFF); // Add this so the ELM doesn't send headers which we ignore anyway
	delay(100);

	// Set data timeout
	// sprintf(command, SET_TIMEOUT_TO_H_X_4MS, dataTimeout / 4);
	// sendCommand_Blocking(command);
	// delay(100);

	// Set protocol
	sprintf(command, TRY_PROT_H_AUTO_SEARCH, protocol);
```
remainder of the function is the same.

ScarNomad, 2023-04-18:
Thank you so much for the response! 
At the moment i'm waiting for a better version of the elm327 bt dongle to ship (one that has the PIC18F25K80 chip) and that should help tremendously with the refresh rate.
Thanks again and i'll let you know if the changes were successful!

closed 2023-12-21 by PowerBroker2