PowerBroker2/ELMduino #157: myELM327.supportedPIDs_1_20()  not working as expected

lcsadmin, 2023-03-01:
Hi there,

I have been trying to obtain the supported PIDs. For my project, it is mandatory to obtain the supported PIDs before reading any other sensor.

Here is my loop:
void loop() {
    int32_t pids = myELM327.supportedPIDs_1_20();
    if (myELM327.nb_rx_state == ELM_SUCCESS) {
        Serial.print("PIDS: ");
        Serial.println(pids);
    }
}

And here is the debug response:
22:07:19.728 -> Normal length query detected.
22:07:19.728 -> Query string: 01001.
22:07:19.728 -> Clearing input serial buffer.
22:07:19.728 -> Sending the following command/query: 01001.
22:07:19.993 -> Received characters: 4100BE3EB811.
22:07:19.993 -> Expected response header: 4100.
22:07:19.993 -> Single response detected.
22:07:20.038 -> 64-bit response:
22:07:20.038 -> 	responseByte_0: 17
22:07:20.038 -> 	responseByte_1: 184
22:07:20.038 -> 	responseByte_2: 62
22:07:20.038 -> 	responseByte_3: 190
22:07:20.038 -> 	responseByte_4: 255
22:07:20.038 -> 	responseByte_5: 255
22:07:20.038 -> 	responseByte_6: 255
22:07:20.038 -> 	responseByte_7: 255
22:07:20.038 -> PIDS: -1
22:07:20.038 -> Service: 1
22:07:20.038 -> PID: 0

It seems that the hexadecimal code "4100BE3EB811" was received correctly, but when trying to obtain the list of 20 supported PIDs, the result obtained was "-1", which suggests that there might be an issue with the process of retrieving the PIDs.

Is there something I should fix? Should I create a custom `ELM327::processPID()` and then create my own response, because I expect an array of boolean values to know if the PID is active or not?

Thanks in advance
Lee

PowerBroker2, 2023-03-01:
There's an issue when it comes to typecasting the output. I'm not quite sure how to fix, but as a temporary workaround, you can use `responseByte_0` through `responseByte_3` to verify which PIDS are supported.

If anyone else has a fix, I'd greatly appreciate a PR.

jimwhitelaw, 2023-12-17:
Note:  I think the PR above may also fix the rpm = 884 issue, but I have not been able to test it yet.

jimwhitelaw, 2023-12-18:
Update: I have re-tested rpm() with emulator and vehicle and it is working properly now. I’m pretty sure the bit-shifting operation was the cause of the wonky rpm() output.

closed 2023-12-19 by PowerBroker2